Akkudoktor-EOS/EOS #5: Python/Flask Best Practices

njoerd114, 2024-09-15:
Wir sollte uns auf ein paar (Python) Best Practices innerhalb dieses Projektes einigen und diese befolgen.

- Code Struktur
- Code Style
- ...

Und diese dann auch per Test sicherstellen.

danimo, 2024-09-16:
Und per Check in PRs erwzwingen, sobald der Code einmalig in Form gebracht wurde.

drbacke, 2024-09-30:
Sehr gerne

michaelosthege, 2024-10-02:
Mit #76 habe ich einen PR auf gemacht der insbesondere `ruff` code style per `pre-commit` und GitHub Action erzwingt.

Die aktuelle Struktur ist für Python-Projekte etwas unüblich, und bei der Arbeit an #76 sind mir schon ein paar Dinge aufgefallen die schnell für mehr Übersichtlichkeit sorgen können:
* Dataclasses statt dicts verwenden (die kann man dann mit [`dataclasses-json`](https://pypi.org/project/dataclasses-json/) auch leicht (de-)serialisieren
* Den code in Submodules organisieren. z.B. `eos.forecast`, `eos.optimization`, `eos.devices`

Ich kann gerne dabei helfen.

drbacke, 2024-10-07:
I think it's done for now? 
Close?

closed 2024-10-07 by michaelosthege