Akkudoktor-EOS/EOS #7: Setup containerisieren

njoerd114, 2024-09-15:
Es wäre wünschenswert, das Projekt als Docker-Image zur Verfügung zu stellen und eine Anleitung (bspw. per docker-compose.yaml) zu geben, um das Projekt inkl. Abhängigkeiten Plattform-Unabhängig zu starten.

dsiebel, 2024-09-16:
@njoerd114 @drbacke kann ich mich gerne drum kümmern

danimo, 2024-09-16:
@dsiebel Super! Schicke gerne einen PR. Ich kann falls nötig noch Dinge in den Projekt-Settings freischalten.

dsiebel, 2024-09-16:
> * multi-platform
Hatte in erster Linie vor das ganze in eine `docker-compose.yaml` zu kippen.
Multi-arch wäre dadurch gegeben, dass das ganze da gebaut wird wo es auch läuft:
`docker compose up --build`

> * distroless?
Nicht unmöglich, aber reichlich schwierig umzusetzen für Python.

PR is allerdings schon fertig, habe die Kommentare hier erst jetzt gesehen.
https://github.com/Akkudoktor-EOS/EOS/pull/11

Konnte noch nicht verifizieren, dass das ganze so auch funktioniert, API-wise.

danimo, 2024-09-16:
Ich denke das `docker-compose.yml` ist für den ersten Aufschlag ok. Sobald wir Releases bauen, können wir fertige Pakete, auch multi-Platform, per CI bauen lassen und publisen.

dsiebel, 2024-09-17:
Ich denke, der hier kann nun geschlossen werden?

danimo, 2024-09-17:
Ja, die MacOS Port-Geschichte sollten wir nicht in Docker lösen, sondern durchgehnd. Dafür mache ich ein separates issue auf.

closed 2024-09-17 by danimo