Akkudoktor-EOS/EOS #10: API-Dokumentation

danimo, 2024-09-16:
Es gibt Swagger/openapi-Integrationen für Flask. Diese sollte verwendet werden, damit `/` auf diee Doku zeigt statt wie aktuellen auf den `/site-map`-Hack, der dann raus kann.

zappience, 2024-10-01:
Fastapi hätte eine Swagger-Funktionalität Out of the Box dabei. Gab es eine technische Entscheidung für Flask oder könnte man das zu Fastapi ändern?

danimo, 2024-10-01:
Nur dass wir nicht noch mehr Baustellen aufmachen sollten. Siehe die Anzahl der PRs. Wir brauchen zu allererst Tests, damit wir die PRs mit confidence mergen können-

NormannK, 2024-10-01:
traust du mir nicht? :-P 
ne mal im ernst, solltet ihr nicht und ich bin absolut bei dir. ich habe mir selber mal nen test geschrieben. aber da ich das noch nie richtig gemacht habe musste einer mal ein Fundament erstellen. 

grundsätzlich habe ich auch noch gar nicht verstanden wieso das ganze über flask läuft (ganz zu schweigen von dem Ansatz Node-Red einzubinden). Wiese keine reine Python implementation ohne extra layer. ALles andere kann man dann optional drauf setzen.

danimo, 2024-10-02:
Ich traue nicht mal mir selbst 🙈.

Die anderen Fragen musst du glaub ich an @drbacke richten 😅. Node Red vermutlich weil er es eh nutzt. Ich bin nur hier weil ich schon das ein oder andere Open Source Projekt (wenn auch nicht unbedingt python, wo ich auch nur Flask als Minimalframework kenne) betreut habe und wenn ich was gelernt habe dabei dann dass man bei Bestandscode vorsichtig sein muss, weil man beim refactoren super schnell was übersieht. Gerade weil die Dokumentation so minimal ist. Daher sollten wir vermutlich die Tests vorziehen und die PRs darauf rebasen, von den trivialeren vllt abgesehen.

drbacke, 2024-10-02:
Genau Nodered einfach nur, weil es bei mir schon da war. Aber es ist alles in JSON gehalten, sodass der Punkt relativ egal sein sollte. 
Ich habe das Install Video + die Beschreibung von Jörg, da findest du weitere Infos zu Benutzung. 
Flask als Server sehe ich jetzt nicht als großes Problem und finde das bei den aktuellen Punkten mit am unwichtigsten. Grundsätzlich habe ich aber nichts gegen eine andere Implementieren. Aber bitte eins nach dem anderen.

drbacke, 2024-10-04:
> Fastapi hätte eine Swagger-Funktionalität Out of the Box dabei. Gab es eine technische Entscheidung für Flask oder könnte man das zu Fastapi ändern?

Nein es gibt keine technischen Hintergrund. Wir können gerne wechseln. Nichts dagegen einzuwenden

drbacke, 2024-10-05:
I will test a flask swagger Implementation.

drbacke, 2024-10-07:
I've creates a new Issue,  with poll an discussion which Web Framework we want to use in the future. 
Lets discuss there

closed 2024-10-07 by drbacke