Akkudoktor-EOS/EOS #12: Wärmepumpe allgemeinerer Ansatz

drbacke, 2024-09-16:
Bisher ist ein sehr einfaches Wärmepumpen Modell unter modules\class_heatpump.py  und im EMS (class_ems.py) integriert. 
Aktuell ist es sogar ausgeschaltet und wird über die Standardlasten prognostiziert. 
Falls eine Steuerung der WP über das EOS vorgenommen werden soll, muss es aus den Last-Prognosen entfernt werden. 
Ansonsten würde die Last inklusive Wärmepumpe von der Lastprognose vorhergesagt werden und danach durch die EOS Planung nochmal. 
Todos: Besseres WP Modell / Lastprognose entprechend anpassen

rschueler, 2024-09-17:
Es scheint, dass das aktuelle Modell für Wärmepumpen nur für Luft-Luft-Wärmepumpen geeignet ist, bei denen die Leistungsaufnahme direkt von der Temperatur abhängt. 

Für Luft-Wasser-Wärmepumpen, die einen Speicher beinhalten, wäre ein komplexeres Modell erforderlich. Möglicherweise müsste hierfür eine spezielle Simulation des Wärme-Kreislaufs erstellt werden, die Speicher, Verluste und eine Wärme-Lastprognose berücksichtigt.

drbacke, 2024-09-30:
Es sollten hier am besten verschiedene Wärmepumpenmodelle vorgesehen werden.

closed 2024-12-10 by drbacke