Akkudoktor-EOS/EOS #27: How to proceed?

NormannK, 2024-09-18:
I'm a bit unsure about how to move forward. It would be really helpful if someone could set clear steps so we know what to do first. Right now, there’s a lot that needs to be done.

I think we should start by making the code easier to read like translating everything into English and cleaning up the dependencies. After that, we could improve the setup process, so more people can join and help faster.
What do you think?

danimo, 2024-09-19:
Agreed: clean up, set a linting standard, then enforce it for PRs should be the first steps. Translation can happen rather independently.

danimo, 2024-09-19:
See #3, #4

NormannK, 2024-09-20:
also some basic automated testing would be great. cleaning the wrong stuff could break a lot.

danimo, 2024-09-20:
Nothing keeps anyone from adding them right now though, while I can see that things like translation, formatting rules, etc may need some kind of concensus and ordering. Can't have linting for pull requests without cleaning up the code first for example.

danimo, 2024-09-20:
I suggest using milestones to sort our priorities and labels for added information.

niklasbeinghaus, 2024-09-20:
I'll probably have some time to finish with my translation, afterwards I'd vote for a refactoring.

NormannK, 2024-09-20:
> also some basic automated testing would be great. cleaning the wrong stuff could break a lot.

i made a pull request with a test for flask_server.py. I never wrote something like that before

danimo, 2024-09-21:
I saw your PRs, but can't get around to review them until next week for personal reasons. Please have patience. If anyone else has the time, please feel free to review, I'll hit merge.

NormannK, 2024-09-22:
Don't worries, take your time. ill use my branch and iron out some bugs

fortrieb, 2024-09-26:
Hi, i can start to review some of the PRs.
If translation and minimal code clean up is started should Python type hints be added to make development process more convenient?

NormannK, 2024-09-27:
very good thanks!
yeah i intentionally did not add hints yet. i was missing the overview over the code.
id do it in a 2nd pass when moving the parameters to the config file or somebody else oc :-)

drbacke, 2024-09-30:
Thank you for your commitment, I have reviewed and committed some PR. I'll try to look through a few more over the next few days.

fortrieb, 2024-10-02:
I will start writing some unit test for these class, starting with heatpump class (see PR https://github.com/Akkudoktor-EOS/EOS/pull/78). Maybe we can create issues or creating a Github Project (https://github.com/Akkudoktor-EOS/EOS/projects?query=is%3Aopen) to discuss and coordinate all testing effort?

NormannK, 2024-10-02:
good idea. Today were some great and big PR. I will need some time to read through these and up my github game.
I guess all continue working on an other way to ingest data to avoid Node-Red. At least for devs.

danimo, 2024-10-04:
I think we have a path going forward: Tests, then refacor. Let's proceed with that and open issues along the way.

closed 2024-10-04 by danimo