Akkudoktor-EOS/EOS #58: Pip install error - conflicting ml-dtypes versions

fortrieb, 2024-09-27:
## General System information

- Python version: 3.12.6
- pip version: 24.2
- OS: MacOS 14.7

## Error message

```sh
INFO: pip is looking at multiple versions of tensorflow to determine which version is compatible with other requirements. This could take a while.
ERROR: Cannot install -r requirements.txt (line 21), -r requirements.txt (line 53) and ml-dtypes==0.5.0 because these package versions have conflicting dependencies.

The conflict is caused by:
    The user requested ml-dtypes==0.5.0
    keras 3.5.0 depends on ml-dtypes
    tensorflow 2.17.0 depends on ml-dtypes<0.5.0 and >=0.3.1

To fix this you could try to:
1. loosen the range of package versions you've specified
2. remove package versions to allow pip to attempt to solve the dependency conflict

ERROR: ResolutionImpossible: for help visit https://pip.pypa.io/en/latest/topics/dependency-resolution/#dealing-with-dependency-conflicts
```

## Possible solution

Is `ml-dtypes` relay needed as a strict dependency for _EOS_ python module. I can't find any imports from `ml_dtypes` and if i got the [package description](https://pypi.org/project/ml-dtypes/) right, it is a standalone version of NumPy types. But NumPy is also installed via _requirements.txt_.

Maybe dependency _ml-dtypes_ can be removed?

NormannK, 2024-09-27:
maybe we should start fresh and check all of them. these are all Third-Party Library Imports for python we currently use:
```
numpy
pandas
mariadb
matplotlib
flask
scikit-learn
tensorflow
deap
joblib
requests
```
e.g. pytz is standard lib since python 3.9 (3 years or so). Do we need to specify a version? Imo unless we find a problem with a specific version id ignore it for now.

NormannK, 2024-09-27:
ok i just did a fresh installation on ubuntu in an lxc container.
I had to install  `apt install git python3-virtualenv libmariadb-dev`
after that only the python packages i listed above.
I assume one of the packages in the requirements collide with a requirement of an other package. Therefore am more inclined to only use specific versions for the packages we use and not sub dependencies. 
`pip freeze > requirements.txt`
gave me:
`click==8.1.7
contourpy==1.3.0
cycler==0.12.1
deap==1.4.1
Flask==3.0.3
flatbuffers==24.3.25
fonttools==4.54.1
gast==0.6.0
google-pasta==0.2.0
grpcio==1.66.1
h5py==3.12.1
idna==3.10
itsdangerous==2.2.0
Jinja2==3.1.4
joblib==1.4.2
absl-py==2.1.0
astunparse==1.6.3
blinker==1.8.2
certifi==2024.8.30
charset-normalizer==3.3.2
keras==3.5.0
kiwisolver==1.4.7
libclang==18.1.1
mariadb==1.1.10
Markdown==3.7
markdown-it-py==3.0.0
MarkupSafe==2.1.5
matplotlib==3.9.2
mdurl==0.1.2
ml-dtypes==0.4.1
namex==0.0.8
numpy==1.26.4
opt_einsum==3.4.0
optree==0.12.1
packaging==24.1
pandas==2.2.3
pillow==10.4.0
protobuf==4.25.5
Pygments==2.18.0
pyparsing==3.1.4
python-dateutil==2.9.0.post0
pytz==2024.2
requests==2.32.3
rich==13.8.1
scikit-learn==1.5.2
scipy==1.14.1
six==1.16.0
tensorboard==2.17.1
tensorboard-data-server==0.7.2
tensorflow==2.17.0
tensorflow-io-gcs-filesystem==0.37.1
termcolor==2.4.0
threadpoolctl==3.5.0
typing_extensions==4.12.2
tzdata==2024.2
urllib3==2.2.3
Werkzeug==3.0.4
wrapt==1.16.0`

danimo, 2024-10-01:
The reason to have specific versions is to avoid bugs stemming from people using random versions. It also helps dependabot.

NormannK, 2024-10-01:
yeah i thought about that after i wrote it. also what if we stop maintaining the project and something is changing in the newer versions. 
some hours ago i wrote the pull request and i also included the version. #73

fortrieb, 2024-10-04:
Dependencies are updated by commit https://github.com/Akkudoktor-EOS/EOS/commit/d9f7f3604fc23810aa1764f6d752091920383f4a