Akkudoktor-EOS/EOS #82: README-DE.md line 110 not a date

simcup, 2024-10-03:
moin, ich kann mir vorstellen das in der readme an zeile 110 (Input für den Flask Server ) der 30.07.2024 stehen soll, finde aber auch ein pullrequest für einen typo an nicht kritischer stelle ist overkill. kthybye

drbacke, 2024-10-04:
Gerne fixen, erstelle einen PR und los 👍

sandboxcode, 2024-10-05:
Würde den fix eben in nem Branch publishen und den PR erstellen. Geht aber ohne Contributor-Rolle nicht, oder @drbacke ? Würdest du mir die Rolle eben assignen bzw. das ausreichende Rechteprofil?

drbacke, 2024-10-06:
Müsste geht gehen, das repo ist offen

sandboxcode, 2024-10-06:
Tatsächlich nicht möglich, kann meinen Branch nicht ins origin repo pushen. Nur als fork, aber das wird das origin repo nicht erkennen denke ich. 
[image]

drbacke, 2024-10-06:
Thats the normal way to do this.
Fork -> PR

sandboxcode, 2024-10-06:
Auch das wird leider irgendwie nicht zugelassen.

[image]

Was mache ich falsch? Habe meinen Branch im Fork gepushed, aber Commit History wird vermutlich nicht mitgepullt?

Vielleicht stelle ich mich auch doof an, aber bisher habe ich immer nur in eigenen Repos der Branches gearbeitet und PRs erstellt.

[image]

sandboxcode, 2024-10-06:
Falls jemand aufs gleiche Problem stoßen sollte in der Zukunft: Problem war bei mir einfach über `git rebase upstream/main` zu lösen, da das die commit histories des forks mit dem origin main wieder angleicht.

closed 2024-10-07 by drbacke