Akkudoktor-EOS/EOS #84: Using uv instead of pip to speed up build processes

ss89, 2024-10-03:
Hi all,

I have seen that the `uv` tool (written in rust) is supposedly way faster than pip and has helped the folks over at Home-Assistant to speed up their workflows a lot, thus reduced the time people have been waiting for builds to finish.
Is there any interest in integrating it? I might create a PR then.

fortrieb, 2024-10-04:
Thanks for that hint. Build time isn't an issue at the moment. Adding tests and refactoring the source code are the first thinks needs to be addressed before adding new features.

But we should label this issues and keeping it in backlog for later iterations.

danimo, 2024-10-04:
I'd also like to stay with pip at this point, as it implicitly exists in all environments. I'd like to close this issue for now.

closed 2024-10-04 by danimo