Akkudoktor-EOS/EOS #92: Pre Commit to restrictive

drbacke, 2024-10-04:
Local checks with ruff passed, with pre commit = Fail

closed 2024-10-04 by drbacke

michaelosthege, 2024-10-04:
The merge of #91 included some changes to which pre-commit was not applied. The check was also :x: in that PR, by the way. Under "Branch protection rules" you can require PRs to have passing ✔️ checks before merging:

[image]

[image]

This would help to prevent accidentally merging PRs that break code style or tests.

drbacke, 2024-10-04:
Yes, I know that, but I thought that ruff counted as standard here. And code that conforms to ruff is not accepted. That seems a bit too restrictive to me

michaelosthege, 2024-10-04:
Can you give an example?
Maybe we have OS-dependent effects?
I'm on Windows and the checks pass locally and in CI, so for current `main` I don't see issues.

michaelosthege, 2024-10-05:
On 0th indentation level both `ruff format .` and `pre-commit run --all` format this as 👇 on my machine

```python
assert (
    len(result["Netzbezug_Wh_pro_Stunde"]) == 47
), "The length of 'Netzbezug_Wh_pro_Stunde' should be 48."
```

`ruff check .` only runs the linter and reports `All checks passed!` regardless of the format.

drbacke, 2024-10-05:
I use pre commit on my local system now. It was Not the ruff Check.
Lets Close this.