Akkudoktor-EOS/EOS #95: [BUG]: Flask startet nicht

Digioso, 2024-10-04:
### Describe the issue:

(venv) digioso@akkueos:/opt/EOS$ python ./flask_server.py
Traceback (most recent call last):
  File "/opt/EOS/./flask_server.py", line 19, in <module>
    from modules.class_optimize import isfloat, optimization_problem
  File "/opt/EOS/modules/class_optimize.py", line 8, in <module>
    from modules.class_ems import EnergieManagementSystem, Wechselrichter
ImportError: cannot import name 'Wechselrichter' from 'modules.class_ems' (/opt/EOS/modules/class_ems.py)
(venv) digioso@akkueos:/opt/EOS$

Zudem muss man vorher noch 'pandas' installieren, weil das ebenfalls eine requirement zu sein scheint.

(venv) digioso@akkueos:/opt/EOS$ grep -i pandas flask_server.py
import pandas as pd
(venv) digioso@akkueos:/opt/EOS$

Installation mit: pip install pandas

### Version information:

Aktuelle Version vom GIT.

NormannK, 2024-10-04:
thanks there is already a fix for that in PR #86

Digioso, 2024-10-04:
Vielen Dank!
Habe ich gemacht. Damit geht es jetzt. :)

closed 2024-10-04 by drbacke