Akkudoktor-EOS/EOS #107: Refactor optimization-tests to load input/expectation data from JSON files

michaelosthege, 2024-10-06:
### Description

Currently located in `tests/test_class_optimize.py` there are about 1000 lines of test data (input and expectation) hard-coded into the Python script.

This should be moved to JSON files, for example:
* `tests/testdata/optimize_input_1.json`
* `tests/testdata/optimize_result_1.json`

Then, the test function can be parametrized using pytest:

```python
DIR_TESTDATA = Path(__file__).parent / "testdata"

@pytest.mark.parametrize("fn_in,fn_out", [("optimize_input_1.json", "optimize_output_1.json")])
def test_optimize(fn_in, fn_out):
    # TODO: load input/expectation data
    pass
```

NormannK, 2024-10-06:
I thought about making a hash out of that result and compare it. Would that be a proper alternative? 
The only downside I see would that you cant compare what actually is different.

michaelosthege, 2024-10-06:
No that's not something you'd do with hashes, for the reason you already mentioned.

`np.testing` has a lot of useful assertion functions to compare arrays. They also report the differences.

NormannK, 2024-10-06:
assign that to me. ill to it later today.

NormannK, 2024-10-06:
@michaelosthege can you look at PR #116 please? I can't figure out why the json flies are not being found.

PR #128 (2024-10-09): Run optimization test in 3 seconds, fix other warnings and side effects of tests
* By making `ngen` a kwarg for `optimize()` and `optimierung_ems` the corresponding test can set `ngen=3` and now runs in about 2-3 seconds.
* Input/result data for the optimization test is now provided via JSON files to keep the Python code short.
* A matplotlib warning related to incorrect use of `set_xticklabels` was dealt with
* The `visualization_results.pdf` was removed and is now git-ignored

Closes #107
closed 2024-10-09 by drbacke