Akkudoktor-EOS/EOS #118: Join Hacktoberfest by adding it as a label to the repo

sandboxcode, 2024-10-06:
### Proposed implementation

```python
Add `hacktoberfest` as a label to the repo.

Why does it makes sense? It encourages developers to work actively on participating repositories.
```

fortrieb, 2024-10-06:
Should identify some simple issue that can be labeled with _hacktoberfest_.

drbacke, 2024-10-07:
How about this one ^^

danimo, 2024-10-07:
From experience, Hacktoberfest tags mostly incentivize trophy seekers that care more about badges than quality. This gets to to the point where the Hacktoberfest website now desperately point out that they really are looking for *quality* submissions. Yet, the burden to decide what's quality and what's nonsense (anything from silly simple text reformatting to plain incorrect changes) still single-handedly falls to maintainers. Having made those experiences, I'd vote to stay out of Hacktoberfest for now.

michaelosthege, 2024-10-07:
I agree with @danimo cause we've seen similar things in advance of Google Summer of Code over in the PyMC project.

If participation in Hacktoberfest is a priority for you, we should definitely raise the bar for contributions:
* Requiring more detailed ticket or discussion contributions
* Much stricter requirements for PRs
* Requiring >90 % test coverage for code changes

IMO the project needs contributions from experienced developers until the code structure and architecture stabilizes.

drbacke, 2024-10-10:
No its Not, lets Close that.

closed 2024-10-10 by drbacke