Akkudoktor-EOS/EOS #121: [BUG]: PyPi Integration Bug: flask Server data

drbacke, 2024-10-07:
### Describe the issue:

flask_severy.py      
gesamtlast_simple, database path (/data) not working anymore:
file_path = os.path.join("data", "load_profiles.npz")

### Reproduceable code example:

```python
start flask_server.py
call:
http://127.0.0.1:8503/gesamtlast_simple?year_energy=2000&
```

### Error message:

```shell
TypeError: 'NoneType' object is not subscriptable
```

### Version information:

-

b0661, 2024-10-07:
@drbacke where shall the static data reside?

Shall it become part of the EOS library package or shall it become part of the server package?

drbacke, 2024-10-07:
Its used in the library for load prediction.
The Call is in the server

b0661, 2024-10-07:
I just added a pull request. Moved the data to the server package.

If the data is default data for the EOS library it may better be integrated in the library. In this case one of the classes in the library should provide access to the data. Just my 2c.

For now it is working. 

Using 
``
start flask_server.py; call: http://127.0.0.1:8503/gesamtlast_simple?year_energy=2000&
``
as a test would be fine.

PR #122 (2024-10-07): Adapt data to package directory structure.
Move static data to server package directory.
Assure data/*.npz data is included in the server package.

Closes #121
closed 2024-10-07 by drbacke