Akkudoktor-EOS/EOS #166: Suggestion: Use stronger types instead of int/float

niklasbeinghaus, 2024-10-11:
### Description

I'd love to suggest to use a library that ensures a plausible value conversion between the different units.

Possible candidates:
https://github.com/vikramsamak/PhsiPy-Api
https://github.com/pacosalces/physunits/tree/master/physunits
https://github.com/juhasch/PhysicalQuantities

What's difficult in communicating with each other doesn't need to be complicated in programming. Using strong types reduces confusion and helps understanding which units are used to calculate something. Also, this would allow for easier extension and additional values involved later on...

akokam, 2024-10-13:
I highly recommend pydantic.
It not only helps you expressing types but it also validates at runtime. With mypy etc you can even have statics checks in your pipelines and/or githooks (pre-commit).

NormannK, 2025-01-05:
We have adopted pydantic and mypy. Calculation is done in SI units. Sometimes user interface units can differ bc of convenience.

closed 2025-01-05 by NormannK