Akkudoktor-EOS/EOS #183: [BUG]: class_akku.py min_soc_percent may not be Null

akrauss, 2024-10-27:
### Describe the issue:

I am currently testing the EOS with Home Assistant. Unfortunately I'm quite a noob when it comes to programming, but I can test ;)

In the class_akku.py the default value for min_soc_percent is Null, which leads to an error in the calculation of min_soc_wh.

Default value should be 0 not zero ....

### Reproduceable code example:

```python
src/akkudoktoreosserver/flask_server.py

        # Optional min SoC PV Battery
        if "min_soc_prozent" not in parameter:
            parameter["min_soc_prozent"] = None
```

### Error message:

```shell
eos-1  | Traceback (most recent call last):
eos-1  |   File "/usr/local/lib/python3.12/site-packages/flask/app.py", line 1498, in __call__
eos-1  |     return self.wsgi_app(environ, start_response)
eos-1  |            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
eos-1  |   File "/usr/local/lib/python3.12/site-packages/flask/app.py", line 1476, in wsgi_app
eos-1  |     response = self.handle_exception(e)
eos-1  |                ^^^^^^^^^^^^^^^^^^^^^^^^
eos-1  |   File "/usr/local/lib/python3.12/site-packages/flask/app.py", line 1473, in wsgi_app
eos-1  |     response = self.full_dispatch_request()
eos-1  |                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
eos-1  |   File "/usr/local/lib/python3.12/site-packages/flask/app.py", line 882, in full_dispatch_request
eos-1  |     rv = self.handle_user_exception(e)
eos-1  |          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
eos-1  |   File "/usr/local/lib/python3.12/site-packages/flask/app.py", line 880, in full_dispatch_request
eos-1  |     rv = self.dispatch_request()
eos-1  |          ^^^^^^^^^^^^^^^^^^^^^^^
eos-1  |   File "/usr/local/lib/python3.12/site-packages/flask/app.py", line 865, in dispatch_request
eos-1  |     return self.ensure_sync(self.view_functions[rule.endpoint])(**view_args)  # type: ignore[no-any-return]
eos-1  |            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
eos-1  |   File "/opt/eos/akkudoktoreosserver/flask_server.py", line 249, in flask_optimize
eos-1  |     result = opt_class.optimierung_ems(parameter=parameter, start_hour=datetime.now().hour)
eos-1  |              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
eos-1  |   File "/opt/eos/akkudoktoreos/class_optimize.py", line 392, in optimierung_ems
eos-1  |     akku = PVAkku(
eos-1  |            ^^^^^^^
eos-1  |   File "/opt/eos/akkudoktoreos/class_akku.py", line 31, in __init__
eos-1  |     self.min_soc_wh = (self.min_soc_prozent / 100) * self.kapazitaet_wh
eos-1  |                        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
eos-1  | TypeError: unsupported operand type(s) for /: 'NoneType' and 'int'
```

### Version information:

docker

NormannK, 2024-10-27:
So I do understand it better. What did you try todo? Do you have a use case where you can work without the soc level?
What command did you call?
In any case this should not happen.

akrauss, 2024-10-28:
You provided the fix faster than I could answer your question. Thank you!

NormannK, 2024-11-10:
fixed