Akkudoktor-EOS/EOS #198: [BUG]: Flask_server calling /strompreis fails

NormannK, 2024-11-12:
### Describe the issue:

when accessing http://127.0.0.1:8503/strompreis we get a crash. 

flask_server line 52 is calling the wrong config as a start `config=config.eos.prediction_hours,` should be `config=config`
after that we get crashes in class_strompreis

also there is no test for some endpoints.

### Reproduceable code example:

```python
access http://127.0.0.1:8503/strompreis
```

### Error message:

```shell
File "C:\Users\Norma\AppData\Local\Programs\Python\Python311\Lib\site-packages\flask\app.py", line 1498, in __call__

    return self.wsgi_app(environ, start_response)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

    File "C:\Users\Norma\AppData\Local\Programs\Python\Python311\Lib\site-packages\flask\app.py", line 1476, in wsgi_app

    response = self.handle_exception(e)
               ^^^^^^^^^^^^^^^^^^^^^^^^

    File "C:\Users\Norma\AppData\Local\Programs\Python\Python311\Lib\site-packages\flask\app.py", line 1473, in wsgi_app

    response = self.full_dispatch_request()
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^

    File "C:\Users\Norma\AppData\Local\Programs\Python\Python311\Lib\site-packages\flask\app.py", line 882, in full_dispatch_request

    rv = self.handle_user_exception(e)
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

    File "C:\Users\Norma\AppData\Local\Programs\Python\Python311\Lib\site-packages\flask\app.py", line 880, in full_dispatch_request

    rv = self.dispatch_request()
         ^^^^^^^^^^^^^^^^^^^^^^^

    File "C:\Users\Norma\AppData\Local\Programs\Python\Python311\Lib\site-packages\flask\app.py", line 865, in dispatch_request

    return self.ensure_sync(self.view_functions[rule.endpoint])(**view_args)  # type: ignore[no-any-return]
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

    File "C:\Users\Norma\Desktop\EOS_own\EOS\src\akkudoktoreosserver\flask_server.py", line 65, in flask_strompreis

    specific_date_prices = price_forecast.get_price_for_daterange(
                           

    File "C:\Users\Norma\Desktop\EOS_own\EOS\src\akkudoktoreos\class_strompreis.py", line 124, in get_price_for_daterange

    daily_prices = self.get_price_for_date(date_str)
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

    File "C:\Users\Norma\Desktop\EOS_own\EOS\src\akkudoktoreos\class_strompreis.py", line 91, in get_price_for_date

    last_price_of_previous_day = [
                                 

    IndexError: list index out of range
```

### Version information:

main

NormannK, 2024-11-12:
In the last line 109 in the class_strompreis, `np.array(date_prices) / (1000.0 * 100.0) + self.charges`, the self.charges are added again. We did that already in the list comprehension before.
^this was always the case even before we started the rewriting https://github.com/Akkudoktor-EOS/EOS/commit/9049a067820e4feda21cfcf1d74185a3bc63f356

NormannK, 2024-11-12:
this was my return from the api:
https://api.akkudoktor.net/prices?start=2024-11-12&end=2024-11-14

`{"meta":{"start_timestamp":"2024-11-12","end_timestamp":"2024-11-14","start":"2024-11-12T00:00:00.000Z","end":"2024-11-14T00:00:00.000Z"},"values":[{"start_timestamp":1731366000000,"end_timestamp":1731369600000,"marketprice":110.31,"unit":"Eur/MWh","start":"2024-11-11T23:00:00.000Z","end":"2024-11-12T00:00:00.000Z","marketpriceEurocentPerKWh":11.03},{"start_timestamp":1731369600000,"end_timestamp":1731373200000,"marketprice":104.64,"unit":"Eur/MWh","start":"2024-11-12T00:00:00.000Z","end":"2024-11-12T01:00:00.000Z","marketpriceEurocentPerKWh":10.46},{"start_timestamp":1731373200000,"end_timestamp":1731376800000,"marketprice":102.69,"unit":"Eur/MWh","start":"2024-11-12T01:00:00.000Z","end":"2024-11-12T02:00:00.000Z","marketpriceEurocentPerKWh":10.27},{"start_timestamp":1731376800000,"end_timestamp":1731380400000,"marketprice":103.99,"unit":"Eur/MWh","start":"2024-11-12T02:00:00.000Z","end":"2024-11-12T03:00:00.000Z","marketpriceEurocentPerKWh":10.4},{"start_timestamp":1731380400000,"end_timestamp":1731384000000,"marketprice":107.5,"unit":"Eur/MWh","start":"2024-11-12T03:00:00.000Z","end":"2024-11-12T04:00:00.000Z","marketpriceEurocentPerKWh":10.75},{"start_timestamp":1731384000000,"end_timestamp":1731387600000,"marketprice":118.01,"unit":"Eur/MWh","start":"2024-11-12T04:00:00.000Z","end":"2024-11-12T05:00:00.000Z","marketpriceEurocentPerKWh":11.8},{"start_timestamp":1731387600000,"end_timestamp":1731391200000,"marketprice":147.67,"unit":"Eur/MWh","start":"2024-11-12T05:00:00.000Z","end":"2024-11-12T06:00:00.000Z","marketpriceEurocentPerKWh":14.77},{"start_timestamp":1731391200000,"end_timestamp":1731394800000,"marketprice":185.52,"unit":"Eur/MWh","start":"2024-11-12T06:00:00.000Z","end":"2024-11-12T07:00:00.000Z","marketpriceEurocentPerKWh":18.55},{"start_timestamp":1731394800000,"end_timestamp":1731398400000,"marketprice":201.64,"unit":"Eur/MWh","start":"2024-11-12T07:00:00.000Z","end":"2024-11-12T08:00:00.000Z","marketpriceEurocentPerKWh":20.16},{"start_timestamp":1731398400000,"end_timestamp":1731402000000,"marketprice":223.33,"unit":"Eur/MWh","start":"2024-11-12T08:00:00.000Z","end":"2024-11-12T09:00:00.000Z","marketpriceEurocentPerKWh":22.33},{"start_timestamp":1731402000000,"end_timestamp":1731405600000,"marketprice":190.24,"unit":"Eur/MWh","start":"2024-11-12T09:00:00.000Z","end":"2024-11-12T10:00:00.000Z","marketpriceEurocentPerKWh":19.02},{"start_timestamp":1731405600000,"end_timestamp":1731409200000,"marketprice":171.08,"unit":"Eur/MWh","start":"2024-11-12T10:00:00.000Z","end":"2024-11-12T11:00:00.000Z","marketpriceEurocentPerKWh":17.11},{"start_timestamp":1731409200000,"end_timestamp":1731412800000,"marketprice":161.59,"unit":"Eur/MWh","start":"2024-11-12T11:00:00.000Z","end":"2024-11-12T12:00:00.000Z","marketpriceEurocentPerKWh":16.16},{"start_timestamp":1731412800000,"end_timestamp":1731416400000,"marketprice":156.47,"unit":"Eur/MWh","start":"2024-11-12T12:00:00.000Z","end":"2024-11-12T13:00:00.000Z","marketpriceEurocentPerKWh":15.65},{"start_timestamp":1731416400000,"end_timestamp":1731420000000,"marketprice":174.37,"unit":"Eur/MWh","start":"2024-11-12T13:00:00.000Z","end":"2024-11-12T14:00:00.000Z","marketpriceEurocentPerKWh":17.44},{"start_timestamp":1731420000000,"end_timestamp":1731423600000,"marketprice":218.78,"unit":"Eur/MWh","start":"2024-11-12T14:00:00.000Z","end":"2024-11-12T15:00:00.000Z","marketpriceEurocentPerKWh":21.88},{"start_timestamp":1731423600000,"end_timestamp":1731427200000,"marketprice":272.73,"unit":"Eur/MWh","start":"2024-11-12T15:00:00.000Z","end":"2024-11-12T16:00:00.000Z","marketpriceEurocentPerKWh":27.27},{"start_timestamp":1731427200000,"end_timestamp":1731430800000,"marketprice":317.89,"unit":"Eur/MWh","start":"2024-11-12T16:00:00.000Z","end":"2024-11-12T17:00:00.000Z","marketpriceEurocentPerKWh":31.79},{"start_timestamp":1731430800000,"end_timestamp":1731434400000,"marketprice":225.33,"unit":"Eur/MWh","start":"2024-11-12T17:00:00.000Z","end":"2024-11-12T18:00:00.000Z","marketpriceEurocentPerKWh":22.53},{"start_timestamp":1731434400000,"end_timestamp":1731438000000,"marketprice":171.35,"unit":"Eur/MWh","start":"2024-11-12T18:00:00.000Z","end":"2024-11-12T19:00:00.000Z","marketpriceEurocentPerKWh":17.13},{"start_timestamp":1731438000000,"end_timestamp":1731441600000,"marketprice":149.12,"unit":"Eur/MWh","start":"2024-11-12T19:00:00.000Z","end":"2024-11-12T20:00:00.000Z","marketpriceEurocentPerKWh":14.91},{"start_timestamp":1731441600000,"end_timestamp":1731445200000,"marketprice":131.9,"unit":"Eur/MWh","start":"2024-11-12T20:00:00.000Z","end":"2024-11-12T21:00:00.000Z","marketpriceEurocentPerKWh":13.19},{"start_timestamp":1731445200000,"end_timestamp":1731448800000,"marketprice":125.21,"unit":"Eur/MWh","start":"2024-11-12T21:00:00.000Z","end":"2024-11-12T22:00:00.000Z","marketpriceEurocentPerKWh":12.52},{"start_timestamp":1731448800000,"end_timestamp":1731452400000,"marketprice":115.59,"unit":"Eur/MWh","start":"2024-11-12T22:00:00.000Z","end":"2024-11-12T23:00:00.000Z","marketpriceEurocentPerKWh":11.56},{"start_timestamp":1731452400000,"end_timestamp":1731456000000,"marketprice":117.15,"unit":"Eur/MWh","start":"2024-11-12T23:00:00.000Z","end":"2024-11-13T00:00:00.000Z","marketpriceEurocentPerKWh":11.71},{"start_timestamp":1731456000000,"end_timestamp":1731459600000,"marketprice":111.71,"unit":"Eur/MWh","start":"2024-11-13T00:00:00.000Z","end":"2024-11-13T0 [... cut ...] -11-13T20:00:00.000Z","end":"2024-11-13T21:00:00.000Z","marketpriceEurocentPerKWh":11.49},{"start_timestamp":1731531600000,"end_timestamp":1731535200000,"marketprice":113.51,"unit":"Eur/MWh","start":"2024-11-13T21:00:00.000Z","end":"2024-11-13T22:00:00.000Z","marketpriceEurocentPerKWh":11.35},{"start_timestamp":1731535200000,"end_timestamp":1731538800000,"marketprice":106.62,"unit":"Eur/MWh","start":"2024-11-13T22:00:00.000Z","end":"2024-11-13T23:00:00.000Z","marketpriceEurocentPerKWh":10.66}]}`

redmoon2711, 2024-11-18:
Hi,  in new version is the calulation for the energyprice is wrong.

_np.array(date_prices) / (1000.0 * 100.0) + self.charges_ in line 109
 is better as 
_entry["marketpriceEurocentPerKWh"] + self.charges_.

You can also calulate the energyprice with a function:
       

```
def calc_price(euro_pro_mwh) -> float:
	"""
	Berechne Strompreis in kWh
	Beschaffung: 1.81
	Netznutzung: 9.3
	Steuern, Abgaben und Umlagen: 5.214
	Mehrwertsteuer: 19% = 1.19
	"""
	return (euro_pro_mwh / 10 + 1.81 + 9.3 + 5.214) * 1.19
```

and use the function in:

        previous_day_prices = [
            calc_price(entry["marketprice"])
            for entry in self.prices
            if previous_day_str in entry["end"]
        ]

and also in

        date_prices = [
            calc_price(entry["marketprice"])
            for entry in self.prices
            if date_str in entry["end"]
        ]

Don't forget to remove self.charges in line 109.

NormannK, 2024-11-19:
very good spotted. somehow that did slip in again even that i wrote it in the commit msg.
Do you want to put up a PR yourself?

redmoon2711, 2024-11-19:
thanks, i don't want to put up a PR by myself.  (i don't know how :-))

NormannK, 2024-11-19:
ah that's the easy part if its just a small one file edit :-)
just navigate to the file/code:
https://github.com/Akkudoktor-EOS/EOS/blob/main/src/akkudoktoreos/class_strompreis.py
top right edit, you make the edit and click top right "Commit changes". Just enter a small comment and/or reference this issues (#198) by simply simply writing hashtag and the number.

There are some code style rules but ill take care of that but it would be your PR :-)

NormannK, 2024-11-19:
an no i noticed i never merged the PR for this bug. Thats why the error was still in :-) 
I'll mention you in the PR

closed 2024-12-11 by drbacke