Akkudoktor-EOS/EOS #201: [BUG]: ...ModuleNotFoundError

mazie-78, 2024-11-16:
### Describe the issue:

The server does not start on a Raspi

### Reproduceable code example:

```python
I add this before line 24 in fastapi_server.py: 24 "from akkudoktoreos.class_load import LoadForecast"

"
pathstr= os.path.abspath(".")
path= '/'.join(pathstr.split('/')[:-1])
path= '/'.join(path.split('/')[:-1])
sys.path.append(path)
"
```

### Error message:

```shell
File "/home/xxxxxxx/EOS/src/akkudoktoreos/server/fastapi_server.py", line 25, in <module>
    from akkudoktoreos.class_load import LoadForecast
ModuleNotFoundError: No module named 'akkudoktoreos'
```

### Version information:

Version from 15.11.24
Raspi 5 
Raspi OS

Lasall, 2024-11-16:
How are you trying to start the server? If you start it from source code, use `fastapi run src/akkudoktoreos/server/fastapi_server.py`

mazie-78, 2024-11-16:
python fastapi_server.py

Lasall, 2024-11-16:
Please start it the recommended way

NormannK, 2024-11-16:
anyway im trying to replicate the behavior. I can run the server using `python fastapi_server.py` without any problem. 
Is that Raspi specific?

mazie-78, 2024-11-16:
I think so

Lasall, 2024-11-16:
The error occurs when the package was not installed in the venv (pip install -e .). Also I realised the instructions were not up-to-date.

commit fe9c8ecb (2024-11-16): Update startup instructions. Closes #201 (#202)
closed 2024-11-16 by NormannK