Akkudoktor-EOS/EOS #205: [BUG]: Running optimize with valid data results mostly (but not everytime) in an exception (File "/usr/lib/python3.11/random.py", line 345, in randrange)

RobertHerter, 2024-11-18:
### Describe the issue:

Runnign Optimize with Data below results often in an exeption.

### Reproduceable code example:

```python
<details>
{"ems":{"pv_prognose_wh":[0,0,0,0,0,0,0,0,304.3,1495.2,3179.5,4164.05,4152.25,3389.5,2288.9,890.9000000000001,158.1,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,264.40000000000003,1182.4,2087.25,2290.7,2251.8,1936.2,1530.7,729.4,173.9,0,0,0,0,0,0,0],"strompreis_euro_pro_wh":[0.0002488498,0.00024762410000000003,0.00024223339999999998,0.0002214441,0.0002402223,0.0002593337,0.00028470450000000005,0.0003157754,0.0003160729,0.0003132764,0.00030369690000000004,0.0002816462,0.000265605,0.0002830147,0.0003013288,0.0003137762,0.00032865119999999997,0.000361281,0.0003635658,0.0003530224,0.0003244386,0.0003109678,0.0003102181,0.000300353,0.0002488498,0.00024762410000000003,0.00024223339999999998,0.0002214441,0.0002402223,0.0002593337,0.00028470450000000005,0.0003157754,0.0003160729,0.0003132764,0.00030369690000000004,0.0002816462,0.000265605,0.0002830147,0.0003013288,0.0003137762,0.00032865119999999997,0.000361281,0.0003635658,0.0003530224,0.0003244386,0.0003109678,0.0003102181,0.000300353],"einspeiseverguetung_euro_pro_wh":[0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075],"preis_euro_pro_wh_akku":0,"gesamtlast":[538.4006913709309,515.1285643594947,504.56656326821417,522.6620816954082,506.2200375900648,550.7818662084169,728.85808933053,839.7446021107708,602.144050611358,637.7297072883508,640.9920376286482,611.7952165808247,663.4045753113331,821.0726546635326,1272.616565570683,810.8849866760464,966.5875747164768,766.5416595200188,1302.0893613143114,913.9179484493379,1059.6693994071156,624.0590283573999,541.9028597517477,533.7397228632124,529.6875511696602,487.73427657182987,482.4310144355359,489.13503796444735,480.06154245773587,505.0503731540093,880.9688738483051,835.2617398931166,585.845560334122,663.4970215893358,496.1895681601025,507.5389538284087,777.2617304930155,577.4184466668224,728.7007405603796,764.7915424197927,656.5257970727289,773.4681770350271,928.3942710702735,738.3776700170713,768.8931931680265,739.630279429749,551.186720741159,536.6973156332999]},"pv_akku":{"kapazitaet_wh":21000,"lade_effizienz":0.92,"entlade_effizienz":0.82,"max_ladeleistung_w":9000,"start_soc_prozent":40,"min_soc_prozent":0,"max_soc_prozent":100},"wechselrichter":{"max_leistung_wh":12410},"eauto":{"kapazitaet_wh":1,"lade_effizienz":0.88,"entlade_effizienz":1,"max_ladeleistung_w":1,"start_soc_prozent":1,"min_soc_prozent":0,"max_soc_prozent":100},"temperature_forecast":[5.19,4.99,4.71,4.65,4.85,4.62,4.69,5.77,6.29,8.04,8.91,9.94,10.18,10.28,10.36,10.23,9.08,8.25,7.88,7.46,7.08,6.56,6.07,4.9,4.31,4.46,4.29,4.03,3.69,3.95,4.33,4.58,5.1,6.63,8.01,9.01,9.96,10.78,11.07,10.46,8.52,7.26,6.34,5.57,6.29,6.73,6.62,6.12],"start_solution":[0]}
</details>
```

### Error message:

```shell
<details>
INFO:     192.168.71.12:33786 - "POST /optimize HTTP/1.1" 500 Internal Server Error
ERROR:    Exception in ASGI application
Traceback (most recent call last):
  File "/opt/EOS/.venv/lib/python3.11/site-packages/uvicorn/protocols/http/httptools_impl.py", line 401, in run_asgi
    result = await app(  # type: ignore[func-returns-value]
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/EOS/.venv/lib/python3.11/site-packages/uvicorn/middleware/proxy_headers.py", line 60, in __call__
    return await self.app(scope, receive, send)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/EOS/.venv/lib/python3.11/site-packages/fastapi/applications.py", line 1054, in __call__
    await super().__call__(scope, receive, send)
  File "/opt/EOS/.venv/lib/python3.11/site-packages/starlette/applications.py", line 113, in __call__
    await self.middleware_stack(scope, receive, send)
  File "/opt/EOS/.venv/lib/python3.11/site-packages/starlette/middleware/errors.py", line 187, in __call__
    raise exc
  File "/opt/EOS/.venv/lib/python3.11/site-packages/starlette/middleware/errors.py", line 165, in __call__
    await self.app(scope, receive, _send)
  File "/opt/EOS/.venv/lib/python3.11/site-packages/starlette/middleware/exceptions.py", line 62, in __call__
    await wrap_app_handling_exceptions(self.app, conn)(scope, receive, send)
  File "/opt/EOS/.venv/lib/python3.11/site-packages/starlette/_exception_handler.py", line 62, in wrapped_app
    raise exc
  File "/opt/EOS/.venv/lib/python3.11/site-packages/starlette/_exception_handler.py", line 51, in wrapped_app
    await app(scope, receive, sender)
  File "/opt/EOS/.venv/lib/python3.11/site-packages/starlette/routing.py", line 715, in __call__
    await self.middleware_stack(scope, receive, send)
  File "/opt/EOS/.venv/lib/python3.11/site-packages/starlette/routing.py", line 735, in app
    await route.handle(scope, receive, send)
  File "/opt/EOS/.venv/lib/python3.11/site-packages/starlette/routing.py", line 288, in handle
    await self.app(scope, receive, send)
  File "/opt/EOS/.venv/lib/python3.11/site-packages/starlette/routing.py", line 76, in app
    await wrap_app_handling_exceptions(app, request)(scope, receive, send)
  File "/opt/EOS/.venv/lib/python3.11/site-packages/starlette/_exception_handler.py", line 62, in wrapped_app
    raise exc
  File "/opt/EOS/.venv/lib/python3.11/site-packages/starlette/_exception_handler.py", line 51, in wrapped_app
    await app(scope, receive, sender)
  File "/opt/EOS/.venv/lib/python3.11/site-packages/starlette/routing.py", line 73, in app
    response = await f(request)
               ^^^^^^^^^^^^^^^^
  File "/opt/EOS/.venv/lib/python3.11/site-packages/fastapi/routing.py", line 301, in app
    raw_response = await run_endpoint_function(
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/EOS/.venv/lib/python3.11/site-packages/fastapi/routing.py", line 214, in run_endpoint_function
    return await run_in_threadpool(dependant.call, **values)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/EOS/.venv/lib/python3.11/site-packages/starlette/concurrency.py", line 39, in run_in_threadpool
    return await anyio.to_thread.run_sync(func, *args)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/EOS/.venv/lib/python3.11/site-packages/anyio/to_thread.py", line 56, in run_sync
    return await get_async_backend().run_sync_in_worker_thread(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/EOS/.venv/lib/python3.11/site-packages/anyio/_backends/_asyncio.py", line 2441, in run_sync_in_worker_thread
    return await future
           ^^^^^^^^^^^^
  File "/opt/EOS/.venv/lib/python3.11/site-packages/anyio/_backends/_asyncio.py", line 943, in run
    result = context.run(func, *args)
             ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/EOS/src/akkudoktoreos/server/fastapi_server.py", line 204, in fastapi_optimize
    result = opt_class.optimierung_ems(parameters=parameters, start_hour=start_hour)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/EOS/src/akkudoktoreos/class_optimize.py", line 580, in optimierung_ems
    start_solution, extra_data = self.optimize(parameters.start_solution, ngen=ngen)
                                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/EOS/src/akkudoktoreos/class_optimize.py", line 500, in optimize
    algorithms.eaMuPlusLambda(
  File "/opt/EOS/.venv/lib/python3.11/site-packages/deap/algorithms.py", line 316, in eaMuPlusLambda
    offspring = varOr(population, toolbox, lambda_, cxpb, mutpb)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/EOS/.venv/lib/python3.11/site-packages/deap/algorithms.py", line 234, in varOr
    ind1, ind2 = toolbox.mate(ind1, ind2)
                 ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/EOS/.venv/lib/python3.11/site-packages/deap/tools/crossover.py", line 50, in cxTwoPoint
    cxpoint2 = random.randint(1, size - 1)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/random.py", line 362, in randint
    return self.randrange(a, b+1)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/random.py", line 345, in randrange
    raise ValueError("empty range for randrange() (%d, %d, %d)" % (istart, istop, width))
ValueError: empty range for randrange() (1, 1, 0)
</details>
```

### Version information:

<details>
Current master
</details>

RobertHerter, 2024-11-18:
Found the Problem, removing start_solution from request

NormannK, 2024-11-18:
I'd like to keep it open. yes start_solution is not needed, but there should be a proper exception if the format expected is not meet.
We need some input validation.

commit c7eb3ec6 (2024-11-18): optimize: Enforce start solution with at least two elements Closes #205 (#207)

* mate -> cxTwoPoints needs at least two elements.
closed 2024-11-18 by NormannK