Akkudoktor-EOS/EOS #206: [BUG]: Ergebnisse selbst bei gleichem Input innerhalb der selben Stunde völlig unterschiedlich

RobertHerter, 2024-11-18:
### Describe the issue:

Ich füttere EOS mit Hilfe einer Datenbank, die die entsprechenden Werte liefert.
Leider erhalte ich sehr oft völlig wirre Ergebnisse, trotz absolut identischem Input zur selben Stunde.
Ich kann nicht ausschließen, dass es ein Fehler in meiner Eingabe gibt, allerdings wäre für mich nur Solution 3 eine sinnvolles Ergebnis.

### Reproduceable code example:

```python
#Request 1
{"ems":{"pv_prognose_wh":[0,0,0,0,0,0,0,0,304.3,1495.2,3179.5,4164.05,4167.950000000001,3404.3,2305.6,890.9000000000001,158.1,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,264.40000000000003,1182.4,2087.25,2290.7,2251.8,1936.2,1530.7,729.4,173.9,0,0,0,0,0,0,0],"strompreis_euro_pro_wh":[0.0002488498,0.00024762410000000003,0.00024223339999999998,0.0002214441,0.0002402223,0.0002593337,0.00028470450000000005,0.0003157754,0.0003160729,0.0003132764,0.00030369690000000004,0.0002816462,0.000265605,0.0002830147,0.0003013288,0.0003137762,0.00032865119999999997,0.000361281,0.0003635658,0.0003530224,0.0003244386,0.0003109678,0.0003102181,0.000300353,0.0002488498,0.00024762410000000003,0.00024223339999999998,0.0002214441,0.0002402223,0.0002593337,0.00028470450000000005,0.0003157754,0.0003160729,0.0003132764,0.00030369690000000004,0.0002816462,0.000265605,0.0002830147,0.0003013288,0.0003137762,0.00032865119999999997,0.000361281,0.0003635658,0.0003530224,0.0003244386,0.0003109678,0.0003102181,0.000300353],"einspeiseverguetung_euro_pro_wh":[0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075],"preis_euro_pro_wh_akku":0,"gesamtlast":[538.4006913709309,515.1285643594947,504.56656326821417,522.6620816954082,506.2200375900648,550.7818662084169,728.85808933053,839.7446021107708,602.144050611358,637.7297072883508,640.9920376286482,611.7952165808247,663.4045753113331,821.0726546635326,1272.616565570683,810.8849866760464,966.5875747164768,766.5416595200188,1302.0893613143114,913.9179484493379,1059.6693994071156,624.0590283573999,541.9028597517477,533.7397228632124,529.6875511696602,487.73427657182987,482.4310144355359,489.13503796444735,480.06154245773587,505.0503731540093,880.9688738483051,835.2617398931166,585.845560334122,663.4970215893358,496.1895681601025,507.5389538284087,777.2617304930155,577.4184466668224,728.7007405603796,764.7915424197927,656.5257970727289,773.4681770350271,928.3942710702735,738.3776700170713,768.8931931680265,739.630279429749,551.186720741159,536.6973156332999]},"pv_akku":{"kapazitaet_wh":21000,"lade_effizienz":0.92,"entlade_effizienz":0.82,"max_ladeleistung_w":9000,"start_soc_prozent":48,"min_soc_prozent":0,"max_soc_prozent":100},"wechselrichter":{"max_leistung_wh":12410},"eauto":{"kapazitaet_wh":68000,"lade_effizienz":0.88,"entlade_effizienz":1,"max_ladeleistung_w":11000,"start_soc_prozent":100,"min_soc_prozent":0,"max_soc_prozent":100},"temperature_forecast":[5.19,4.99,4.71,4.65,4.85,4.62,4.69,5.77,6.29,8.04,8.91,9.94,10.57,10.67,10.66,10.51,9.55,8.73,7.88,7.46,7.08,6.56,6.07,4.9,4.31,4.46,4.29,4.03,3.69,3.95,4.33,4.58,5.1,6.63,8.01,9.01,9.96,10.78,11.07,10.46,8.52,7.26,6.34,5.57,6.29,6.73,6.62,6.12]}
#Request 2
{"ems":{"pv_prognose_wh":[0,0,0,0,0,0,0,0,304.3,1495.2,3179.5,4164.05,4167.950000000001,3404.3,2305.6,890.9000000000001,158.1,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,264.40000000000003,1182.4,2087.25,2290.7,2251.8,1936.2,1530.7,729.4,173.9,0,0,0,0,0,0,0],"strompreis_euro_pro_wh":[0.0002488498,0.00024762410000000003,0.00024223339999999998,0.0002214441,0.0002402223,0.0002593337,0.00028470450000000005,0.0003157754,0.0003160729,0.0003132764,0.00030369690000000004,0.0002816462,0.000265605,0.0002830147,0.0003013288,0.0003137762,0.00032865119999999997,0.000361281,0.0003635658,0.0003530224,0.0003244386,0.0003109678,0.0003102181,0.000300353,0.0002488498,0.00024762410000000003,0.00024223339999999998,0.0002214441,0.0002402223,0.0002593337,0.00028470450000000005,0.0003157754,0.0003160729,0.0003132764,0.00030369690000000004,0.0002816462,0.000265605,0.0002830147,0.0003013288,0.0003137762,0.00032865119999999997,0.000361281,0.0003635658,0.0003530224,0.0003244386,0.0003109678,0.0003102181,0.000300353],"einspeiseverguetung_euro_pro_wh":[0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075],"preis_euro_pro_wh_akku":0,"gesamtlast":[538.4006913709309,515.1285643594947,504.56656326821417,522.6620816954082,506.2200375900648,550.7818662084169,728.85808933053,839.7446021107708,602.144050611358,637.7297072883508,640.9920376286482,611.7952165808247,663.4045753113331,821.0726546635326,1272.616565570683,810.8849866760464,966.5875747164768,766.5416595200188,1302.0893613143114,913.9179484493379,1059.6693994071156,624.0590283573999,541.9028597517477,533.7397228632124,529.6875511696602,487.73427657182987,482.4310144355359,489.13503796444735,480.06154245773587,505.0503731540093,880.9688738483051,835.2617398931166,585.845560334122,663.4970215893358,496.1895681601025,507.5389538284087,777.2617304930155,577.4184466668224,728.7007405603796,764.7915424197927,656.5257970727289,773.4681770350271,928.3942710702735,738.3776700170713,768.8931931680265,739.630279429749,551.186720741159,536.6973156332999]},"pv_akku":{"kapazitaet_wh":21000,"lade_effizienz":0.92,"entlade_effizienz":0.82,"max_ladeleistung_w":9000,"start_soc_prozent":48,"min_soc_prozent":0,"max_soc_prozent":100},"wechselrichter":{"max_leistung_wh":12410},"eauto":{"kapazitaet_wh":68000,"lade_effizienz":0.88,"entlade_effizienz":1,"max_ladeleistung_w":11000,"start_soc_prozent":100,"min_soc_prozent":0,"max_soc_prozent":100},"temperature_forecast":[5.19,4.99,4.71,4.65,4.85,4.62,4.69,5.77,6.29,8.04,8.91,9.94,10.57,10.67,10.66,10.51,9.55,8.73,7.88,7.46,7.08,6.56,6.07,4.9,4.31,4.46,4.29,4.03,3.69,3.95,4.33,4.58,5.1,6.63,8.01,9.01,9.96,10.78,11.07,10.46,8.52,7.26,6.34,5.57,6.29,6.73,6.62,6.12]}
#Request 3
{"ems":{"pv_prognose_wh":[0,0,0,0,0,0,0,0,304.3,1495.2,3179.5,4164.05,4167.950000000001,3404.3,2305.6,890.9000000000001,158.1,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,264.40000000000003,1182.4,2087.25,2290.7,2251.8,1936.2,1530.7,729.4,173.9,0,0,0,0,0,0,0],"strompreis_euro_pro_wh":[0.0002488498,0.00024762410000000003,0.00024223339999999998,0.0002214441,0.0002402223,0.0002593337,0.00028470450000000005,0.0003157754,0.0003160729,0.0003132764,0.00030369690000000004,0.0002816462,0.000265605,0.0002830147,0.0003013288,0.0003137762,0.00032865119999999997,0.000361281,0.0003635658,0.0003530224,0.0003244386,0.0003109678,0.0003102181,0.000300353,0.0002488498,0.00024762410000000003,0.00024223339999999998,0.0002214441,0.0002402223,0.0002593337,0.00028470450000000005,0.0003157754,0.0003160729,0.0003132764,0.00030369690000000004,0.0002816462,0.000265605,0.0002830147,0.0003013288,0.0003137762,0.00032865119999999997,0.000361281,0.0003635658,0.0003530224,0.0003244386,0.0003109678,0.0003102181,0.000300353],"einspeiseverguetung_euro_pro_wh":[0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075]," [... cut ...] 1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1],"discharge_array":[1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1],"entlade_effizienz":1,"hours":48,"kapazitaet_wh":68000,"lade_effizienz":0.88,"max_ladeleistung_w":11000,"soc_wh":68000,"start_soc_prozent":100},"start_solution":[4,1,1,4,6,6,5,1,6,1,4,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,0,1,1,1,1,1,1,1,1,1,1,1],"spuelstart":null}
```

### Version information:

Current master

NormannK, 2024-11-18:
If the input is the same every time the only possibility I think of is the optimization. Since its based on the random function sometimes its possible to not find the best solution. That's why we have the starting solution which hints the optimization in the right direction and can improve the result. 
So in a ideal case every time you rerun the simulation you use the last solution data as an input.
I can't find the post right now but I did a statistical analysis on the current optimization settings to see how often we don't get the best ("correct") solution. Ill rerun it again for your input this evening. Maybe you were very unlucky :-)
If you want to test it yourself: run it more often and see if one solutions is found more often than others.

RobertHerter, 2024-11-18:
Hello Normann,
we are not talking about a slightly different solution, the Solution is way off. as you see in my example, Request 1/2/3 are identical. But Solution 1 and 2 is way off a good solution and solution 3 is the only nearly fitting

NormannK, 2024-11-18:
yes i understand. let me run some tests. Ill report back.

NormannK, 2024-11-18:
here a minimal test using your input.
So far I also have different results for every different seed. 
Here a short script to batch calculate results:
[single_test_optimization_Robert.zip](https://github.com/user-attachments/files/17801439/single_test_optimization_Robert.zip)
100 Results so far for my and they are all over the place. 
Something is rotten in the state of Denmark.

NormannK, 2024-11-18:
The intended approach is to run the script every hour, ensuring it always includes an additional hour of data. This minimizes changes between iterations and gradually improves the overall solution.

When running the script with ngen = 2000 (compared to the default of 400, where ngen represents the number of generations in the genetic algorithm), it produces better results overall. However, the outcomes are still not optimal.

One potential improvement could involve running a modified optimization process when no starting solution is provided. This could help refine the initial results. The drawback is that such an approach is computationally intensive, so it might be practical to apply it only to the first iteration.

Another idea that comes to mind while writing this is to run the optimization multiple times until the difference between the starting solution and the final one becomes sufficiently small. This iterative approach might improve convergence, but we need further discussion and testing to validate its feasibility.

Here the results with ngen 2000 after ~500 runs.
Highest number: 3.7498956337232783
Lowest number: 0.0

Number Group (rounded to 1 decimal) | Count
                       3.12927 | 17
                       3.12525 | 15
                        3.1041 | 12
                           0.0 | 11
                       3.15077 | 10
                       3.13352 | 9
                       3.14189 | 5
                       3.05471 | 5
                       3.16462 | 5
                       3.13685 | 5
                       3.01258 | 4

NormannK, 2024-11-18:
[image]

I'm doing 10 Iteration but im not even 100 data sets in. 
The plots shows the distribution of the results for each iteration. First ist not using a starting_solution but later do use them. 
Vertical lines do show the average of all values for that iteration.

Idk which the best solution is. Should it be the one with the value 0? Why 0 why not negative? Is 0 even a useful solution?

NormannK, 2024-11-18:
yes its Gesamtbilanz_euro
[image]

Final data with 750+ data points for each iteration. 0,05 sized buckets. mean value of each iteration are the vertical lines.

Looking at this im not sure what todo. Doing several iterations looks promising, but how many would be best i can't say.  Maybe even more than 10 is beneficial. 
I would move this into a discussion, but if you have any other idea how to improve it or present the data let me know.

Ultimately if would be great to have multiple real datasets of like 1 week for every hour from different people. 
Using this all the parameters could be tuned.

RobertHerter, 2024-11-19:
So if you need Datasets i cloud provide any real Data you need.

Maybe i missunderstood the using: I fill the array on request with hourly Data from 00:00 on.
Using the result-arrays from the current hour on. So the only change in Data is the current SOC Value until a new Day begins.

For me validating the "best" Solution is simple using grafana and compairing the results with lowest Price and Charge/Discharge Results by Hand. Could provide more Feedback on days with an Autarky < 100.

RobertHerter, 2024-11-19:
So no for me there are some new data every hour (especially SOC and Forcast). 
What data do you need in what format, should i send you the json request every hour? I could provide you access to my influxdb (1.8) Database with Data of my System (stored pv and consumption data every 2s) ?
Would it be better to fill the request from 00:00 to current hour with the real data or the "old" forecast values ?

NormannK, 2024-12-07:
oh I had forgotten to answer that. I think at the current time we should concentrate more on the code itself and do the optimization later. Ill when the time has come to do so ill come back to you

drbacke, 2024-12-07:
It becomes stable if you take the previous optimization / result as the starting solution. This is a continuous optimization that is restarted several times per hour and should be used in this way. Without it, you start randomly.
However, the algorithm in deap is unfortunately very poor. In the long run, we should use something more stable here.

closed 2024-12-10 by drbacke

NormannK, 2024-12-15:
@RobertHerter 
pr #242 is improving your problem drastically! We now have better results way faster than before!
One iteration is enough.

RobertHerter, 2024-12-15:
@NormannK  tried to but get an exception even won doc

`Traceback (most recent call last):
  File "/opt/EOS/.venv/lib/python3.11/site-packages/uvicorn/protocols/http/httptools_impl.py", line 409, in run_asgi
    result = await app(  # type: ignore[func-returns-value]
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/EOS/.venv/lib/python3.11/site-packages/uvicorn/middleware/proxy_headers.py", line 60, in __call__
    return await self.app(scope, receive, send)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/EOS/.venv/lib/python3.11/site-packages/fastapi/applications.py", line 1054, in __call__
    await super().__call__(scope, receive, send)
  File "/opt/EOS/.venv/lib/python3.11/site-packages/starlette/applications.py", line 113, in __call__
    await self.middleware_stack(scope, receive, send)
  File "/opt/EOS/.venv/lib/python3.11/site-packages/starlette/middleware/errors.py", line 187, in __call__
    raise exc
  File "/opt/EOS/.venv/lib/python3.11/site-packages/starlette/middleware/errors.py", line 165, in __call__
    await self.app(scope, receive, _send)
  File "/opt/EOS/.venv/lib/python3.11/site-packages/starlette/middleware/exceptions.py", line 62, in __call__
    await wrap_app_handling_exceptions(self.app, conn)(scope, receive, send)
  File "/opt/EOS/.venv/lib/python3.11/site-packages/starlette/_exception_handler.py", line 53, in wrapped_app
    raise exc
  File "/opt/EOS/.venv/lib/python3.11/site-packages/starlette/_exception_handler.py", line 42, in wrapped_app
    await app(scope, receive, sender)
  File "/opt/EOS/.venv/lib/python3.11/site-packages/starlette/routing.py", line 715, in __call__
    await self.middleware_stack(scope, receive, send)
  File "/opt/EOS/.venv/lib/python3.11/site-packages/starlette/routing.py", line 735, in app
    await route.handle(scope, receive, send)
  File "/opt/EOS/.venv/lib/python3.11/site-packages/starlette/routing.py", line 288, in handle
    await self.app(scope, receive, send)
  File "/opt/EOS/.venv/lib/python3.11/site-packages/starlette/routing.py", line 76, in app
    await wrap_app_handling_exceptions(app, request)(scope, receive, send)
  File "/opt/EOS/.venv/lib/python3.11/site-packages/starlette/_exception_handler.py", line 53, in wrapped_app
    raise exc
  File "/opt/EOS/.venv/lib/python3.11/site-packages/starlette/_exception_handler.py", line 42, in wrapped_app
    await app(scope, receive, sender)
  File "/opt/EOS/.venv/lib/python3.11/site-packages/starlette/routing.py", line 73, in app
    response = await f(request)
               ^^^^^^^^^^^^^^^^
  File "/opt/EOS/.venv/lib/python3.11/site-packages/fastapi/routing.py", line 301, in app
    raw_response = await run_endpoint_function(
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/EOS/.venv/lib/python3.11/site-packages/fastapi/routing.py", line 212, in run_endpoint_function
    return await dependant.call(**values)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/EOS/src/akkudoktoreos/server/fastapi_server.py", line 275, in proxy_get
    return await proxy(request, path)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/EOS/src/akkudoktoreos/server/fastapi_server.py", line 298, in proxy
    response = await client.get(url, headers=headers)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/EOS/.venv/lib/python3.11/site-packages/httpx/_client.py", line 1768, in get
    return await self.request(
           ^^^^^^^^^^^^^^^^^^^
  File "/opt/EOS/.venv/lib/python3.11/site-packages/httpx/_client.py", line 1540, in request
    return await self.send(request, auth=auth, follow_redirects=follow_redirects)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/EOS/.venv/lib/python3.11/site-packages/httpx/_client.py", line 1629, in send
    response = await self._send_handling_auth(
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/EOS/.venv/lib/python3.11/site-packages/httpx/_client.py", line 1657, in _send_handling_auth
    response = await self._send_handling_redirects(
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/EOS/.venv/lib/python3.11/site-packages/httpx/_client.py", line 1694, in _send_handling_redirects
    response = await self._send_single_request(request)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/EOS/.venv/lib/python3.11/site-packages/httpx/_client.py", line 1730, in _send_single_request
    response = await transport.handle_async_request(request)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/EOS/.venv/lib/python3.11/site-packages/httpx/_transports/default.py", line 393, in handle_async_request
    with map_httpcore_exceptions():
  File "/usr/lib/python3.11/contextlib.py", line 158, in __exit__
    self.gen.throw(typ, value, traceback)
  File "/opt/EOS/.venv/lib/python3.11/site-packages/httpx/_transports/default.py", line 118, in map_httpcore_exceptions
    raise mapped_exc(message) from exc
httpx.ConnectError: All connection attempts failed`

Lasall, 2024-12-15:
Did you install the new dependencies (is fasthtml installed)? Which endpoint did you access there?

RobertHerter, 2024-12-15:
Ah my fault using url without endpoint, but with optimize i get:
      INFO   192.168.71.12:51862 - "POST /optimize HTTP/1.1" 500
     ERROR   Exception in ASGI application
Traceback (most recent call last):
  File "/opt/EOS/.venv/lib/python3.11/site-packages/uvicorn/protocols/http/httptools_impl.py", line 409, in run_asgi
    result = await app(  # type: ignore[func-returns-value]
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/EOS/.venv/lib/python3.11/site-packages/uvicorn/middleware/proxy_headers.py", line 60, in __call__
    return await self.app(scope, receive, send)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/EOS/.venv/lib/python3.11/site-packages/fastapi/applications.py", line 1054, in __call__
    await super().__call__(scope, receive, send)
  File "/opt/EOS/.venv/lib/python3.11/site-packages/starlette/applications.py", line 113, in __call__
    await self.middleware_stack(scope, receive, send)
  File "/opt/EOS/.venv/lib/python3.11/site-packages/starlette/middleware/errors.py", line 187, in __call__
    raise exc
  File "/opt/EOS/.venv/lib/python3.11/site-packages/starlette/middleware/errors.py", line 165, in __call__
    await self.app(scope, receive, _send)
  File "/opt/EOS/.venv/lib/python3.11/site-packages/starlette/middleware/exceptions.py", line 62, in __call__
    await wrap_app_handling_exceptions(self.app, conn)(scope, receive, send)
  File "/opt/EOS/.venv/lib/python3.11/site-packages/starlette/_exception_handler.py", line 53, in wrapped_app
    raise exc
  File "/opt/EOS/.venv/lib/python3.11/site-packages/starlette/_exception_handler.py", line 42, in wrapped_app
    await app(scope, receive, sender)
  File "/opt/EOS/.venv/lib/python3.11/site-packages/starlette/routing.py", line 715, in __call__
    await self.middleware_stack(scope, receive, send)
  File "/opt/EOS/.venv/lib/python3.11/site-packages/starlette/routing.py", line 735, in app
    await route.handle(scope, receive, send)
  File "/opt/EOS/.venv/lib/python3.11/site-packages/starlette/routing.py", line 288, in handle
    await self.app(scope, receive, send)
  File "/opt/EOS/.venv/lib/python3.11/site-packages/starlette/routing.py", line 76, in app
    await wrap_app_handling_exceptions(app, request)(scope, receive, send)
  File "/opt/EOS/.venv/lib/python3.11/site-packages/starlette/_exception_handler.py", line 53, in wrapped_app
    raise exc
  File "/opt/EOS/.venv/lib/python3.11/site-packages/starlette/_exception_handler.py", line 42, in wrapped_app
    await app(scope, receive, sender)
  File "/opt/EOS/.venv/lib/python3.11/site-packages/starlette/routing.py", line 73, in app
    response = await f(request)
               ^^^^^^^^^^^^^^^^
  File "/opt/EOS/.venv/lib/python3.11/site-packages/fastapi/routing.py", line 301, in app
    raw_response = await run_endpoint_function(
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/EOS/.venv/lib/python3.11/site-packages/fastapi/routing.py", line 214, in run_endpoint_function
    return await run_in_threadpool(dependant.call, **values)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/EOS/.venv/lib/python3.11/site-packages/starlette/concurrency.py", line 39, in run_in_threadpool
    return await anyio.to_thread.run_sync(func, *args)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/EOS/.venv/lib/python3.11/site-packages/anyio/to_thread.py", line 56, in run_sync
    return await get_async_backend().run_sync_in_worker_thread(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/EOS/.venv/lib/python3.11/site-packages/anyio/_backends/_asyncio.py", line 2505, in run_sync_in_worker_thread
    return await future
           ^^^^^^^^^^^^
  File "/opt/EOS/.venv/lib/python3.11/site-packages/anyio/_backends/_asyncio.py", line 1005, in run
    result = context.run(func, *args)
             ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/EOS/src/akkudoktoreos/server/fastapi_server.py", line 238, in fastapi_optimize
    prediction_eos.update_data()
  File "/opt/EOS/src/akkudoktoreos/core/dataabc.py", line 1230, in update_data
    provider.update_data(force_enable=force_enable, force_update=force_update)
  File "/opt/EOS/src/akkudoktoreos/prediction/predictionabc.py", line 207, in update_data
    self._update_data(force_update=force_update)
  File "/opt/EOS/src/akkudoktoreos/prediction/weatherclearoutside.py", line 136, in _update_data
    response = self._request_forecast(force_update=force_update)  # type: ignore
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/EOS/src/akkudoktoreos/utils/cacheutil.py", line 671, in wrapper
    result = func(*args, **kwargs)
             ^^^^^^^^^^^^^^^^^^^^^
  File "/opt/EOS/src/akkudoktoreos/prediction/weatherclearoutside.py", line 94, in _request_forecast
    latitude = round(self.config.latitude, 2)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

Where to add latitude ?

RobertHerter, 2024-12-15:
@Lasall Found the new config with a bunch of new parameters, added lattitude, lonigute but fails at optimization with

`      INFO   192.168.71.12:45526 - "POST /optimize HTTP/1.1" 500
     ERROR   Exception in ASGI application
Traceback (most recent call last):
  File "/opt/EOS/.venv/lib/python3.11/site-packages/uvicorn/protocols/http/httptools_impl.py", line 409, in run_asgi
    result = await app(  # type: ignore[func-returns-value]
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/EOS/.venv/lib/python3.11/site-packages/uvicorn/middleware/proxy_headers.py", line 60, in __call__
    return await self.app(scope, receive, send)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/EOS/.venv/lib/python3.11/site-packages/fastapi/applications.py", line 1054, in __call__
    await super().__call__(scope, receive, send)
  File "/opt/EOS/.venv/lib/python3.11/site-packages/starlette/applications.py", line 113, in __call__
    await self.middleware_stack(scope, receive, send)
  File "/opt/EOS/.venv/lib/python3.11/site-packages/starlette/middleware/errors.py", line 187, in __call__
    raise exc
  File "/opt/EOS/.venv/lib/python3.11/site-packages/starlette/middleware/errors.py", line 165, in __call__
    await self.app(scope, receive, _send)
  File "/opt/EOS/.venv/lib/python3.11/site-packages/starlette/middleware/exceptions.py", line 62, in __call__
    await wrap_app_handling_exceptions(self.app, conn)(scope, receive, send)
  File "/opt/EOS/.venv/lib/python3.11/site-packages/starlette/_exception_handler.py", line 53, in wrapped_app
    raise exc
  File "/opt/EOS/.venv/lib/python3.11/site-packages/starlette/_exception_handler.py", line 42, in wrapped_app
    await app(scope, receive, sender)
  File "/opt/EOS/.venv/lib/python3.11/site-packages/starlette/routing.py", line 715, in __call__
    await self.middleware_stack(scope, receive, send)
  File "/opt/EOS/.venv/lib/python3.11/site-packages/starlette/routing.py", line 735, in app
    await route.handle(scope, receive, send)
  File "/opt/EOS/.venv/lib/python3.11/site-packages/starlette/routing.py", line 288, in handle
    await self.app(scope, receive, send)
  File "/opt/EOS/.venv/lib/python3.11/site-packages/starlette/routing.py", line 76, in app
    await wrap_app_handling_exceptions(app, request)(scope, receive, send)
  File "/opt/EOS/.venv/lib/python3.11/site-packages/starlette/_exception_handler.py", line 53, in wrapped_app
    raise exc
  File "/opt/EOS/.venv/lib/python3.11/site-packages/starlette/_exception_handler.py", line 42, in wrapped_app
    await app(scope, receive, sender)
  File "/opt/EOS/.venv/lib/python3.11/site-packages/starlette/routing.py", line 73, in app
    response = await f(request)
               ^^^^^^^^^^^^^^^^
  File "/opt/EOS/.venv/lib/python3.11/site-packages/fastapi/routing.py", line 301, in app
    raw_response = await run_endpoint_function(
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/EOS/.venv/lib/python3.11/site-packages/fastapi/routing.py", line 214, in run_endpoint_function
    return await run_in_threadpool(dependant.call, **values)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/EOS/.venv/lib/python3.11/site-packages/starlette/concurrency.py", line 39, in run_in_threadpool
    return await anyio.to_thread.run_sync(func, *args)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/EOS/.venv/lib/python3.11/site-packages/anyio/to_thread.py", line 56, in run_sync
    return await get_async_backend().run_sync_in_worker_thread(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/EOS/.venv/lib/python3.11/site-packages/anyio/_backends/_asyncio.py", line 2505, in run_sync_in_worker_thread
    return await future
           ^^^^^^^^^^^^
  File "/opt/EOS/.venv/lib/python3.11/site-packages/anyio/_backends/_asyncio.py", line 1005, in run
    result = context.run(func, *args)
             ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/EOS/src/akkudoktoreos/server/fastapi_server.py", line 241, in fastapi_optimize
    result = opt_class.optimierung_ems(parameters=parameters, start_hour=start_hour)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/EOS/src/akkudoktoreos/optimization/genetic.py", line 506, in optimierung_ems
    start_solution, extra_data = self.optimize(parameters.start_solution, ngen=ngen)
                                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/EOS/src/akkudoktoreos/optimization/genetic.py", line 408, in optimize
    population = self.toolbox.population(n=300)
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/EOS/.venv/lib/python3.11/site-packages/deap/tools/init.py", line 23, in initRepeat
    return container(func() for _ in range(n))
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/EOS/.venv/lib/python3.11/site-packages/deap/tools/init.py", line 23, in <genexpr>
    return container(func() for _ in range(n))
                     ^^^^^^
  File "/opt/EOS/src/akkudoktoreos/optimization/genetic.py", line 205, in create_individual
    individual_components = [
                            ^
  File "/opt/EOS/src/akkudoktoreos/optimization/genetic.py", line 206, in <listcomp>
    self.toolbox.attr_discharge_state() for _ in range(self.config.prediction_hours)
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/random.py", line 362, in randint
    return self.randrange(a, b+1)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/random.py", line 345, in randrange
    raise ValueError("empty range for randrange() (%d, %d, %d)" % (istart, istop, width))
ValueError: empty range for randrange() (0, 0, 0)`

Lasall, 2024-12-15:
The config item optimization_ev_available_charge_rates_percent still has to be set to non-empty list

RobertHerter, 2024-12-16:
Found the Problem with the optimization_ev_available_charge_rates_percent.

But the new Version is way worse than the old one. May be i making something wrong on the new, but let me explain the Problem:

Today i started with an soc of 0% at 00:00, Prices are from 00:00 to 07:00 17.1 ct/kWh (0 ct without Net-Fee), Prices over the Day 07:00 to 22:00 are 25 ct/kWh, Solar-Forecast is bad and not capable of loading the Batterie. So Optimization should load the Batterie anytime from 00:00 to 06:00. The "old" version sets an AC-Load during that time, which looks ok, but the "new" one 
schedule not an AC-Load and does strange things with discharge.

I provide you both requests an results and Grafana Charts to show the Problem:
`Version new (start_hour 00:00)
Request:
{"pv_prognose_wh":[0,0,0,0,0,0,0,0,34.1,283.09999999999997,677.2,876.4000000000001,982,1279.8,1340.5,808,155.79999999999998,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,49.4,656.05,1821.5499999999997,2857.95,3453.85,3397.75,2670.7,1497.55,225.7,0,0,0,0,0,0,0],"strompreis_euro_pro_wh":[0.00017110710000000002,0.000171119,0.0001710595,0.0001710833,0.00017098810000000002,0.0001730825,0.0001801987,0.000241567,0.0002564301,0.00026133289999999995,0.00026079740000000003,0.0002663547,0.00027241179999999997,0.000268223,0.000268223,0.000268223,0.00025906000000000003,0.00026805640000000005,0.0002566919,0.0002554186,0.000244185,0.0002238479,0.00022383600000000002,0.00019074210000000002,0.00022891829166666667,0.00022891829166666667,0.00022891829166666667,0.00022891829166666667,0.00022891829166666667,0.00022891829166666667,0.00022891829166666667,0.00022891829166666667,0.00022891829166666667,0.00022891829166666667,0.00022891829166666667,0.00022891829166666667,0.00022891829166666667,0.00022891829166666667,0.00022891829166666667,0.00022891829166666667,0.00022891829166666667,0.00022891829166666667,0.00022891829166666667,0.00022891829166666667,0.00022891829166666667,0.00022891829166666667,0.00022891829166666667,0.00022891829166666667],"einspeiseverguetung_euro_pro_wh":[0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075],"preis_euro_pro_wh_akku":0.0000777075,"gesamtlast":[512.8613901989584,525.0538795677081,520.4247436905302,519.7860403852872,511.9323595526547,962.4810654826389,1014.9809697928049,617.0592964391396,740.4517354579858,698.0433199850694,841.7152548913194,992.7673612694448,802.8124522329862,955.3170582899306,630.246830901736,688.4267576906251,889.1859661024305,1000.9549201055552,802.0751929475693,684.4860865503472,698.9133949014217,606.9549062770504,572.5358705743054,578.699046015972,498.29172418784725,488.6758173506945,500.4477586493982,527.977783395741,518.0636631930555,874.8042313812496,1042.6946524125256,665.5320389558767,803.9230082068916,672.4621286069975,624.8217563724611,1457.6097921330945,840.6505636788194,944.3708200458329,988.4640774531254,925.4533766163192,912.2666061281254,1038.7351379954741,891.8147506181767,762.3481078657687,620.4458260004418,585.9546274703968,562.6862267313393,558.0905828277776]}
Result:
{"ac_charge":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"dc_charge":[1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1],"discharge_allowed":[1,1,1,0,0,1,1,1,1,1,1,0,1,0,0,0,0,1,1,1,1,1,1,1,1,0,0,0,1,1,1,1,1,1,0,0,0,1,0,0,1,1,1,1,1,1,1,1],"eautocharge_hours_float":null,"result":{"Last_Wh_pro_Stunde":[512.8613901989584,525.0538795677081,520.4247436905302,519.7860403852872,511.9323595526547,962.4810654826389,1014.9809697928049,617.0592964391396,740.4517354579858,698.0433199850694,841.7152548913194,992.7673612694448,802.8124522329862,955.3170582899306,630.246830901736,688.4267576906251,889.1859661024305,1000.9549201055552,802.0751929475693,684.4860865503472,698.9133949014217,606.9549062770504,572.5358705743054,578.699046015972,498.29172418784725,488.6758173506945,500.4477586493982,527.977783395741,518.0636631930555,874.8042313812496,1042.6946524125256,665.5320389558767,803.9230082068916,672.4621286069975,624.8217563724611,1457.6097921330945,840.6505636788194,944.3708200458329,988.4640774531254,925.4533766163192,912.2666061281254,1038.7351379954741,891.8147506181767,762.3481078657687,620.4458260004418,585.9546274703968,562.6862267313393,558.0905828277776],"EAuto_SoC_pro_Stunde":[100,100,100,100,100,100,100,100,100,100,100,100,100,100,100,100,100,100,100,100,100,100,100,100,100,100,100,100,100,100,100,100,100,100,100,100,100,100,100,100,100,100,100,100,100,100,100,100],"Einnahmen_Euro_pro_Stunde":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"Gesamt_Verluste":1428.1080215989614,"Gesamtbilanz_Euro":3.740417434233791,"Gesamteinnahmen_Euro":0,"Gesamtkosten_Euro":3.740417434233791,"Home_appliance_wh_per_hour":[null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null, [... cut ...] 1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1],"discharge_array":[1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1],"entlade_effizienz":1,"hours":48,"kapazitaet_wh":68000,"lade_effizienz":0.88,"max_ladeleistung_w":11000,"soc_wh":68000,"start_soc_prozent":100},"start_solution":[1,4,0,5,2,0,0,1,1,1,1,1,1,0,1,0,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1],"washingstart":null}`

[image]
[image]

RobertHerter, 2024-12-16:
Sorry for the confusion, found that i filled optimization_ev_available_charge_rates_percent wrong.

RobertHerter, 2024-12-20:
So after a while with the new Implementation i have one Question. Running multiple times through the day i fill starting_solution. 
But at 00:00 on a day change, would it be better to fill starting_solution with the data of the previous day request, or to start a new day with no staring_solution ?

NormannK, 2024-12-20:
> So after a while with the new Implementation i have one Question. Running multiple times through the day i fill starting_solution. But at 00:00 on a day change, would it be better to fill starting_solution with the data of the previous day request, or to start a new day with no staring_solution ?

you should always provide the last solution you calculated as long as at least some part of the new one overlaps with the old. 
We also have some Ideas to provide a out of the box starting solution to speed things up, but that will take some time.