Akkudoktor-EOS/EOS #211: [BUG]: akku_soc_pro_stunde looks like 1 hour off

RobertHerter, 2024-11-19:
### Describe the issue:

Showing the Results of an solution in Grafana to visualize teh solution, it look to me that akku_soc_pro_stunde ist one hour off.

In the Picture you could see, that on discharge 0 results the soc is one hour before.

[image]

### Reproduceable code example:

```python
#Request
{"ems":{"pv_prognose_wh":[0,0,0,0,0,0,0,0,214.3,954.2,1892.1000000000001,2958.7,3042.75,2006,1383.9,882.8,194.99999999999997,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,152.6,676.8499999999999,1256.6,1477.6999999999998,1728.0500000000002,1841.1,1697.8,1122.5,302.05,0,0,0,0,0,0,0],"strompreis_euro_pro_wh":[0.00029840140000000003,0.000291666,0.0002847402,0.0002801587,0.0002843832,0.00028848869999999996,0.0003093613,0.00032759210000000006,0.0003555214,0.0003585083,0.0003569494,0.00035284389999999996,0.0003473461,0.0003425266,0.00033777849999999996,0.0003242482,0.00032913910000000003,0.00033782610000000003,0.00033123349999999996,0.0003208567,0.00031034899999999997,0.00030115029999999997,0.0002962475,0.0002779096,0.0002901666,0.00028048,0.00027405400000000003,0.0002680088,0.0002685324,0.0002820151,0.0002989845,0.0003323878,0.0003410153,0.00033770710000000006,0.00031421650000000004,0.00029924629999999997,0.00028673939999999997,0.0002868584,0.0002979016,0.0003266996,0.00033175710000000005,0.0003399799999999999,0.0003406702,0.00033790940000000003,0.0003328162,0.00031742950000000005,0.0003095398,0.000294165],"einspeiseverguetung_euro_pro_wh":[0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075],"preis_euro_pro_wh_akku":0.0000777075,"gesamtlast":[529.6875511696602,487.73427657182987,482.4310144355359,489.13503796444735,480.06154245773587,505.0503731540093,880.9688738483051,835.2617398931166,585.845560334122,663.4970215893358,496.1895681601025,507.5389538284087,777.2617304930155,577.4184466668224,728.7007405603796,764.7915424197927,656.5257970727289,773.4681770350271,928.3942710702735,738.3776700170713,768.8931931680265,739.630279429749,551.186720741159,536.6973156332999,535.0729311326247,516.0131991551251,510.6118506343517,533.3646786598673,503.4867522274335,805.088018509237,702.8692867669787,814.5587399434157,707.3633864736238,666.3565068295518,519.5863102233698,555.1306370668368,853.9651737122313,614.3921425056077,920.5238974553305,982.2751825938461,829.6118460362158,780.6291653462995,870.078935289223,638.0371849929747,648.4093246160255,628.1192136833364,534.610377432871,524.4443023604393]},"pv_akku":{"kapazitaet_wh":21000,"lade_effizienz":0.92,"entlade_effizienz":0.82,"max_ladeleistung_w":9000,"start_soc_prozent":45,"min_soc_prozent":0,"max_soc_prozent":100},"wechselrichter":{"max_leistung_wh":12410},"eauto":{"kapazitaet_wh":68000,"lade_effizienz":0.88,"entlade_effizienz":1,"max_ladeleistung_w":11000,"start_soc_prozent":100,"min_soc_prozent":0,"max_soc_prozent":100},"temperature_forecast":[6.23,6.23,6.22,5.74,5.53,5.49,5.52,5.75,6.33,7.73,8.86,9.66,10.5,10.71,10.81,10.67,9.92,8.3,7.03,6.04,6.25,6.29,6.48,6.71,6.74,6.29,5.86,2.97,3.09,3.16,3.25,1.78,1.68,1.62,1.96,2.64,3.19,3.38,3.31,2.55,1.44,-0.34,-0.93,-1.63,-2.04,-2.28,-1.4,-0.34],"start_solution":[3,6,2,6,5,1,4,2,6,1,1,1,4,1,1,1,1,1,1,1,1,1,1,1,1,0,0,0,0,0,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1]}

#Solution
{"ac_charge":[0.4,1,0.2,1,0.8,0,0.6,0.2,1,0,0,0,0.6,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"dc_charge":[1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1],"discharge_allowed":[0,0,0,0,0,1,0,0,0,1,1,1,0,1,1,1,1,1,1,1,1,1,1,1,1,0,0,0,0,0,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1],"result":{"Last_Wh_pro_Stunde":[577.4184466668224,728.7007405603796,764.7915424197927,656.5257970727289,773.4681770350271,928.3942710702735,738.3776700170713,768.8931931680265,739.630279429749,551.186720741159,536.6973156332999,535.0729311326247,516.0131991551251,510.6118506343517,533.3646786598673,503.4867522274335,805.088018509237,702.8692867669787,814.5587399434157,707.3633864736238,666.3565068295518,519.5863102233698,555.1306370668368,853.9651737122313,614.3921425056077,920.5238974553305,982.2751825938461,829.6118460362158,780.6291653462995,870.078935289223,638.0371849929747,648.4093246160255,628.1192136833364,534.610377432871,524.4443023604393],"EAuto_SoC_pro_Stunde":[100,100,100,100,100,100,100,100,100,100,100,100,100,100,100,100,100,100,100,100,100,100,100,100,100,100,100,100,100,100,100,100,100,100,100],"Einnahmen_Euro_pro_Stunde":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"Gesamt_Verluste":3393.2231535437063,"Gesamtbilanz_Euro":0.8811689327856047,"Gesamteinnahmen_Euro":0,"Gesamtkosten_Euro":0.8811689327856047,"Haushaltsgeraet_wh_pro_stunde":[null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null],"Kosten_Euro_pro_Stunde":[0,0,0,0,0,0,0,0,0,0,0,0,0.1447313820990295,0.13993522011374662,0.14294642749001663,0.13520250594383806,0.22704697804868434,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0.09130641909028947],"Netzbezug_Wh_pro_Stunde":[0,0,0,0,0,0,0,0,0,0,0,0,516.0131991551251,510.6118506343517,533.3646786598673,503.4867522274335,805.088018509237,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,310.39185181884136],"Netzeinspeisung_Wh_pro_Stunde":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"Verluste_Pro_Stunde":[114.28652426665417,52.41594075516957,9.440676606416574,101.31054082084302,169.78569739793284,203.7938643812796,162.08290317447916,168.78143264664004,162.35786621628642,120.99220699196178,117.8116058707244,117.45503366325909,0,0,0,0,0,154.28838002201974,178.8055770607499,121.7773287381126,0.8394794536358496,58.96109518213041,73.80554903465304,69.92678610302153,98.13662859955139,62.18208820357347,11.217985392492295,115.80625888599866,171.35762166138284,190.99293701470765,140.0569430472384,142.33375418400567,137.87982739390316,117.35349748526437,46.98712328961906],"akku_soc_pro_stunde":[51.25854775745964,54.128944513099874,54.6459339463084,51.96576090871996,47.47407579237253,42.08270371879371,37.79479622740538,33.32967896162125,29.034497315687744,25.833645278863358,22.716936128844196,19.60966010600665,19.60966010600665,19.60966010600665,19.60966010600665,19.60966010600665,19.60966010600665,15.527956930820944,10.797650659372538,7.576028205983317,7.621999699872897,10.850821578894324,14.892554026030085,18.721878026909835,24.09602673593288,27.50123632803334,28.115554575717443,25.05189693323071,20.518626518908416,15.465903317461125,11.7606931839363,7.9952499515552,4.347635470235012,1.243045589672462,0]},"eauto_obj":{"charge_array":[1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1],"discharge_array":[1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1],"entlade_effizienz":1,"hours":48,"kapazitaet_wh":68000,"lade_effizienz":0.88,"max_ladeleistung_w":11000,"soc_wh":68000,"start_soc_prozent":100},"start_solution":[3,6,2,6,5,1,4,2,6,1,1,1,4,1,1,1,1,1,1,1,1,1,1,1,1,0,0,0,0,0,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1],"spuelstart":null}
```

### Version information:

current master

drbacke, 2024-12-07:
I can confirm this.

redmoon2711, 2024-12-17:
Hi, I think the calculation of Akku SOC is correct, but shifted one hour into the past.
In the simulation (ems.py), the initial value is set, but later overwritten.

```
# Set initial state
akku_soc_pro_stunde[0] = self.akku.ladezustand_in_prozent()
if self.eauto:
    eauto_soc_pro_stunde[0] = self.eauto.ladezustand_in_prozent()

for stunde in range(start_stunde, ende):
    stunde_since_now = stunde - start_stunde        # At start_stunde: stunde_since_now = 0
    ...

    if self.eauto:
        eauto_soc_pro_stunde[stunde_since_now] = self.eauto.ladezustand_in_prozent() # initial state is overwritten
    ...
    
    # Akku SOC tracking
    akku_soc_pro_stunde[stunde_since_now] = self.akku.ladezustand_in_prozent()    # initial state is overwritten
```

You can fix the issue with:

```
if stunde_since_now + 1 < total_hours:
    akku_soc_pro_stunde[stunde_since_now + 1] = self.akku.ladezustand_in_prozent()
    if self.eauto:
        eauto_soc_pro_stunde[stunde_since_now + 1] = self.eauto.ladezustand_in_prozent()
```

drbacke, 2024-12-22:
> Hi, I think the calculation of Akku SOC is correct, but shifted one hour into the past. In the simulation (ems.py), the initial value is set, but later overwritten.
>
> ```
> # Set initial state [...]

This leads to an error index > than array entries. The fix is to set the battery States at the begining. Fixed

drbacke, 2024-12-22:
Fixed in https://github.com/Akkudoktor-EOS/EOS/pull/276

closed 2024-12-22 by drbacke