Akkudoktor-EOS/EOS #212: [BUG]: ...starting description is with a leading dot and that does not work

briearn, 2024-11-20:
### Describe the issue:

In the URL https://github.com/Akkudoktor-EOS/EOS its stating:

---snip---
Run from source

Install dependencies in virtual environment:

python -m venv .venv
.venv/bin/pip install -r requirements.txt
Finally, start EOS fastapi server:

.venv/bin/fastapi run --port 8503 src/akkudoktoreos/server/fastapi_server.py
Docker

docker compose up --build
Configuration
---snap---

if you try to start it up like that its leading to an error:
---snip---
optimize@optimize:/opt/EOS$ date
Wed Nov 20 12:48:53 PM CET 2024
optimize@optimize:/opt/EOS$ pwd
/opt/EOS
optimize@optimize:/opt/EOS$ source venv/bin/activate
(venv) optimize@optimize:/opt/EOS$ .venv/bin/fastapi run --port 8503 src/akkudoktoreos/server/fastapi_server.py
-bash: .venv/bin/fastapi: No such file or directory
(venv) optimize@optimize:/opt/EOS$ ls -la venv/bin/fastapi 
-rwxr-xr-x 1 optimize optimize 223 Nov 20 09:24 venv/bin/fastapi
(venv) optimize@optimize:/opt/EOS$ . venv/bin/fastapi run --port 8503 src/akkudoktoreos/server/fastapi_server.py
-bash: import: command not found
-bash: import: command not found
-bash: from: command not found
-bash: venv/bin/fastapi: line 7: syntax error near unexpected token `('
-bash: venv/bin/fastapi: line 7: `    sys.argv[0] = re.sub(r'(-script\.pyw|\.exe)?$', '', sys.argv[0])'
(venv) optimize@optimize:/opt/EOS$
---snap---

### Reproduceable code example:

```python
(venv) optimize@optimize:/opt/EOS$ .venv/bin/fastapi run --port 8503 src/akkudoktoreos/server/fastapi_server.py
-bash: .venv/bin/fastapi: No such file or directory
(venv) optimize@optimize:/opt/EOS$ ls -la venv/bin/fastapi 
-rwxr-xr-x 1 optimize optimize 223 Nov 20 09:24 venv/bin/fastapi
(venv) optimize@optimize:/opt/EOS$
```

### Error message:

```shell
-bash: .venv/bin/fastapi: No such file or directory
(venv) optimize@optimize:/opt/EOS$
```

### Version information:

I have a Debian 12 VM running with up to date patchset: Linux optimize 6.1.0-27-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.1.115-1 (2024-11-01) x86_64

EOS Installation was performed with git clone and following the steps from the installation routine of this GIT projects start page:

python -m venv .venv
.venv/bin/pip install -r requirements.txt

Lasall, 2024-11-20:
What python version do you have? `python --version` Try with python3 instead.

briearn, 2024-11-21:
Hi Lasall,

I am using python 3 already:

(venv) optimize@optimize:/opt/EOS$ date
Thu Nov 21 08:01:23 AM CET 2024
(venv) optimize@optimize:/opt/EOS$ python --version
Python 3.11.2
(venv) optimize@optimize:/opt/EOS$

drbacke, 2024-12-10:
If you use Windows it's .venv\Scripts\pip ....