Akkudoktor-EOS/EOS #215: [BUG]: AC Charge Verluste werden in Gesamtbilanz nichtberechnet

celle1234, 2024-11-21:
### Describe the issue:

Mir ist aufgefallen, dass beim AC Charge zwar der Energiegehalt und die Verluste richtig berechnet werde, aber nicht beim Gesamtverbrauch betrachtet und damit berechnet wird.

Hier ein Fix:

```
diff --git a/src/akkudoktoreos/class_ems.py b/src/akkudoktoreos/class_ems.py
index e52204f..fe1ecec 100644
--- a/src/akkudoktoreos/class_ems.py
+++ b/src/akkudoktoreos/class_ems.py
@@ -157,7 +157,9 @@ class EnergieManagementSystem:
                 )
                 # print(stunde, " ", geladene_menge, " ",self.ac_charge_hours[stunde]," ",self.akku.ladezustand_in_prozent())
                 verbrauch += geladene_menge
+                verbrauch += verluste_wh
                 netzbezug += geladene_menge
+                netzbezug += verluste_wh
                 verluste_wh_pro_stunde[stunde_since_now] += verluste_wh

             netzeinspeisung_wh_pro_stunde[stunde_since_now] = netzeinspeisung
```

### Version information:

commit e18201dcb34083c955489856ff0fb25d73d94cd6

celle1234, 2024-12-07:
The file after the reorganisation is src/akkudoktoreos/prediction/ems.py

celle1234, 2024-12-07:
I have noticed that although the energy content and losses are calculated correctly for the AC Charge, the losses is not considered for the total consumption. And so the price is not calculated correctly. So the optimization is also wrong.

drbacke, 2024-12-08:
Please open a PR

celle1234, 2024-12-09:
I did some tests. 
The Scenario is:
The first hours the price is low (10ct/kWh). Then the price rises to 20 ct/kWh. So the alogithim charges the battery in the beginning. The needed power is always 500W and there is no pv.

Here are the results:
```
# charge and discharge effincy 100%
{"ac_charge":[0.4,0.4,1.0,0.6,0.2,0.4,0.4,0.8,0.0,0.0,0.6,0.6,0.8,0.4,0.6,1.0,1.0,1.0,0.4,0.0,0.6,0.8,1.0,0.4,0.8,0.2,0.8,0.8,0.6,1.0,0.6,1.0,0.8,0.0,0.6,0.4,0.8,1.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0],"dc_charge":[1.0,1.0,1.0,1.0,1.0,1.0,1.0,1.0,1.0,1.0,1.0,1.0,1.0,1.0,1.0,1.0,1.0,1.0,1.0,1.0,1.0,1.0,1.0,1.0,1.0,1.0,1.0,1.0,1.0,1.0,1.0,1.0,1.0,1.0,1.0,1.0,1.0,1.0,1.0,1.0,1.0,1.0,1.0,1.0,1.0,1.0,1.0,1.0],"discharge_allowed":[0,0,0,0,0,0,0,0,1,1,0,0,0,0,0,0,0,0,0,1,0,0,0,0,0,0,0,0,0,0,0,0,0,1,0,0,0,0,1,1,1,1,1,1,1,1,1,1],"eautocharge_hours_float":null,"result":{"Last_Wh_pro_Stunde":[1500.0,1500.0,3000.0,1000.0,500.0,500.0,500.0,500.0,500.0,500.0,1500.0,500.0,500.0,500.0,500.0,500.0,500.0,500.0,500.0,500.0,1000.0,500.0,500.0,500.0,500.0,500.0,500.0,500.0,500.0,500.0,500.0,500.0,500.0,500.0,1000.0,500.0,500.0,500.0,500.0,500.0,500.0,500.0,500.0,500.0,500.0,500.0,500.0,500.0],"EAuto_SoC_pro_Stunde":[60.0,60.0,60.0,60.0,60.0,60.0,60.0,60.0,60.0,60.0,60.0,60.0,60.0,60.0,60.0,60.0,60.0,60.0,60.0,60.0,60.0,60.0,60.0,60.0,60.0,60.0,60.0,60.0,60.0,60.0,60.0,60.0,60.0,60.0,60.0,60.0,60.0,60.0,60.0,60.0,60.0,60.0,60.0,60.0,60.0,60.0,60.0,60.0],"Einnahmen_Euro_pro_Stunde":[0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0],"Gesamt_Verluste":0.0,"Gesamtbilanz_Euro":4.1,"Gesamteinnahmen_Euro":0.0,"Gesamtkosten_Euro":4.1,"Home_appliance_wh_per_hour":[null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null],"Kosten_Euro_pro_Stunde":[0.15,0.15,0.3,0.1,0.1,0.1,0.1,0.1,0.0,0.0,0.3,0.1,0.1,0.1,0.1,0.1,0.1,0.1,0.1,0.0,0.2,0.1,0.1,0.1,0.1,0.1,0.1,0.1,0.1,0.1,0.1,0.1,0.1,0.0,0.2,0.1,0.1,0.1,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0],"Netzbezug_Wh_pro_Stunde":[1500.0,1500.0,3000.0,1000.0,500.0,500.0,500.0,500.0,0.0,0.0,1500.0,500.0,500.0,500.0,500.0,500.0,500.0,500.0,500.0,0.0,1000.0,500.0,500.0,500.0,500.0,500.0,500.0,500.0,500.0,500.0,500.0,500.0,500.0,0.0,1000.0,500.0,500.0,500.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0],"Netzeinspeisung_Wh_pro_Stunde":[0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0],"Verluste_Pro_Stunde":[0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0],"akku_soc_pro_stunde":[20.0,40.0,90.0,100.0,100.0,100.0,100.0,100.0,90.0,80.0,100.0,100.0,100.0,100.0,100.0,100.0,100.0,100.0,100.0,90.0,100.0,100.0,100.0,100.0,100.0,100.0,100.0,100.0,100.0,100.0,100.0,100.0,100.0,90.0,100.0,100.0,100.0,100.0,90.0,80.0,70.0,60.0,50.0,40.0,30.0,20.0,10.0,0.0]},"eauto_obj":{"charge_array":[1.0,1.0,1.0,1.0,1.0,1.0,1.0,1.0,1.0,1.0,1.0,1.0,1.0,1.0,1.0,1.0,1.0,1.0,1.0,1.0,1.0,1.0,1.0,1.0,1.0,1.0,1.0,1.0,1.0,1.0,1.0,1.0,1.0,1.0,1.0,1.0,1.0,1.0,1.0,1.0,1.0,1.0,1.0,1.0,1.0,1.0,1.0,1.0],"discharge_array":[1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1],"entlade_effizienz":1.0,"hours":48,"kapazitaet_wh":39000,"lade_effizienz":0.88,"max_ladeleistung_w":11000,"soc_wh":23400.0,"start_soc_prozent":100},"start_solution":[3.0,3.0,6.0,4.0,2.0,3.0,3.0,5.0,1.0,1.0,4.0,4.0,5.0,3.0,4.0,6.0,6.0,6.0,3.0,1.0,4.0,5.0,6.0,3.0,5.0,2.0,5.0,5.0,4.0,6.0,4.0,6.0,5.0,1.0,4.0,3.0,5.0,6.0,1.0,1.0,1.0,1.0,1.0,1.0,1.0,1.0,1.0,1.0],"washingstart":null}
"Gesamtkosten_Euro":4.1
"Gesamt_Verluste":0.0

# charge effincy 80% and discharge effincy 100%
{"ac_charge":[0.2,0.8,1.0,1.0,0.8,0.6,0.0,0.4,0.8,0.8,0.8,1.0,0.6,0.4,1.0,0.8,1.0,0.0,0.2,0.8,0.4,0.4,0.8,0.0,1.0,0.2,0.2,1.0,0.4,1.0,0.6,0.4,1.0,0.6,0.0,0.8,0.2,0.6,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0],"dc_charge":[1.0,1.0,1.0,1.0,1.0,1.0,1.0,1.0,1.0,1.0,1.0,1.0,1.0,1.0,1.0,1.0,1.0,1.0,1.0,1.0,1.0,1.0,1.0,1.0,1.0,1.0,1.0,1.0,1.0,1.0,1.0,1.0,1.0,1.0,1.0,1.0,1.0,1.0,1.0,1.0,1.0,1.0,1.0,1.0,1.0,1.0,1.0,1.0],"discharge_allowed":[0,0,0,0,0,0,1,0,0,0,0,0,0,0,0,0,0,1,0,0,0,0,0,1,0,0,0,0,0,0,0,0,0,0,1,0,0,0,1,1,1,1,1,1,1,1,1,1],"eautocharge_hours_float":null,"result":{"Last_Wh_pro_Stunde":[900.0,2100.0,2500.0,1500.0,500.0,500.0,500.0,1000.0,500.0,500.0,500.0,500.0,500.0,500.0,500.0,500.0,500.0,500.0,900.0,600.0,500.0,500.0,500.0,500.0,1000.0,500.0,500.0,500.0,500.0,500.0,500.0,500.0,500.0,500.0,500.0,1000.0,500.0,500.0,500.0,500.0,500.0,500.0,500.0,500.0,500.0,500.0,500.0,500.0],"EAuto_SoC_pro_Stunde":[60.0,60.0,60.0,60.0,60.0,60.0,60.0,60.0,60.0,60.0,60.0,60.0,60.0,60.0,60.0,60.0,60.0,60.0,60.0,60.0,60.0,60.0,60.0,60.0,60.0,60.0,60.0,60.0,60.0,60.0,60.0,60.0,60.0,60.0,60.0,60.0,60.0,60.0,60.0,60.0,60.0,60.0,60.0,60.0,60.0,60.0,60.0,60.0],"Einnahmen_Euro_pro_Stunde":[0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0],"Gesamt_Verluste":1750.0,"G [... cut ...] start_soc_prozent":100},"start_solution":[5.0,6.0,4.0,2.0,6.0,5.0,5.0,4.0,3.0,2.0,6.0,6.0,5.0,3.0,3.0,5.0,3.0,4.0,4.0,2.0,2.0,2.0,4.0,4.0,2.0,4.0,4.0,4.0,3.0,4.0,4.0,5.0,6.0,6.0,3.0,4.0,4.0,3.0,1.0,1.0,1.0,1.0,1.0,1.0,1.0,1.0,1.0,1.0],"washingstart":null}
"Gesamtkosten_Euro":4.225
"Gesamt_Verluste":1250.0
```

This is the same, as when I calculate it manually.

What do you think?

I think this is one we need to fix first, before the optimization, otherwise the algorithm can not find the optimum.

closed 2024-12-11 by drbacke