Akkudoktor-EOS/EOS #216: [ENH]: Bessere Simulationsergebnisse durch Optimierung der Zustände

celle1234, 2024-11-21:
### Proposed implementation

```python
Die verschiedenen Zustände für Laden, Entladen, ... werden nur als eine Zahl codiert.

The states are:
0: No action ("idle")
1: Discharge ("discharge")
2-6: AC charging with different power levels ("ac_charge")
7-8: DC charging Dissallowed/allowed ("dc_charge")

Wenn ich den Code richtig verstehe werden bei den Permutationen die Werte per Random gefüllt. Dabei wird Logischerweise das AC Charge bevorzugt und man kommt oft aus den lokalen Minima der Kosten nicht mehr raus.

Zum testen habe ich die Zustände bei mir wie folgt geändert:
0-4 idle
5-9 discharge 
10-14 AC Charge 0.2, 0.4, 0.6, 0.8, 1

Das gibt deutlich bessere Simulationsergebnisse, die nicht sinnlos den Akku lädt, obwohl das nur Geld kostet und nichts bringt.

diff --git a/src/akkudoktoreos/class_optimize.py b/src/akkudoktoreos/class_optimize.py
index bc0b3f5..b062bde 100644
--- a/src/akkudoktoreos/class_optimize.py
+++ b/src/akkudoktoreos/class_optimize.py
@@ -208,7 +208,8 @@ class optimization_problem:

         # Create ac_charge array: Only consider values between 2 and 6 (AC charging power levels), set the rest to 0
         ac_charge = np.where(
-            (discharge_hours_bin >= 2) & (discharge_hours_bin <= 6), discharge_hours_bin - 1, 0
+            #(discharge_hours_bin >= 2) & (discharge_hours_bin <= 6), discharge_hours_bin - 1, 0
+            (discharge_hours_bin >= 10) & (discharge_hours_bin <= 14), discharge_hours_bin - 9, 0
         )
         ac_charge = ac_charge / 5.0  # Normalize AC charge to range between 0 and 1

@@ -217,12 +218,14 @@ class optimization_problem:
         if self.optimize_dc_charge:
             dc_charge = np.where(discharge_hours_bin == 8, 1, 0)
         else:
-            dc_charge = np.ones_like(
-                discharge_hours_bin
-            )  # Set DC charge to 0 if optimization is disabled
+            dc_charge = np.where(discharge_hours_bin <= 100, 1, 0)
+            #dc_charge = np.ones_like(
+            #    discharge_hours_bin
+            #)  # Set DC charge to 0 if optimization is disabled

         # Create discharge array: Only consider value 1 (Discharge), set the rest to 0 (binary output)
-        discharge = np.where(discharge_hours_bin == 1, 1, 0)
+        #discharge = np.where(discharge_hours_bin == 1, 1, 0)
+        discharge = np.where(np.logical_and(discharge_hours_bin >= 5, discharge_hours_bin <= 9) , 1, 0)

         return ac_charge, dc_charge, discharge

@@ -252,7 +255,8 @@ class optimization_problem:
         if self.optimize_dc_charge:
             charge_discharge_mutated = np.clip(charge_discharge_mutated, 0, 8)
         else:
-            charge_discharge_mutated = np.clip(charge_discharge_mutated, 0, 6)
+            #charge_discharge_mutated = np.clip(charge_discharge_mutated, 0, 6)
+            charge_discharge_mutated = np.clip(charge_discharge_mutated, 0, 14)

         # Use split_charge_discharge to split the mutated array into AC charge, DC charge, and discharge components
         # ac_charge, dc_charge, discharge = self.split_charge_discharge(charge_discharge_mutated)
@@ -353,7 +357,8 @@ class optimization_problem:
         if self.optimize_dc_charge:
             self.toolbox.register("attr_discharge_state", random.randint, 0, 8)
         else:
-            self.toolbox.register("attr_discharge_state", random.randint, 0, 6)
+            #self.toolbox.register("attr_discharge_state", random.randint, 0, 6)
+            self.toolbox.register("attr_discharge_state", random.randint, 0, 14)

         if self.optimize_ev:
             self.toolbox.register(
@@ -379,7 +384,8 @@ class optimization_problem:
             )
         else:
             self.toolbox.register(
-                "mutate_charge_discharge", tools.mutUniformInt, low=0, up=6, indpb=0.2
+                #"mutate_charge_discharge", tools.mutUniformInt, low=0, up=6, indpb=0.2
+                "mutate_charge_discharge", tools.mutUniformInt, low=0, up=14, indpb=0.2
             )
         # - Float mutation for EV charging values
         self.toolbox.register(
@@ -427,7 +433,10 @@ class optimization_problem:
             ems.set_ev_charge_hours(eautocharge_hours_float)
         else:
             ems.set_ev_charge_hours(np.full(self.prediction_hours, 0))
-        return ems.simuliere(start_hour)
+        output = ems.simuliere(start_hour)
+        #with open("/tmp/output.txt", "a") as f:
+        #    print(output, file=f)
+        return(output)

     def evaluate(
         self,
@@ -447,11 +456,18 @@ class optimization_problem:

         discharge_hours_bin, eautocharge_hours_float, _ = self.split_individual(individual)

+        #f = open("/tmp/output2.txt", "a")
+        #print(discharge_hours_bin, file=f)
+        #print(gesamtbilanz, end=' - ', file=f)
         # Small Penalty for not discharging
-        gesamtbilanz += sum(
-            0.01 for i in range(self.prediction_hours) if discharge_hours_bin[i] == 0.0
-        )
-
+        #gesamtbilanz += sum(
+        #    0.01 for i in range(self.prediction_hours) if discharge_hours_bin[i] == 0.0
+        #)
+
+        #gesamtbilanz += sum(
+        #    0.10 for i in range(self.prediction_hours) if discharge_hours_bin[i] > 1
+        #)
+
         # Penalty for not meeting the minimum SOC (State of Charge) requirement
         # if parameters.eauto_min_soc_prozent - ems.eauto.ladezustand_in_prozent() <= 0.0 and  self.optimize_ev:
         #     gesamtbilanz += sum(
@@ -476,6 +492,7 @@ class optimization_problem:
                 (parameters.eauto.min_soc_prozent - ems.eauto.ladezustand_in_prozent())
                 * self.strafe,
             )
+        #print(gesamtbilanz, file=f)

         return (gesamtbilanz,)

@@ -577,7 +594,10 @@ class optimization_problem:
             "evaluate",
             lambda ind: self.evaluate(ind, ems, parameters, start_hour, worst_case),
         # Extract the price from 00:00 of the previous day
-        last_price_of_previous_day = [                 -            entry["marketpriceEurocentPerKWh"] + self.charges
-            for entry in self.prices
-            if previous_day_str in entry["end"]
-        ][-1]
+        #last_price_of_previous_day = [                +        #    entry["marketpriceEurocentPerKWh"] + self.charges                                                 +        #    for entry in self.prices
+        #    if previous_day_str in entry["end"]
+        #][-1]

         # Extract all prices for the specified date             date_prices = [
```
```

drbacke, 2024-12-07:
Please in english and sorry but could you explain your Changes more detailled. 
If I have understood you correctly, do you want equal probabilities for all states?
However, you now have significantly more states than before. Where does the statement that this works better come from? 
If so, it would make more sense to increase the probabilities of the states in the mutation.

celle1234, 2024-12-07:
Yes the problem for me is, that as far as I understand the code the mutations are equally weighted randomized. The problem with this is, that the possibility to get some kind of charge state is much higher, than the other states. 
In tests with my small battery (5kw) I often see, that EOS shows, that I should charge the battery in many scenarios, where it does not make any sense. 
For testing I added some more logging. It seems, that the program finds local minimas of costs but it did not find the global minima. 
I added the additional states to get a better distribution of the logical states (do nothing, allow discharge and charge the battery (with different levels)). This works much better for me. If the distribution can be changed in another way, than this should also solve the problem.

drbacke, 2024-12-07:
Thanks for the explanation. Could you make a short Test and share it with us? Sounds interesting for me. But maybe the Change in propab would be more efficient

celle1234, 2024-12-07:
Yes there is a bug with ac charge price calculation. The ac charge efficiency is not calculated correctly. I opened this as  [https://github.com/Akkudoktor-EOS/EOS/issues/215](https://github.com/Akkudoktor-EOS/EOS/issues/215)

drbacke, 2024-12-08:
> Yes there is a bug with ac charge price calculation. The ac charge efficiency is not calculated correctly. I opened this as  [https://github.com/Akkudoktor-EOS/EOS/issues/215](https://github.com/Akkudoktor-EOS/EOS/issues/215)

Please be carefull here. I'm quite sure that the loses are implemented somewhere else. Please do Not Change that without Testing.
@NormannK shall we implement that Changes? Would be nice If this is the only Problem.

celle1234, 2024-12-09:
For the other problem I opened the pull request [https://github.com/Akkudoktor-EOS/EOS/pull/226](https://github.com/Akkudoktor-EOS/EOS/pull/226)

Back to the optimization problem. First I searched the bug, why the optimization are almost every time wrong for small battery capacity. First I find the bug with AC charge efficiency. After I fixed this the results were a little bit better, but most of the time the code never finds the optimum and the result was very depended on the start solution. So I searched further and I found that the mutations seemed to be imbalanced. I do not know how to change the probability of the mutations, so i increased the number of states. But it should also work to change the probability of the states in the mutations.

drbacke, 2024-12-11:
@celle1234 
I approved the PR. You can go ahead with the ac charge states :-) 
I think it's ok to solve this throug more AC charge states. We can switch to probab. later on.

redmoon2711, 2024-12-12:
I took the idea from Celle and tested and adapted it. The levels can now be set via configuration:
```
self.state_levels = {
    'idle': (0, 4),
    'discharge': (5, 9),
    'ac': (10, 14),
    'dc': (15, 19)
}
```

The results with optimize_input_1.json are:
start_hour=0, ngen=300 after 3 iterations:

main:
Total balance_Euro 0.07506518757851482
Total losses 5518.667417355373

change:
Total balance_Euro -0.20943269628099148
Total losses 5671.084338842975

The goal is reached much faster with more states. ac_charge is no longer as dominant as before.

Lasall, 2024-12-12:
Could you make a Pull-Request for your commit with the state changes please? (For the other commit you can make a separate PR)

NormannK, 2024-12-12:
very nice. when there is PR ill make detailed statistics and compare the quality

celle1234, 2024-12-13:
Am I correct that redmoon2711 will create the PR for this issue?

Lasall, 2024-12-13:
Anybody really, this commit https://github.com/redmoon2711/EOS/commit/340ad9da53ad3784f368bf8b95afaa0030e9057c would be a nice start to look at.

drbacke, 2024-12-15:
Thank you for the hint. Issues is fixed now

closed 2024-12-15 by drbacke