Akkudoktor-EOS/EOS #249: [BUG]: After pr #242 discharge is often 0 without result.Netzbezug_Wh_pro_Stunde >0

RobertHerter, 2024-12-16:
### Describe the issue:

Starting with pr#242 discharge_allowed[x] is often 0 without sense and
result.Netzbezug_Wh_pro_Stunde ist 0 at the same time.

### Version information:

pr#242

[image]

RobertHerter, 2024-12-16:
Request:
`{"ems":{"pv_prognose_wh":[0,0,0,0,0,0,0,0,88.67962884423201,329.5603840174533,762.6,1335.4,1353.1999999999998,1314,1340.5,808,155.79999999999998,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,49.4,656.05,1821.5499999999997,2857.95,3453.85,3397.75,2670.7,1497.55,225.7,0,0,0,0,0,0,0],"strompreis_euro_pro_wh":[0.00017110710000000002,0.000171119,0.0001710595,0.0001710833,0.00017098810000000002,0.0001730825,0.0001801987,0.000241567,0.0002564301,0.00026133289999999995,0.00026079740000000003,0.0002663547,0.00027241179999999997,0.000268223,0.000268223,0.000268223,0.00025906000000000003,0.00026805640000000005,0.0002566919,0.0002554186,0.000244185,0.0002238479,0.00022383600000000002,0.00019074210000000002,0.00022891829166666667,0.00022891829166666667,0.00022891829166666667,0.00022891829166666667,0.00022891829166666667,0.00022891829166666667,0.00022891829166666667,0.00022891829166666667,0.00022891829166666667,0.00022891829166666667,0.00022891829166666667,0.00022891829166666667,0.00022891829166666667,0.00022891829166666667,0.00022891829166666667,0.00022891829166666667,0.00022891829166666667,0.00022891829166666667,0.00022891829166666667,0.00022891829166666667,0.00022891829166666667,0.00022891829166666667,0.00022891829166666667,0.00022891829166666667],"einspeiseverguetung_euro_pro_wh":[0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075],"preis_euro_pro_wh_akku":0.0000777075,"gesamtlast":[717.7727321,717.9599999999999,717.9599999999999,717.9599999999999,717.9599999999999,717.9599999999999,717.9599999999999,717.9599999999999,717.9599999999999,0.18726790000000001,841.7152548913194,992.7673612694448,802.8124522329862,955.3170582899306,630.246830901736,688.4267576906251,889.1859661024305,1000.9549201055552,802.0751929475693,684.4860865503472,698.9133949014217,606.9549062770504,572.5358705743054,578.699046015972,498.29172418784725,488.6758173506945,500.4477586493982,527.977783395741,518.0636631930555,874.8042313812496,1042.6946524125256,665.5320389558767,803.9230082068916,672.4621286069975,624.8217563724611,1457.6097921330945,840.6505636788194,944.3708200458329,988.4640774531254,925.4533766163192,912.2666061281254,1038.7351379954741,891.8147506181767,762.3481078657687,620.4458260004418,585.9546274703968,562.6862267313393,558.0905828277776]},"pv_akku":{"kapazitaet_wh":21000,"lade_effizienz":0.98,"entlade_effizienz":0.85,"max_ladeleistung_w":9140,"start_soc_prozent":82,"min_soc_prozent":0,"max_soc_prozent":100},"wechselrichter":{"max_leistung_wh":12410},"eauto":{"kapazitaet_wh":68000,"lade_effizienz":0.88,"entlade_effizienz":1,"max_ladeleistung_w":11000,"start_soc_prozent":100,"min_soc_prozent":0,"max_soc_prozent":100},"temperature_forecast":[4.66,5.12,4.28,4.04,4.98,5.06,6.03,7.13,7.65,8.15,8.98,9.52,9.5,9.92,10.18,9.58,7.89,7.04,6.59,6.13,6.03,5.89,5.26,4.78,4.67,4.79,5.14,5.3,5.64,6.21,6.36,5.45,5.75,6.42,7.2,8.24,9.18,9.76,9.67,8.86,6.82,5.72,5.11,4.89,4.82,4.84,4.53,4.51],"start_solution":[27,32,3,18,25,25,28,3,12,5,17,19,3,20,3,2,15,11,11,16,13,20,20,14,14,18,14,17,12,14,17,19,19,13,5,22,9,8,18,21,13,11,19,18,20,20,20,16]}`

Result:

`{"ac_charge":[0,0.6,0,0.1,0,0,0,0.1,1,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"dc_charge":[1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1],"discharge_allowed":[0,0,1,0,1,1,1,0,0,0,1,1,1,0,0,0,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,0,1,1,1,0,1,1,1,1,1,1,1,1],"eautocharge_hours_float":null,"result":{"Last_Wh_pro_Stunde":[841.7152548913194,992.7673612694448,802.8124522329862,955.3170582899306,630.246830901736,688.4267576906251,889.1859661024305,1000.9549201055552,802.0751929475693,684.4860865503472,698.9133949014217,606.9549062770504,572.5358705743054,578.699046015972,498.29172418784725,488.6758173506945,500.4477586493982,527.977783395741,518.0636631930555,874.8042313812496,1042.6946524125256,665.5320389558767,803.9230082068916,672.4621286069975,624.8217563724611,1457.6097921330945,840.6505636788194,944.3708200458329,988.4640774531254,925.4533766163192,912.2666061281254,1038.7351379954741,891.8147506181767,762.3481078657687,620.4458260004418,585.9546274703968,562.6862267313393,558.0905828277776],"EAuto_SoC_pro_Stunde":[100,100,100,100,100,100,100,100,100,100,100,100,100,100,100,100,100,100,100,100,100,100,100,100,100,100,100,100,100,100,100,100,100,100,100,100,100,100],"Einnahmen_Euro_pro_Stunde":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"Gesamt_Verluste":3301.9640613622682,"Gesamtbilanz_Euro":0,"Gesamteinnahmen_Euro":0,"Gesamtkosten_Euro":0,"Home_appliance_wh_per_hour":[null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null],"Kosten_Euro_pro_Stunde":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"Netzbezug_Wh_pro_Stunde":[0,0,0,0,0,0,0,0,0,0,0,0, [... cut ...] 1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1],"discharge_array":[1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1],"entlade_effizienz":1,"hours":48,"kapazitaet_wh":68000,"lade_effizienz":0.88,"max_ladeleistung_w":11000,"soc_wh":68000,"start_soc_prozent":100},"start_solution":[22,28,11,23,18,12,18,23,32,2,18,11,17,10,7,4,13,18,19,20,19,16,13,21,17,11,15,12,15,13,20,12,19,15,18,6,20,14,15,2,17,12,16,20,11,18,15,13],"washingstart":null}`

drbacke, 2024-12-16:
Sure the PV Power is Not priorized, because of battery losses?

drbacke, 2024-12-17:
I dont think this is a Bug, it's a little more complicated. Please Post the PDF (visualize) or Something more understandable

RobertHerter, 2024-12-18:
@drbacke Ich habe die json Eingabe und Ausgabe beigefügt. ich versuche mal in Deutsch das Problem zu beschreiben.

Um 13:00 geht discharged_allowed auf 0 und bleibt so bis 16:00. Zu diesem Zeitpunkt reicht die geschätzte solare Produktion um die Lastschätzung zu bedienen, der Preis ist auch relativ hoch, Im Result ist zu diesen Zeiten der berechnete Netzbezug_Wh_pro_Stunde 0, so das ein discharged_allowed=0 keinen Sinn macht, denn es wirkt sich ja weder auf akku_soc_pro_stunde noch auf kosten_euro_pro_stundeaus.

Allerdings wirkt sich ein Sperren der Entladung des Akkus schon im System aus, da dann Lastspitzen oder Einbrüche der solaren Erzeugung nicht aus dem Akku genommen werden, sondern auf den Bezug gehen. Selbes Problem ist um 11:00 und 15:00 des Folgetages.

Gelöst habe ich es, das ich in der Steuerung meines Systems das Entladen des Akkus nicht sperre, wenn discharged_allowed = 0 UND Netzbezug_Wh_pro_Stunde = 0

drbacke, 2024-12-18:
@RobertHerter Ich machs mal kurz auf Deutsch: Ja verstehe das ist in der Tat ein Problem. Es fehlt noch folgendes https://github.com/Akkudoktor-EOS/EOS/issues/14 damit wäre das dann alles korrekt. Also genau diese Lastspitzen kennt das System aktuell nicht. Das möchte ich aber noch einbauen, wir haben das bereits für unser PV Tool getan. Ist aber nicht ganz unkompliziert und wird noch ein wenig dauern. 

@redmoon2711 This is a quick hack to avoid bigger problems due to https://github.com/Akkudoktor-EOS/EOS/issues/14

RobertHerter, 2024-12-18:
@drbacke I think that this would not be a solution, neither forecast (solar nor consumption) would be so exact. One small Cloud or one Cup of Coffee would result in grid-usage and not battery-usage. In my Opinion it is essential to priorize discharging if forcast says either one wold be cost-neutral.

drbacke, 2024-12-18:
> @drbacke I think that this would not be a solution, neither forecast (solar nor consumption) would be so exact. One small Cloud or one Cup of Coffee would result in grid-usage and not battery-usage. [...]

Yes we already do this (you can rise the penalty 0.01 to a higher value, this is the priorisation ->  @redmoon2711 post) , but which penalty do you use for this? For Issues https://github.com/Akkudoktor-EOS/EOS/issues/14 we will model the statistic load distribution, so we get a more detailed view of short peak loads. I think this should help. 

But another point maybe the phyical simulation. So maybe (I'm not sure) the Battery losses are added when discharge is 1 and the battery is empty. So the energy has to pass the battery in every case. I can look at this later or maybe someone else has earlier time for this

redmoon2711, 2024-12-18:
@drbacke in my tests are no battery losses, when discharge is 1 and no energy go out. 
But i think the same problem could ocour if ac_charge is 1 and no energy is charged. 

i have try a other solution - flip the state to discharge, when the costs are equal (at the end of optimize).
The small penalty is not nessesary.   
 
```
# Perform final evaluation on the best solution
o = self.evaluate_inner(start_solution, ems, start_hour)

# Force discharge if costs do not increase
for i in range(start_hour, self.prediction_hours):
    
    # check state if idle or ac_charge
    if start_solution[i] < self.len_ac or 2 * self.len_ac <= start_solution[i] < 3 * self.len_ac:

        # try switch to discharge
        old_state = start_solution[i]
        start_solution[i] = self.len_ac
        out = self.evaluate_inner(start_solution, ems, start_hour)

        if out["Gesamtbilanz_Euro"] > o["Gesamtbilanz_Euro"]:
            start_solution[i] = old_state
        else:
            o = out
```
The results are: [visualization_results.pdf](https://github.com/user-attachments/files/18188251/visualization_results.pdf)

what is your opinion on this?

drbacke, 2024-12-19:
@RobertHerter Could you try out this PR: https://github.com/Akkudoktor-EOS/EOS/pull/268
I've added (first try) a constant small self consumption rate. This should fix this issues. In future we need to calculate this rate by my probability load table. But this should help now. 
@redmoon2711 No this is not a good idea, because you should solve the problem at its roots and this is the lack of a self consumption rate due to high peak loads / pv generation << 1h.

RobertHerter, 2024-12-20:
@drbacke Looks better now, but have to test a few days to check

drbacke, 2024-12-20:
> @drbacke Looks better now, but have to test a few days to check
Thanks for Testing😊

RobertHerter, 2024-12-20:
So didn't take long to get the Problem again:

13:00 to 15:00 Next Day: Discharge is going to 0 again when solar is enough to serve charge and consumption.
Request [20.12.2024, 08:15:00] 
```
{"ems":{"pv_prognose_wh":[0,0,0,0,0,0,0,0,15.5,193.9,420.9,579.2,664.1,622.7,605.5,352.70000000000005,63,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,16,237,646.05,977.9000000000001,1315.5,1342.3,1031.15,380.6,60.5,0,0,0,0,0,0,0],"strompreis_euro_pro_wh":[0.0002277868,0.00020846120000000002,0.00019040890000000002,0.00019174170000000001,0.00019815580000000002,0.00022870310000000003,0.0002497185,0.00029285600000000003,0.0003015668,0.0002979135,0.0002937366,0.00028695360000000004,0.0002751964,0.0002707696,0.0002757557,0.0002884173,0.00029623560000000004,0.0003017691,0.0002914042,0.0002883935,0.00027777870000000003,0.00025646580000000005,0.0002547046,0.00024032940000000004,0.00026188426666666666,0.00026188426666666666,0.00026188426666666666,0.00026188426666666666,0.00026188426666666666,0.00026188426666666666,0.00026188426666666666,0.00026188426666666666,0.00026188426666666666,0.00026188426666666666,0.00026188426666666666,0.00026188426666666666,0.00026188426666666666,0.00026188426666666666,0.00026188426666666666,0.00026188426666666666,0.00026188426666666666,0.00026188426666666666,0.00026188426666666666,0.00026188426666666666,0.00026188426666666666,0.00026188426666666666,0.00026188426666666666,0.00026188426666666666],"einspeiseverguetung_euro_pro_wh":[0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075,0.0000777075],"preis_euro_pro_wh_akku":0.0000777075,"gesamtlast":[510.9488749840559,483.83288667510953,492.19992404014033,497.0596584673614,511.69556811354175,729.4906933388891,809.5756074533343,650.0450166504851,659.6965636725695,725.4698199725692,681.4450092593752,966.4896076611103,679.8590361843625,1047.6947257027189,987.9949799952107,934.6399933951386,839.4373879152778,1098.2586248118055,755.6356805954856,810.8847254263887,653.6064676145832,645.3896876156249,616.1744979826389,623.2950833534721,594.8787128003471,600.0998508152779,594.4037516944442,590.4052838149306,594.8801742871531,863.2662758555559,858.7942874329864,649.5943344456377,612.2377633769318,1024.1865312652778,628.3315629597221,700.7526648690973,761.1067034253471,822.5515592006943,1059.0963958128473,780.8814741843346,936.695584689624,1036.795930965278,1010.9066945565975,862.4403998312504,645.3558516027779,625.2972394225695,629.7641745263887,618.7367612607641]},"pv_akku":{"kapazitaet_wh":21000,"lade_effizienz":0.98,"entlade_effizienz":0.85,"max_ladeleistung_w":9300,"start_soc_prozent":96,"min_soc_prozent":0,"max_soc_prozent":100},"wechselrichter":{"max_leistung_wh":12410},"eauto":{"kapazitaet_wh":68000,"lade_effizienz":0.88,"entlade_effizienz":1,"max_ladeleistung_w":11000,"start_soc_prozent":100,"min_soc_prozent":0,"max_soc_prozent":100},"temperature_forecast":[3.36,3.22,2.24,1.83,2.17,2.12,2.07,1.65,1.54,1.48,1.4,1.42,1.7,2.3,2.62,2.27,1.47,1.2,0.51,-0.37,-0.18,-1.3,-1.2,-1.48,-2.08,-2.66,-2.97,-3,-3.07,-2.65,-2.67,-2.28,-2.17,-1.83,-0.02,1.92,3.33,4.3,4.64,3.91,2.71,2.25,1.5,0.29,0.36,0.25,0,0.14],"start_solution":[14,0,12,5,3,15,8,8,11,13,10,11,12,7,8,9,12,10,12,12,10,1,2,5,9,7,13,11,11,13,9,7,13,10,7,12,6,6,11,9,9,9,9,7,3,7,14,8]}
```

Result [20.12.2024, 08:15:47]:
```
{"ac_charge":[0,0,0,0.6,0,0,0,0.4,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"dc_charge":[1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1],"discharge_allowed":[0,1,0,0,0,0,0,0,1,1,1,1,1,1,1,1,1,1,1,1,1,0,0,0,1,1,1,1,0,1,0,1,1,1,1,1,1,0,0,1,1,1,1,1,1,1,1,1],"eautocharge_hours_float":null,"result":{"Last_Wh_pro_Stunde":[659.6965636725695,725.4698199725692,681.4450092593752,966.4896076611103,679.8590361843625,1047.6947257027189,987.9949799952107,934.6399933951386,839.4373879152778,1098.2586248118055,755.6356805954856,810.8847254263887,653.6064676145832,645.3896876156249,616.1744979826389,623.2950833534721,594.8787128003471,600.0998508152779,594.4037516944442,590.4052838149306,594.8801742871531,863.2662758555559,858.7942874329864,649.5943344456377,612.2377633769318,1024.1865312652778,628.3315629597221,700.7526648690973,761.1067034253471,822.5515592006943,1059.0963958128473,780.8814741843346,936.695584689624,1036.795930965278,1010.9066945565975,862.4403998312504,645.3558516027779,625.2972394225695,629.7641745263887,618.7367612607641],"EAuto_SoC_pro_Stunde":[100,100,100,100,100,100,100,100,100,100,100,100,100,100,100,100,100,100,100,100,100,100,100,100,100,100,100,100,100,100,100,100,100,100,100,100,100,100,100,100],"Einnahmen_Euro_pro_Stunde":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"Gesamt_Verluste":3241.193041389337,"Gesamtbilanz_Euro":1.162629493611469,"Gesamteinnahmen_Euro":0,"Gesamtkosten_Euro":1.162629493611469,"Home_appliance_wh_per_hour":[null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null, [... cut ...] ,8,10,13,9,11,9,8],"washingstart":null}
```

EOS.config.json
```
{
  "directories": {
    "output": "output",
    "cache": "cache"
  },
  "eos": {
    "prediction_hours": 48,
    "optimization_hours": 48,
    "penalty": 10,
    "available_charging_rates_in_percentage": [
        0.0, 0.2, 0.4, 0.5, 0.6, 0.8, 1.0
    ],
    "feed_in_tariff_eur_per_wh": 48
  }
}
```
visualization_results.pdf:

[visualization_results.pdf](https://github.com/user-attachments/files/18207420/visualization_results.pdf)

drbacke, 2024-12-20:
Tahnk

> So didn't take long to get the Problem again:
>
> 13:00 to 15:00 Next Day: Discharge is going to 0 again when solar is enough to serve charge and consumption. Request [20.12.2024, 08:15:00]
>
> ``` [...]

Thanks I will test that

RobertHerter, 2024-12-20:
Same here form 11:00-14:00 next Day:
[eos_20241220_1240.log](https://github.com/user-attachments/files/18210427/eos_20241220_1240.log)
[visualization_results_20241220_1240.pdf](https://github.com/user-attachments/files/18210428/visualization_results_20241220_1240.pdf)

If needed i could provide access to all request/results and vizualization of my system and access to a grafana Instance.

drbacke, 2024-12-20:
I think this is due to the small EVQ value. I will try to get the prob. prediction into it.

drbacke, 2024-12-20:
@RobertHerter It's finished. Could you pull the PR https://github.com/Akkudoktor-EOS/EOS/pull/268 
But it's much slower now, I will get on this in the next days

closed 2024-12-21 by drbacke