Akkudoktor-EOS/EOS #261: [BUG]: Docker image not available any more?

CSchoch, 2024-12-17:
### Describe the issue:

I tried to run the docker image with portainer, but got the following error.
Also searced on dockerhub an could not find it there

### Reproduceable code example:

```python
Use docker compose in portainer and try to run.
```

### Error message:

```shell
Failed to deploy a stack: eos Pulling eos Warning pull access denied for akkudoktor/eos, repository does not exist or may require 'docker login'
```

### Version information:

Using Docker compose from main branch

Lasall, 2024-12-17:
Atm there are no published images yet, so you would need to build it yourself.

Lasall, 2024-12-17:
To build the docker image, you can use: `docker compose build`

If there are more questions about how to use/build the docker image, feel free to reopen.

closed 2024-12-17 by Lasall