Akkudoktor-EOS/EOS #287: Home Assistant add-on

Duetting, 2024-12-27:
Hi,
I tested the last few days integrating the EOS into my Home Assistant instance without all the Node-Red stuff.
To make it the usage even easier for me, I also created an add-on repo with this repo as submodule. 
The logo i used was just created by Copilot.
I hope i didn't do anything wrong here or made something that's already existing.

I could provide more information about all the intermediate sensors that i used to pass the information to the optimize endpoint, but i fear they are very specific for my setup.

The repo can also be moved somewhere else, I just wanted to get it working before.

Before i forget it, here is the link to the repo: https://github.com/Duetting/ha_eos_addon

The log messages from the s6 service still doesn't seem to work reliably, but the service works for me.

Lasall, 2024-12-27:
Pretty cool integration! We aim to provide docker images that can then just be added in config.yaml (see #277). Then the HA addon could be integrated just like https://github.com/evcc-io/hassio-addon . The goal is to have a seemless integration into the HA ecosystem (UI). One example could be https://github.com/marq24/ha-evcc .

As a first step your integration shown in https://github.com/Duetting/ha_eos_addon/blob/main/eos/DOCS.md is quite neat and we could definitely start from there as a very first step of HA support (documentation on how to use EOS with direct API calls).

Duetting, 2024-12-27:
I used my own dockerfile, cause it was easier for me to use one of the home assistant base images.
To be able to use it as add-on in home assistant, the image should have bashio and tempio installed and use s6.
Maybe you should consider at least s6, for your docker image and when there is a stable docker image available, the home assistant add-on dockerfile should only add bashio and maybe tempio and adapt the s6 scripts.
Providing all the necessary data for the optimize method in a home assistant friendly and customizable way, will be difficult I fear. Only the home assistant api provides a guaranteed way of accessing historical values for sensors, e.g.
An integration could easily provide the temperature forecast and soc values. But making the consumption, PV forecast and dynamic price values configurable, will be difficult imho.