Akkudoktor-EOS/EOS #324: in Windows accessing the server as 0.0.0.0 does not work

mschnell1, 2025-01-02:
### Describe the issue:

I suppose this in fact is not possible.
`localhost` does work
so it seems the comments written by the server might be modified.,

### Reproduceable code example:

```python
..
```

### Error message:

```shell
depends on browser
```

### Version information:

newest

NormannK, 2025-01-16:
We have just pushed a substantial update to the main repository. This update includes fixes to many errors, significant improvements to the code, and much better documentation.

We encourage you to update to the latest code and give it another try. Additionally, we've introduced significantly improved debugging methods to help address any potential issues.

For now, I'm are closing this issue. If you encounter the same error with the updated code, please reopen it.

mschnell1, 2025-01-16:
Hmm. I now get an error when installing the requirements:
` ERROR: Failed building wheel for pendulum`

Lasall, 2025-01-17:
For python 3.13 you have to install pendulum manually from source for now

mschnell1, 2025-01-17:
OK. No idea how to do that (and what pendulum is at all).
I'm not in a hurry, so I'll sit and watch for now.
Thanks !

NormannK, 2025-01-19:
reopened. server start up does still show:

server   Server started at http://0.0.0.0:8503
server   Documentation at http://0.0.0.0:8503/docs

PR #394 (2025-01-21): Fix EOSdash startup.
Revert to EOSdash startup as a sub-process.

Add arguments to control the complete startup of both servers. Also development reload can now be given as an argument.

Closes #324
Closes #388
closed 2025-01-21 by NormannK