Akkudoktor-EOS/EOS #327: [BUG]:  tests/conftest.py:76: assert not Path(user_dir).joinpath(ConfigEOS.CONFIG_FILE_NAME).exists()

b0661, 2025-01-02:
### Describe the issue:

Test fails on Github, but not local. See #321 

### Reproduceable code example:

```python
See #321 test.
```

### Error message:

```shell
tests/conftest.py:76: AssertionError
___ ERROR at teardown of TestPydanticDateTimeDataFrame.test_valid_dataframe ____

request = <SubRequest 'cfg_non_existent' for <Function test_valid_dataframe>>

    @pytest.fixture(autouse=True)
    def cfg_non_existent(request):
        yield
        if bool(request.config.getoption("--check-config-side-effect")):
            from platformdirs import user_config_dir
    
            user_dir = user_config_dir(ConfigEOS.APP_NAME)
>           assert not Path(user_dir).joinpath(ConfigEOS.CONFIG_FILE_NAME).exists()
E           AssertionError: assert not True
E            +  where True = exists()
E            +    where exists = PosixPath('/home/runner/.config/net.akkudoktor.eos/EOS.config.json').exists
E            +      where PosixPath('/home/runner/.config/net.akkudoktor.eos/EOS.config.json') = joinpath('EOS.config.json')
E            +        where joinpath = PosixPath('/home/runner/.config/net.akkudoktor.eos').joinpath
E            +          where PosixPath('/home/runner/.config/net.akkudoktor.eos') = Path('/home/runner/.config/net.akkudoktor.eos')
E            +        and   'EOS.config.json' = ConfigEOS.CONFIG_FILE_NAME
```

### Version information:

feature/config_overhaul

b0661, 2025-01-02:
Error persists also after feature/config-overhaul actualisation.

Lasall, 2025-01-02:
The flag `--check-config-side-effect` checks for side effects (config generation) of the test run. If there are side effects in module imports, all tests will fail, otherwise (if one test case has a side effect) all test cases starting from that test case will fail.

This check is disabled by default because valid user configuration might exist before the local test.

closed 2025-01-02 by Lasall