Akkudoktor-EOS/EOS #358: [Bug] feature - Strange AC Charge

drbacke, 2025-01-09:
Input:
{"ems":{"pv_prognose_wh":[0,0,0,0,0,0,0,0,0,8,46,170,247,227,282,307,247,35,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,19,174,478,576,770,833,813,415,73,0,0,0,0,0,0],"strompreis_euro_pro_wh":[0.0003219,0.0003205,0.0003193,0.0003157,0.0003188,0.0003232,0.000354,0.0003693,0.0003854,0.0003767,0.0003681,0.0003607,0.0003487,0.0003417,0.0003444,0.000348,0.0003443,0.0003514,0.0003552,0.0003459,0.0003397,0.0003255,0.0003216,0.0003087,0.0003053,0.0003007,0.0002918,0.0002895,0.0002904,0.0003044,0.0003196,0.0003444,0.00036,0.0003574,0.0003441,0.0003341,0.0003279,0.0003255,0.0003305,0.0003492,0.0003554,0.0003631,0.0003646,0.0003616,0.0003553,0.000343,0.0003403,0.0003322],"einspeiseverguetung_euro_pro_wh":[0.00007,0.00007,0.00007,0.00007,0.00007,0.00007,0.00007,0.00007,0.00007,0.00007,0.00007,0.00007,0.00007,0.00007,0.00007,0.00007,0.00007,0.00007,0.00007,0.00007,0.00007,0.00007,0.00007,0.00007,0.00007,0.00007,0.00007,0.00007,0.00007,0.00007,0.00007,0.00007,0.00007,0.00007,0.00007,0.00007,0.00007,0.00007,0.00007,0.00007,0.00007,0.00007,0.00007,0.00007,0.00007,0.00007,0.00007,0.00007],"preis_euro_pro_wh_akku":0.00022,"gesamtlast":[1478.03399933635,1272.262380050942,1498.373383129938,1636.7493211833453,1295.6866132641019,1482.4456516840155,1360.2147509587844,966.7680382953181,991.3070862563956,1183.1591581400091,1289.46978785848,1558.0527261605644,1230.577730609642,1335.3761298980212,1368.4981775396077,1120.1244007062442,1191.5843859963172,1409.9533303250262,1509.2391739363943,1722.0558173180586,1516.3847799242,1645.7831458566825,1522.5900632115681,1317.987650783864,1473.9220450624916,1267.4140001094895,1497.2955354593612,1636.0155615452913,1294.9142689706143,1481.9394856278716,1360.3643320600954,967.2034186312657,991.123590079357,1128.2067695793417,1105.5298155310381,1334.4058630371724,1218.7639233316008,1468.948999766965,1469.3547891249632,1131.704669759033,1163.9732480056693,1357.3938497446375,1811.7715323698076,1814.1602853536729,1065.0995073427525,1177.655414304526,910.6215434298542,1013.3100013103456]},"pv_akku":{"capacity_wh":26400,"charging_efficiency":0.88,"discharging_efficiency":0.88,"max_charge_power_w":5000,"initial_soc_percentage":0,"min_soc_percentage":0,"max_soc_percentage":100},"wechselrichter":{"max_power_wh":10000},"eauto":{"capacity_wh":60000,"charging_efficiency":0.95,"discharging_efficiency":1,"max_charge_power_w":11040,"initial_soc_percentage":75,"min_soc_percentage":0,"max_soc_percentage":100},"dishwasher":null,"temperature_forecast":[1.7,1.7,1.9,2.2,1.7,1.6,1.6,1.2,1.1,0.9,0.6,0.3,0.2,0,0,0.2,0.3,0.3,0.2,0.3,0.3,0.5,0.3,0.2,0.2,0.1,-0.1,-0.2,-0.4,-0.5,-0.7,-0.4,-0.4,-0.3,-0.2,0.2,0.5,0.8,1.3,1.2,0.6,0.1,0,-0.2,-0.6,-0.5,-0.6,-0.6],"start_solution":[3,6,4,11,5,3,6,0,7,4,9,3,8,5,9,3,8,1,7,9,9,9,9,9,9,9,9,9,9,9,9,9,9,9,9,9,9,9,9,9,9,9,9,9,9,9,9,9]}

Output (AC Charge is not correct)
[visualization_results.pdf](https://github.com/user-attachments/files/18362256/visualization_results.pdf)

{"ac_charge":[0,0,0.6,0,0.6,0,0,0,0.6,0,0,0,0,0,0.3,0.3,0.3,0.3,0.3,0.3,0.3,0.3,0.3,0,0,0.3,0.3,0.3,0.3,0.3,0.3,0.3,0.3,0.3,0.3,0.3,0.3,0.3,0.3,0.3,0.3,0.3,0.3,0.3,0.3,0.3,0.3,0.3],"dc_charge":[1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1],"discharge_allowed":[0,1,0,1,0,0,1,0,0,0,0,1,1,0,0,0,0,0,0,0,0,0,0,0,1,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"eautocharge_hours_float":null,"result":{"Last_Wh_pro_Stunde":[2868.4981775396077,2620.124400706244,2691.584385996317,2909.9533303250264,3009.2391739363943,3222.055817318059,3016.3847799242003,3145.7831458566825,3022.590063211568,1317.987650783864,1473.9220450624916,2767.4140001094893,2997.2955354593614,3136.0155615452913,2794.914268970614,2981.9394856278714,2860.3643320600954,2467.203418631266,2491.123590079357,2628.2067695793417,2605.5298155310384,2834.4058630371724,2718.763923331601,1872.2574257257606,1469.3547891249632,1131.704669759033,1163.9732480056693,1357.3938497446375,1811.7715323698076,1814.1602853536729,1065.0995073427525,1177.655414304526,910.6215434298542,1013.3100013103456],"EAuto_SoC_pro_Stunde":[75,75,75,75,75,75,75,75,75,75,75,75,75,75,75,75,75,75,75,75,75,75,75,75,75,75,75,75,75,75,75,75,75,75],"Einnahmen_Euro_pro_Stunde":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"Gesamt_Verluste":4029.386380896304,"Gesamtbilanz_Euro":23.57079084504485,"Gesamteinnahmen_Euro":0,"Gesamtkosten_Euro":23.57079084504485,"Home_appliance_wh_per_hour":[null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null],"Kosten_Euro_pro_Stunde":[0.890789972344641,0.804967291445773,0.841670404098532,1.0102586002762142,1.0688817545822074,1.1145091072103166,1.0246659097402508,1.02395241397635,0.9720649643288404,0.4068627877969788,0,0.8321613898329234,0.8746108372470416,0.9078765050673618,0.8116431037090663,0.9077023794251241,0.9141724405264066,0.849704857376608,0.8968044924285686,0.9325304994476568,0.8366894095242303,0.7872751988407193,0.7026122904604319,0.3587847920737351,0.21031525780580035,0.11129167067985435,0.2661850923412149,0.46636340684227784,0.6605719007020319,0.6560003591838881,0.37842985495888,0.4039358071064524,0.30988451122917937,0.3366215824352968],"Netzbezug_Wh_pro_Stunde":[2586.4981775396077,2313.124400706244,2444.584385996317,2874.9533303250264,3009.2391739363943,3222.055817318059,3016.3847799242003,3145.7831458566825,3022.590063211568,1317.987650783864,0,2767.4140001094893,2997.2955354593614,3136.0155615452913,2794.914268970614,2981.9394856278714,2860.3643320600954,2467.203418631266,2491.123590079357,2609.2067695793417,2431.5298155310384,2356.4058630371724,2142.763923331601,1102.2574257257606,636.3547891249632,318.7046697590331,748.9732480056693,1284.3938497446375,1811.7715323698076,1814.1602853536729,1065.0995073427525,1177.655414304526,910.6215434298542,1013.3100013103456],"Netzeinspeisung_Wh_pro_Stunde":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"Verluste_Pro_Stunde":[180,180,180,180,180,180,180,180,180,0,200.9893697812488,180,180,180,180,180,180,180,180,180,180,180,180,48.39701111505548,0,0,0,0,0,0,0,0,0,0],"akku_soc_pro_stunde":[0,5,10,15,20,25,30,35,40,45,45,38.65563858013735,43.65563858013735,48.65563858013735,53.65563858013734,58.65563858013735,63.65563858013735,68.65563858013735,73.65563858013735,78.65563858013735,83.65563858013735,88.65563858013735,93.65563858013735,98.65563858013735,100,100,100,100,100,100,100,100,100,100],"Electricity_price":[0.0003444,0.000348,0.0003443,0.0003514,0.0003552,0.0003459,0.0003397,0.0003255,0.0003216,0.0003087,0.0003053,0.0003007,0.0002918,0.0002895,0.0002904,0.0003044,0.0003196,0.0003444,0.00036,0.0003574,0.0003441,0.0003341,0.0003279,0.0003255,0.0003305,0.0003492,0.0003554,0.0003631,0.0003646,0.0003616,0.0003553,0.000343,0.0003403,0.0003322]},"eauto_obj":{"charge_array":[1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1],"discharge_array":[1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1],"discharging_efficiency":1,"hours":48,"capacity_wh":60000,"charging_efficiency":0.95,"max_charge_power_w":11040,"soc_wh":45000,"initial_soc_percentage":75},"start_solution":[3,5,10,6,10,1,6,0,10,2,2,6,7,3,9,9,9,9,9,9,9,9,9,1,6,9,9,9,9,9,9,9,9,9,9,9,9,9,9,9,9,9,9,9,9,9,9,9],"washingstart":null}

Lasall, 2025-01-09:
input data main branch:
```json
{
  "ems": {
    "pv_prognose_wh": [
      0, 0, 0, 0, 0, 0, 0, 0, 0, 8, 46, 170, 247, 227, 282, 307, 247, 35, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 19, 174, 478, 576, 770, 833, 813, 415, 73, 0, 0, 0, 0, 0, 0
    ],
    "strompreis_euro_pro_wh": [
      0.0003219, 0.0003205, 0.0003193, 0.0003157, 0.0003188, 0.0003232, 0.000354, 0.0003693, 0.0003854, 0.0003767, 0.0003681, 0.0003607, 0.0003487, 0.0003417, 0.0003444, 0.000348, 0.0003443, 0.0003514, 0.0003552, 0.0003459, 0.0003397, 0.0003255, 0.0003216, 0.0003087, 0.0003053, 0.0003007, 0.0002918, 0.0002895, 0.0002904, 0.0003044, 0.0003196, 0.0003444, 0.00036, 0.0003574, 0.0003441, 0.0003341, 0.0003279, 0.0003255, 0.0003305, 0.0003492, 0.0003554, 0.0003631, 0.0003646, 0.0003616, 0.0003553, 0.000343, 0.0003403, 0.0003322
    ],
    "einspeiseverguetung_euro_pro_wh": [
      0.00007, 0.00007, 0.00007, 0.00007, 0.00007, 0.00007, 0.00007, 0.00007, 0.00007, 0.00007, 0.00007, 0.00007, 0.00007, 0.00007, 0.00007, 0.00007, 0.00007, 0.00007, 0.00007, 0.00007, 0.00007, 0.00007, 0.00007, 0.00007, 0.00007, 0.00007, 0.00007, 0.00007, 0.00007, 0.00007, 0.00007, 0.00007, 0.00007, 0.00007, 0.00007, 0.00007, 0.00007, 0.00007, 0.00007, 0.00007, 0.00007, 0.00007, 0.00007, 0.00007, 0.00007, 0.00007, 0.00007, 0.00007
    ],
    "preis_euro_pro_wh_akku": 0.00022,
    "gesamtlast": [
      1478.03399933635, 1272.262380050942, 1498.373383129938, 1636.7493211833453, 1295.6866132641019, 1482.4456516840155, 1360.2147509587844, 966.7680382953181, 991.3070862563956, 1183.1591581400091, 1289.46978785848, 1558.0527261605644, 1230.577730609642, 1335.3761298980212, 1368.4981775396077, 1120.1244007062442, 1191.5843859963172, 1409.9533303250262, 1509.2391739363943, 1722.0558173180586, 1516.3847799242, 1645.7831458566825, 1522.5900632115681, 1317.987650783864, 1473.9220450624916, 1267.4140001094895, 1497.2955354593612, 1636.0155615452913, 1294.9142689706143, 1481.9394856278716, 1360.3643320600954, 967.2034186312657, 991.123590079357, 1128.2067695793417, 1105.5298155310381, 1334.4058630371724, 1218.7639233316008, 1468.948999766965, 1469.3547891249632, 1131.704669759033, 1163.9732480056693, 1357.3938497446375, 1811.7715323698076, 1814.1602853536729, 1065.0995073427525, 1177.655414304526, 910.6215434298542, 1013.3100013103456
    ]
  },
  "pv_akku": {
    "kapazitaet_wh": 26400,
    "lade_effizienz": 0.88,
    "entlade_effizienz": 0.88,
    "max_ladeleistung_w": 5000,
    "start_soc_prozent": 0,
    "min_soc_prozent": 0,
    "max_soc_prozent": 100
  },
  "wechselrichter": {
    "max_power_wh": 10000
  },
  "eauto": {
    "kapazitaet_wh": 60000,
    "lade_effizienz": 0.95,
    "entlade_effizienz": 1,
    "max_ladeleistung_w": 11040,
    "start_soc_prozent": 75,
    "min_soc_prozent": 0,
    "max_soc_prozent": 100
  },
  "dishwasher": null,
  "temperature_forecast": [
    1.7, 1.7, 1.9, 2.2, 1.7, 1.6, 1.6, 1.2, 1.1, 0.9, 0.6, 0.3, 0.2, 0, 0, 0.2, 0.3, 0.3, 0.2, 0.3, 0.3, 0.5, 0.3, 0.2, 0.2, 0.1, -0.1, -0.2, -0.4, -0.5, -0.7, -0.4, -0.4, -0.3, -0.2, 0.2, 0.5, 0.8, 1.3, 1.2, 0.6, 0.1, 0, -0.2, -0.6, -0.5, -0.6, -0.6
  ],
  "start_solution": [
    3, 6, 4, 11, 5, 3, 6, 0, 7, 4, 9, 3, 8, 5, 9, 3, 8, 1, 7, 9, 9, 9, 9, 9, 9, 9, 9, 9, 9, 9, 9, 9, 9, 9, 9, 9, 9, 9, 9, 9, 9, 9, 9, 9, 9, 9, 9, 9
  ]
}
```

changes between feature and main (both /optimize with start_hour=14):
```diff
--- output_358_main.json
+++ output_358_feature.json
@@ -52,13 +52,13 @@
     "discharge_array": [
       1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1
     ],
-    "entlade_effizienz": 1,
+    "discharging_efficiency": 1,
     "hours": 48,
-    "kapazitaet_wh": 60000,
-    "lade_effizienz": 0.95,
-    "max_ladeleistung_w": 11040,
+    "capacity_wh": 60000,
+    "charging_efficiency": 0.95,
+    "max_charge_power_w": 11040,
     "soc_wh": 45000,
-    "start_soc_prozent": 75
+    "initial_soc_percentage": 75
   },
   "start_solution": [
     3, 6, 4, 11, 5, 3, 6, 0, 7, 4, 9, 3, 8, 5, 9, 9, 9, 9, 9, 9, 9, 9, 9, 9, 9, 9, 9, 9, 9, 9, 9, 9, 9, 9, 9, 9, 9, 9, 9, 9, 9, 9, 9, 9, 9, 9, 9, 9
```

drbacke, 2025-01-09:
One optimization is not enough. And the Code was completely Clean.

b0661, 2025-01-21:
@drbacke, does this bug persist in new main? Can the issue be closed?

drbacke, 2025-01-21:
Please be patient, I'm still waiting for a similar constellation as back then. But I have my eye on the issues

closed 2025-04-16 by github-actions[bot]