Akkudoktor-EOS/EOS #365: Tibber API & tax/fees automatic

drbacke, 2025-01-12:
Tibber API would be nice and then an automatic determination of tax /  fees.

Harry79, 2025-04-04:
I am fetching the prices from https://api.tibber.com/v1-beta/gql using Fhem and some Perl code to parse the JSON and write it to a file.

This is more standalone for plotting with R:

  #+begin_src R :cache yes :file tibberpreis-my-really.svg :exports both :results graphics file :width 14 :session none
      library(jsonlite)
      library(httr)

      url <- "https://api.tibber.com/v1-beta/gql"
      body <- '{ "query": "{viewer {homes {currentSubscription {priceRating {hourly {entries { time energy total tax difference level}} }}}}}" }'
      config <- add_headers(c(
                            'Authorization' = 'Bearer ***YOUR KEY GO IN HERE***',
                            'Content-Type' = 'application/json'))
      result <- POST(url, config = config, body = body, encode = "raw", verbose())

      entries <- content(result)$data$viewer$homes[[1]]$currentSubscription$priceRating$hourly$entries

      df <- data.frame(matrix(NA, nrow=length(entries), ncol=6))
      colnames(df) <- c("time", "energy", "total", "tax", "difference", "level")
      for (x in 1:length(entries)) {
        df$time[x] <- entries[x][[1]]$time
        df$energy[x] <- entries[x][[1]]$energy
        df$total[x] <- entries[x][[1]]$total
        df$tax[x] <- entries[x][[1]]$tax
        df$difference[x] <- entries[x][[1]]$difference
        df$level[x] <- entries[x][[1]]$level
      }

      write.csv(df, "tibberpreis-my-really.csv", row.names = FALSE)

      df$time <- as.POSIXct(sub("([+-][0-9]{2}):([0-9]{2})$", "\\1\\2",df$time), format = "%Y-%m-%dT%H:%M:%OS%z", tz = "UTC")
      plot(df$time, 100*df$total, ylab="Preis / (ct/kWh)", type = "s", ylim=c(0,max(100*df$total)), lwd=2)
      lines(df$time, 100*df$tax, type = "s")
      grid(nx = 0, ny = NULL, lty = 1)
      abline(v=Sys.time(), col = "red")

      abline(h = 0, col = "black", lwd = 2)
    #+end_src

This is what it looks like:

[image]

SchaafAlexander, 2025-08-30:
I integrated Tibber with the API in my NodeRed pipeline for the EOS prediction.
The good thing is it has my real prices.
Bad thing is that when I run it at 00:05 the path Tibber Tomorrow cannot retreive values, because Tibber does only provide them at ~13:00 each day.
Any idea how to integrate it and overcome this issue with forecasting of the local taxes and in regards to coming dynamic "Netzentgelte"?
Thanks

[image]