Akkudoktor-EOS/EOS #370: [BUG]:  PVForecast Endpoint fails with 500

sushi86, 2025-01-12:
### Describe the issue:

Tried to call pvforcast endpoint with plain url value or with urlencoded but alway get Internal Server Error.

### Reproduceable code example:

```python
curl --location 'localhost:8503/pvforecast?url=https%3A%2F%2Fapi.akkudoktor.net%2Fforecast%3Flat%3D52.52%26lon%3D13.405%26power%3D5000%26azimuth%3D-10%26tilt%3D7%26powerInvertor%3D10000%26horizont%3D20%2C27%2C22%2C20%26power%3D4800%26azimuth%3D-90%26tilt%3D7%26powerInvertor%3D10000%26horizont%3D30%2C30%2C30%2C50%26power%3D1400%26azimuth%3D-40%26tilt%3D60%26powerInvertor%3D2000%26horizont%3D60%2C30%2C0%2C30%26power%3D1600%26azimuth%3D5%26tilt%3D45%26powerInvertor%3D1400%26horizont%3D45%2C25%2C30%2C60%26past_days%3D5%26cellCoEff%3D-0.36%26inverterEfficiency%3D0.8%26albedo%3D0.25%26timezone%3DEurope%2FBerlin%26hourly%3Drelativehumidity_2m%2Cwindspeed_10m%0A'

or

curl --location 'localhost:8503/pvforecast?url=https%3A%2F%2Fapi.akkudoktor.net%2Fforecast%3Flat%3D52.52%26lon%3D13.405%26power%3D5000%26azimuth%3D-10%26tilt%3D7%26powerInvertor%3D10000%26horizont%3D20%2C27%2C22%2C20%26power%3D4800%26azimuth%3D-90%26tilt%3D7%26powerInvertor%3D10000%26horizont%3D30%2C30%2C30%2C50%26power%3D1400%26azimuth%3D-40%26tilt%3D60%26powerInvertor%3D2000%26horizont%3D60%2C30%2C0%2C30%26power%3D1600%26azimuth%3D5%26tilt%3D45%26powerInvertor%3D1400%26horizont%3D45%2C25%2C30%2C60%26past_days%3D5%26cellCoEff%3D-0.36%26inverterEfficiency%3D0.8%26albedo%3D0.25%26timezone%3DEurope%252FBerlin%26hourly%3Drelativehumidity_2m%252Cwindspeed_10m%0A'
```

### Error message:

```shell
eos-1  | INFO:     172.18.0.1:62014 - "GET /pvforecast?url=https%3A%2F%2Fapi.akkudoktor.net%2Fforecast%3Flat%3D52.52%26lon%3D13.405%26power%3D5000%26azimuth%3D-10%26tilt%3D7%26powerInvertor%3D10000%26horizont%3D20%2C27%2C22%2C20%26power%3D4800%26azimuth%3D-90%26tilt%3D7%26powerInvertor%3D10000%26horizont%3D30%2C30%2C30%2C50%26power%3D1400%26azimuth%3D-40%26tilt%3D60%26powerInvertor%3D2000%26horizont%3D60%2C30%2C0%2C30%26power%3D1600%26azimuth%3D5%26tilt%3D45%26powerInvertor%3D1400%26horizont%3D45%2C25%2C30%2C60%26past_days%3D5%26cellCoEff%3D-0.36%26inverterEfficiency%3D0.8%26albedo%3D0.25%26timezone%3DEurope%252FBerlin%26hourly%3Drelativehumidity_2m%252Cwindspeed_10m%0A HTTP/1.1" 500 Internal Server Error
eos-1  | ERROR:    Exception in ASGI application
eos-1  | Traceback (most recent call last):
eos-1  |   File "/usr/local/lib/python3.12/site-packages/uvicorn/protocols/http/httptools_impl.py", line 409, in run_asgi
eos-1  |     result = await app(  # type: ignore[func-returns-value]
eos-1  |              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
eos-1  |   File "/usr/local/lib/python3.12/site-packages/uvicorn/middleware/proxy_headers.py", line 60, in __call__
eos-1  |     return await self.app(scope, receive, send)
eos-1  |            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
eos-1  |   File "/usr/local/lib/python3.12/site-packages/fastapi/applications.py", line 1054, in __call__
eos-1  |     await super().__call__(scope, receive, send)
eos-1  |   File "/usr/local/lib/python3.12/site-packages/starlette/applications.py", line 113, in __call__
eos-1  |     await self.middleware_stack(scope, receive, send)
eos-1  |   File "/usr/local/lib/python3.12/site-packages/starlette/middleware/errors.py", line 187, in __call__
eos-1  |     raise exc
eos-1  |   File "/usr/local/lib/python3.12/site-packages/starlette/middleware/errors.py", line 165, in __call__
eos-1  |     await self.app(scope, receive, _send)
eos-1  |   File "/usr/local/lib/python3.12/site-packages/starlette/middleware/exceptions.py", line 62, in __call__
eos-1  |     await wrap_app_handling_exceptions(self.app, conn)(scope, receive, send)
eos-1  |   File "/usr/local/lib/python3.12/site-packages/starlette/_exception_handler.py", line 53, in wrapped_app
eos-1  |     raise exc
eos-1  |   File "/usr/local/lib/python3.12/site-packages/starlette/_exception_handler.py", line 42, in wrapped_app
eos-1  |     await app(scope, receive, sender)
eos-1  |   File "/usr/local/lib/python3.12/site-packages/starlette/routing.py", line 715, in __call__
eos-1  |     await self.middleware_stack(scope, receive, send)
eos-1  |   File "/usr/local/lib/python3.12/site-packages/starlette/routing.py", line 735, in app
eos-1  |     await route.handle(scope, receive, send)
eos-1  |   File "/usr/local/lib/python3.12/site-packages/starlette/routing.py", line 288, in handle
eos-1  |     await self.app(scope, receive, send)
eos-1  |   File "/usr/local/lib/python3.12/site-packages/starlette/routing.py", line 76, in app
eos-1  |     await wrap_app_handling_exceptions(app, request)(scope, receive, send)
eos-1  |   File "/usr/local/lib/python3.12/site-packages/starlette/_exception_handler.py", line 53, in wrapped_app
eos-1  |     raise exc
eos-1  |   File "/usr/local/lib/python3.12/site-packages/starlette/_exception_handler.py", line 42, in wrapped_app
eos-1  |     await app(scope, receive, sender)
eos-1  |   File "/usr/local/lib/python3.12/site-packages/starlette/routing.py", line 73, in app
eos-1  |     response = await f(request)
eos-1  |                ^^^^^^^^^^^^^^^^
eos-1  |   File "/usr/local/lib/python3.12/site-packages/fastapi/routing.py", line 301, in app
eos-1  |     raw_response = await run_endpoint_function(
eos-1  |                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
eos-1  |   File "/usr/local/lib/python3.12/site-packages/fastapi/routing.py", line 214, in run_endpoint_function
eos-1  |     return await run_in_threadpool(dependant.call, **values)
eos-1  |            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
eos-1  |   File "/usr/local/lib/python3.12/site-packages/starlette/concurrency.py", line 39, in run_in_threadpool
eos-1  |     return await anyio.to_thread.run_sync(func, *args)
eos-1  |            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
eos-1  |   File "/usr/local/lib/python3.12/site-packages/anyio/to_thread.py", line 56, in run_sync
eos-1  |     return await get_async_backend().run_sync_in_worker_thread(
eos-1  |            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
eos-1  |   File "/usr/local/lib/python3.12/site-packages/anyio/_backends/_asyncio.py", line 2505, in run_sync_in_worker_thread
eos-1  |     return await future
eos-1  |            ^^^^^^^^^^^^
eos-1  |   File "/usr/local/lib/python3.12/site-packages/anyio/_backends/_asyncio.py", line 1005, in run
eos-1  |     result = context.run(func, *args)
eos-1  |              ^^^^^^^^^^^^^^^^^^^^^^^^
eos-1  |   File "/opt/eos/akkudoktoreos/server/fastapi_server.py", line 184, in fastapi_pvprognose
eos-1  |     PVforecast = PVForecast(
eos-1  |                  ^^^^^^^^^^^
eos-1  |   File "/opt/eos/akkudoktoreos/prediction/pv_forecast.py", line 269, in __init__
eos-1  |     self.process_data(
eos-1  |   File "/opt/eos/akkudoktoreos/prediction/pv_forecast.py", line 374, in process_data
eos-1  |     raise ValueError(f"Prediction input data are of unknown format: '{data_format}'.")
eos-1  | ValueError: Prediction input data are of unknown format: 'None'.
```
```

### Version information:

Commit: eabc2a795a92e36ce5a423f682efce7ab8d366f2

Started via docker-compose

Lasall, 2025-01-12:
there is a `%0A` at the end. Nevertheless the server should not 500. However this is fixed in the development branch `feature/config-overhaul`.

sushi86, 2025-01-13:
In `feature/config-overhaul` I get following error (again 500):
```
FileNotFoundError: [Errno 2] No usable temporary directory found in ['/tmp', '/var/tmp', '/usr/tmp', '/opt/eos']
``` 

Docker problem?

Lasall, 2025-01-13:
Can you try the fixed Dockerfile from #361 ?

sushi86, 2025-01-13:
There is no change in dockerfile

`CMD ["python", "-m", "akkudoktoreos.server.eos"]` 

has already been changed in the config-overhaul branch?

I also tested it starting with python directly, now I get a 404:

Response:
```
{
    "detail": "Can not get the PV forecast: \"No data found for key 'pvforecast_ac_power'.\". Did you configure the PV forecast provider?"
}
```
How can I configure the PV forecast provider, I thought this is done via passing url?

Lasall, 2025-01-13:
I'll take a look at the docker error later. For your 404: The interface changed quite a bit, the pvforevast provider has to be enabled: https://akkudoktor-eos.readthedocs.io/en/latest/akkudoktoreos/prediction.html#pvforecastakkudoktor-provider

Lasall, 2025-01-13:
Which docker image did you use? Did you use `akkudoktor/eos:feature-config-overhaul` or did you rebuild the image?

NormannK, 2025-01-16:
We have just pushed a substantial update to the main repository. This update includes fixes to many errors, significant improvements to the code, and much better documentation.

We encourage you to update to the latest code and give it another try. Additionally, we've introduced significantly improved debugging methods to help address any potential issues.

For now, I'm are closing this issue. If you encounter the same error with the updated code, please reopen it.

closed 2025-01-16 by NormannK