Akkudoktor-EOS/EOS #381: [ENH]: replace akkudoctor energyprice api by energy-charts.info

NormannK, 2025-01-14:
### Link to discussion and related issues

```python
replace akkudoctor energyprice api by energy-charts.info
https://api.energy-charts.info/#/prices/day_ahead_price_price_get
```

### Proposed implementation

```python
basically copy ElecPriceAkkudoktor to ElecPriceenergycharts
```

NormannK, 2025-01-14:
can be done in combination #365 
if tibber is used tax values should be set/checked by tibber data and energy-charts.info can be used as fall back information.
maybe even if there is more up to date data available there.

b0661, 2025-01-15:
Do Not replace, but add a new prediction. Sich you can Switch predictions by config.prediction_provider.

b0661, 2025-04-20:
Reopen due to stale policy change for enhancements.

b0661, 2025-06-23:
Now in main.

closed 2025-06-23 by b0661