Akkudoktor-EOS/EOS #387: [BUG]: Time-Axis in visualization_results.pdf are wrong (an now Marker)

RobertHerter, 2025-01-17:
### Describe the issue:

With new master the visualization_results.pdf Time-Axis is wrong an starts at 00:00 for "Now".

[image]

### Reproduceable code example:

```python

```

### Error message:

```shell

```

### Version information:

Master

NormannK, 2025-01-17:
I need to make it clearer how to interpret the graph. The red dashed line represents the current time ("now") and corresponds to the server time, adjusted for the configured time zone. The x-axis at the top shows the number of hours since the first recorded value, as it did in the old graphs. The x-axis at the bottom represents the actual time.

In the future we will also show past data before now, that is why the indicator is helpful.

Does that help?

RobertHerter, 2025-01-17:
Hello Normann,

i agree that the indictor of now is helpfull, but the lower x starts at 8:00 (witch is the time i run the solution) but the Graph and top x shows starting from 00:00. Looking at the bottom x-axis the interpretation is that i have maximum pv around midnight ;-).

RobertHerter, 2025-01-17:
Last run at 18:15 CET

[visualization_results.pdf](https://github.com/user-attachments/files/18457866/visualization_results.pdf)

NormannK, 2025-01-17:
from the request right the first item is this:

```
"pv_prognose_wh": [
                0,
                0,
                0,
                0,
                0,
                0,
                0,
                0,
                146.20268649912595,
                512.847910990359,
                856.2969903403294,
                1070.1335984765183,
                816.3972436141687,
                591.9229498875227,
                465.4621868374305,
                196.70586731633315,
                26.431601158551082,
                0,
                0,
                0,
                0,
                0,
                0,
                0,
                0,
                0,
                0,
                0,
                0,
                0,
                0,
                0,
                366.8,
                1820.15,
                3341.025,
                4550,
                5138.625,
                4983.825,
                4015.475,
                2760.625,
                1240.2,
                15.375,
                0,
                0,
                0,
                0,
                0,
                0,
            ],
```
which does fit to the PV Forecast chart. How did you generate the PV Forecast data?

NormannK, 2025-01-17:
From your request the first item is this:

```
"pv_prognose_wh": [
                0,
                0,
                0,
                0,
                0,
                0,
                0,
                0,
                146.20268649912595,
                512.847910990359,
                856.2969903403294,
                1070.1335984765183,
                816.3972436141687,
                591.9229498875227,
                465.4621868374305,
                196.70586731633315,
                26.431601158551082,
                0,
                0,
                0,
                0,
                0,
                0,
                0,
                0,
                0,
                0,
                0,
                0,
                0,
                0,
                0,
                366.8,
                1820.15,
                3341.025,
                4550,
                5138.625,
                4983.825,
                4015.475,
                2760.625,
                1240.2,
                15.375,
                0,
                0,
                0,
                0,
                0,
                0,
            ],
```

which does fit to the PV Forecast chart. How did you generate the PV Forecast data?

RobertHerter, 2025-01-17:
This is the forcast from solcast for my pv. Starting from 00:00 today, as all parameters. Getting my data from my Influxdb. Timezone is Europe/Berlin for both me and the server, timezone not set in config, but lat/long.

celle1234, 2025-01-22:
I think the problem is, that different graphs start at different times. If the graph starts at start_hour it seems nearly ok (not sure if 1h off), but if the graph starts at 0 o'clock it is completely wrong. I do not really understand the code, but from my perspective the now marker and time range code is the same for all graphs. To fix this problem, it need to be different, regarding of the start time of the graph.

RobertHerter, 2025-01-25:
Hello Normann,

the Data are filled starring at today 0:00 at array Position 0, starting with Data at current Hour at array Position 0 as you suggest is, as i am looking at the optimization, wrong.

Maybe i am totaly wrong, but input-data has to be filled with Data starting today 00:00 and results starts at current hour. With that in Mind your DateTime Axis and now Marker is wrong. 

Not a big Deal for me because i store results ind influx to visualize with Grafana.

NormannK, 2025-01-25:
that is the issue. the input data has not to start with 00:00 but with the next hour after now().
Technically there is no issue in general if all the input data starts at the same time but so far that was not the intended way of doing it. 

For some of the predictions we relay on past data to interpolate or use other methods to have future data. All those assume that we want to optimize starting at the next full hour.

RobertHerter, 2025-01-25:
So i am confused, a Parameter to optimize is starting_hour and the results array size ls 48-current_hour. 
starting the input with current hour would result in totaly wrong data.

celle1234, 2025-01-26:
I tried to fix the problem with the timescale. One problem was the different beginnings of the graphs. The more complacted problem was the timezone thing. It is importent to set timezone in the configuration. When you can use the fixed code.

```
diff --git a/src/akkudoktoreos/utils/visualize.py b/src/akkudoktoreos/utils/visualize.py
index 3c63ab3..1fe54ca 100644
--- a/src/akkudoktoreos/utils/visualize.py
+++ b/src/akkudoktoreos/utils/visualize.py
@@ -147,14 +147,14 @@ class VisualizationReport(ConfigMixin):

             # Format the time axis
             plt.gca().xaxis.set_major_formatter(
-                mdates.DateFormatter("%Y-%m-%d")
+                mdates.DateFormatter("%Y-%m-%d", tz=self.config.timezone)
             )  # Show date and time
             plt.gca().xaxis.set_major_locator(
-                mdates.DayLocator(interval=1, tz=None)
+                mdates.DayLocator(interval=1, tz=self.config.timezone)
             )  # Major ticks every day
-            plt.gca().xaxis.set_minor_locator(mdates.HourLocator(interval=3, tz=None))
+            plt.gca().xaxis.set_minor_locator(mdates.HourLocator(interval=2, tz=self.config.timezone))
             # Minor ticks every 6 hours
-            plt.gca().xaxis.set_minor_formatter(mdates.DateFormatter("%H"))
+            plt.gca().xaxis.set_minor_formatter(mdates.DateFormatter("%H", tz=self.config.timezone))
             # plt.gcf().autofmt_xdate(rotation=45, which="major")
             # Auto-format the x-axis for readability

@@ -174,6 +174,7 @@ class VisualizationReport(ConfigMixin):

             # Add vertical line for the current date if within the axis range
             current_time = pendulum.now(self.config.timezone)
+            #current_time = pendulum.now().add(hours=1)
             if timestamps[0].subtract(hours=2) <= current_time <= timestamps[-1]:
                 plt.axvline(current_time, color="r", linestyle="--", label="Now")
                 plt.text(current_time, plt.ylim()[1], "Now", color="r", ha="center", va="bottom")
@@ -419,10 +420,11 @@ def prepare_visualize(
     start_hour: Optional[int] = 0,
 ) -> None:
     report = VisualizationReport(filename)
-    next_full_hour_date = pendulum.now(report.config.timezone).start_of("hour").add(hours=1)
+    next_full_hour_date = pendulum.now(report.config.timezone).start_of("hour").add(hours=0)
+    full_hour_date_today = pendulum.now(report.config.timezone).start_of("day").add(hours=0)
     # Group 1:
     report.create_line_chart_date(
-        next_full_hour_date,  # start_date
+        full_hour_date_today,  # start_date
         [
             parameters.ems.gesamtlast,
         ],
@@ -432,7 +434,7 @@ def prepare_visualize(
         labels=["Total Load (Wh)"],
     )
     report.create_line_chart_date(
-        next_full_hour_date,  # start_date
+        full_hour_date_today,  # start_date
         [
             parameters.ems.pv_prognose_wh,
         ],
@@ -442,7 +444,7 @@ def prepare_visualize(
     )

     report.create_line_chart_date(
-        next_full_hour_date,  # start_date
+        full_hour_date_today,  # start_date
         [np.full(len(parameters.ems.gesamtlast), parameters.ems.einspeiseverguetung_euro_pro_wh)],
         title="Remuneration",
         # xlabel="Hours", # not enough space
@@ -451,7 +453,7 @@ def prepare_visualize(
     )
     if parameters.temperature_forecast:
         report.create_line_chart_date(
-            next_full_hour_date,  # start_date
+            full_hour_date_today,  # start_date
             [
                 parameters.temperature_forecast,
             ],
@@ -501,7 +503,7 @@ def prepare_visualize(
         markers=["o", "x"],
     )
     report.create_line_chart_date(
-        next_full_hour_date,  # start_date
+        full_hour_date_today,  # start_date
         [parameters.ems.strompreis_euro_pro_wh],
         # title="Electricity Price", # not enough space
         # xlabel="Date", # not enough space

```

NormannK, 2025-01-26:
it would be great if you could directly create a pull request.

commit 6cc9a5fd (2025-02-07): visualize: fix timestamps on diagrams (#430) Closes #387

* visualize: fix timestamps on diagrams
* set start time in all graphs to the same beginning hour

---------
closed 2025-02-07 by Lasall