Akkudoktor-EOS/EOS #388: [BUG]: EOSDash is not starting

RobertHerter, 2025-01-17:
### Describe the issue:

EOS Dash is not starting. But config is set:

  "server_eos_verbose": true,
  "server_eos_startup_eosdash": true,
  "server_eos_host": "0.0.0.0",
  "server_eos_port": 8503,
  "server_eosdash_host": "0.0.0.0",
  "server_eosdash_port": 8504,

[image]

### Reproduceable code example:

```python

```

### Error message:

```shell

```

### Version information:

Master

b0661, 2025-01-17:
@RobertHerter , thank you for the report. I have some changes in the pipeline but it will still take some time.  To be sure this isn´t a timing problem could you check whether you can access http://localhost:8504 some time after the startup?

RobertHerter, 2025-01-17:
@b0661 No there no process is listening on port 8504 on any ip.

b0661, 2025-01-17:
Thanks, I will prepare a PR next week. In the meantime if you really need EOSdash (which is a proof of concept today) you can start it manually.

RobertHerter, 2025-01-20:
After doing an "make install", eosdash ist starting with fastapi-server.

PR #394 (2025-01-21): Fix EOSdash startup.
Revert to EOSdash startup as a sub-process.

Add arguments to control the complete startup of both servers. Also development reload can now be given as an argument.

Closes #324
Closes #388
closed 2025-01-21 by NormannK