Akkudoktor-EOS/EOS #391: [ENH]: enhance privacy by limiting accuracy of lat and long values

NormannK, 2025-01-19:
### Link to discussion and related issues

```python
Any latitude and longitude values managed by the system should be restricted to an accuracy level that prevents exposing the user's precise location when transmitted via an API or included in outputs, such as debug data in a GitHub issue. If these values are stored in a configuration file, the stored data should reflect the same limited accuracy.

Accuracy versus decimal places at the equator

decimal  degrees    distance
places
-------------------------------  
0        1.0        111 km
1        0.1        11.1 km
2        0.01       1.11 km
3        0.001      111 m
4        0.0001     11.1 m
5        0.00001    1.11 m
6        0.000001   0.111 m
7        0.0000001  1.11 cm
8        0.00000001 1.11 mm

I propose a accuracy of 1.11 km so two decimal places.
```

### Proposed implementation

```python

```

b0661, 2025-01-19:
We should delegate private configuration data to a Secrets File such that in normal data only the reference to the Secret Key but Not the value itself is visible. See Home Assistent for an example.

NormannK, 2025-01-20:
that's a great way to do it! On the other hand if we don't need the information/accuracy we should not save it in the first place.
What else could we store in the Secret File currently?

b0661, 2025-01-20:
> that's a great way to do it! On the other hand if we don't need the information/accuracy we should not save it in the first place. What else could we store in the Secret File currently?

Private API-keys.

b0661, 2025-04-20:
Reopen due to stale policy change for enhancements.