Akkudoktor-EOS/EOS #392: [ENH]: Charges and Fees correct

drbacke, 2025-01-20:
### Link to discussion and related issues

```python
Fees are percentual, at the moment there is only one fix charge added, which is not correct. We should implement the single parts of fees ...
```

### Proposed implementation

```python

```

RenatusRo, 2025-01-20:
watching this from the sidelines:
even this is still too simplistic.
in other markets some of the fees depend on the day of the week or the hours of the day or even the season of the year.
[having seen all this stuff recently added to evcc, which now allows configuring a custom formula and more for tariffs]

sandboxcode, 2025-01-21:
Keep the following in mind - and it will be probably tedious to complete all data:

- time variable grid fees (german specific -> §14a EnWG) which are specific vor EVERY SINGLE distribution system operator (DSO)
- As these cost parts will greatly enable cost saving from Q2 or Q3 2025 we have to consider them
- Integration maybe via static or cost profile flag in config

As I am not fluent with the current config setup I am not able to suggest the best way of implementation. Still I wanted to mention it right here.

drbacke, 2025-01-21:
Yes i know about §14a Module 3. But this should BE only an Array for Netzentgelte 
It would be nice to have a API to get this Data automatically. But i could not find one.
And of course any Help is welcome😊

sandboxcode, 2025-01-22:
@drbacke I will talk to some people that spezialized in exactly that market domain (those API providers earn a lot of money, because they operate the databases that EnWG "konforme" private customer electricity bills can be generated).
Based on that I want to gather the information to come up with an idea to gather the most pricing details as possible respecting pareto principle. 😄 

@Lasall Should we have an exchange on that? Maybe we should start a free API service for that data haha?

drbacke, 2025-01-22:
Before that we should Change simply the fix Fees with the single parts. @sandboxcode Plesse make a new issues. This Problem is urgent

sandboxcode, 2025-04-02:
I am very packed with work right now, however I am still interested in contributing to the issue. Maybe someone else wants to pair up with me?

LordMike, 2025-10-25:
I stumbled across this, and I think we have the same in DK plus a project that uncovers this: [Strømligning](https://stromligning.dk/live). They started as a rating portal for energy companies, and now offers an API which calculates the energy prices with all taxes, tariffs and so on included (which also vary per hour, time of year, area of the country, if you're using energy for heating and so on).

The API even offers an ["optimize"](https://stromligning.dk/api/docs/#/Scheduling/post_api_scheduling_optimize) which seems to mimic Eos on the surface.

Not that its a solution, but more so you know others have it and what they're doing.