Akkudoktor-EOS/EOS #403: [ENH]: Time variable grid fees

Lasall, 2025-01-22:
### Link to discussion and related issues

```python

```

### Proposed implementation

```python

```

sandboxcode, 2025-03-11:
I talked to many different IT companies that would have been great partners to get some of the required data. Unfortunately they were not open to assist as this would destroy parts of their core business model.

I guess we have to bring the data together ourselves. For that we have to create an index / dict that holds all fee data for all DSOs (roughly +800 in Germany).

Actually this could also be a small startup idea for a simple API provider. Anybody motivated to join me on that as I have limited time? :D

TobHaeg, 2025-07-21:
https://www.variable-netzentgelte.de