Akkudoktor-EOS/EOS #408: Docker auf Synology: "ndarray.pyi stub file could not be generated: [Errno 13] Permission denied"

WolfImBusch, 2025-01-25:
Hallo,

vorab: ich bin vom der Idee und der Software ziemlich begeistert und habe mir heute endlich mal Zeit genommen, EOS in Betrieb zu nehmen. Genauer: als Docker auf meiner Synology NAS. Dazu habe ich mir soeben den aktuellen Stand von "main" heruntergeladen. Port 8503 habe ich von Docker nach außen freischaltet.

Laut Protokoll sieht es auch gar nicht so ganz schlecht aus. Ich bin aber leider ziemlicher Anfänger. Ich bitte euch schon jetzt um Entschuldigung, wenn meine Frage dämlich ist - aber ich komme nicht weiter und bitte euch um eure Hilfe. Laut Log sieht es so aus ....

```
2025/01/25 15:03:20 | stdout | [32mINFO[0m:     Uvicorn running on [1mhttp://0.0.0.0:8503[0m (Press CTRL+C to quit)
2025/01/25 15:03:20 | stdout | [32mINFO[0m:     Application startup complete.
2025/01/25 15:03:20 | stdout | [32mINFO[0m:     Waiting for application startup.
2025/01/25 15:03:20 | stdout | [32mINFO[0m:     Started server process [[36m1[0m]
2025/01/25 15:03:17 | stdout | warn(f"ndarray.pyi stub file could not be generated: {e}", stacklevel=1)
2025/01/25 15:03:17 | stdout | /usr/local/lib/python3.12/site-packages/numpydantic/meta.py:66: UserWarning: ndarray.pyi stub file could not be generated: [Errno 13] Permission denied: '/usr/local/lib/python3.12/site-packages/numpydantic/ndarray.pyi'
```

Ich habe dann versucht zu überprüfen, ob EOS trotzdem läuft, indem ich ein REST-API GET wie folgt durchgeführt habe, der leider kein Ergebnis lieferte (Timeout)
`http://192.168.178.22:8503/strompreis`

Mir ist klar, dass es sich um ein "Permission denied" Fehler handelt. Aber habt ihr eine Idee, wie ich weiter suchen könnte?

Ich danke euch und viele Grüße

Lasall, 2025-01-25:
Hallo Wolf, die Warnung kannst du ignorieren, EOS funktioniert auch mit der Warnung. Kommst du an den Endpunkt `http://192.168.178.22:8503/v1/config` ?

WolfImBusch, 2025-01-25:
Hi @Lasall 

herzlichen Dank für Deine Hilfe!!!!!

Nein - leider erreiche ich auch diesen Endpunkt nicht, also
`http://192.168.178.22:8503/v1/config`

Sind für eine erste Inbetriebnahme Änderung an der Konfigurations-Datei "EOS.config.json" erforderlich? Ich vermute nein ..

btw: EOS habe ich im Synology-Container-Manager wie folgt konfiguriert (s. Screenshot). Alle Verzeichnisse bleiben auch nach Start von EOS leer. Im Verzeichnis "config" habe ich die o.a. "EOS.config.json" eingefügt.

[image]

Herzlichen Dank und viele Grüße!!!

Lasall, 2025-01-25:
Der Zugriff auf andere Container im Container-Manager funktioniert vermutlich?

NormannK, 2025-01-25:
und welche DSM-Version hast du am laufen?

WolfImBusch, 2025-01-25:
Ja - die anderen Container funktioniern ohne Probleme (evcc, influxdb, grafana, ....)

Meine DSM-Version ist die "DSM 7.2.2-72806 Update 2" (Synology DS220+).

WolfImBusch, 2025-01-25:
Hi @Lasall , hi @NormannK ,

ihr habt mich auf eine Idee gebracht - und es funktioniert. Ich glaub's nicht ...

Ich habe jetzt komplett andere Ports verwendet (s. Screenshot). Der Endpunkt

`http://192.168.178.22:7072/v1/config`

liefert jetzt Daten zurück. Ich habe nicht die geringste Ahnung, warum das so ist - bei anderen Containern konnte ich stets den gleichen Port wie die Container selbst verwenden. Der Port 8503 ist auch nicht belegt.

Ich danke euch vielmals und hoffe, dass ich euch nicht eure wertvolle Zeit geraubt habe. Vielleicht hat irgendjemand mal das gleiche Problem und das Ganze war jetzt nicht umsonst - vielleicht sogar ein Punkt für die Doku .. :-)

Nochmal vielen Dank und viele Grüße aus Stuttgart!

[image]

Lasall, 2025-01-26:
According to https://kb.synology.com/en-me/DSM/tutorial/What_network_ports_are_used_by_Synology_services the MailPlus Server is using ports 8500-8520. I will update the docs.

WolfImBusch, 2025-04-07:
From my point of view the issue can be closed as @Lasall has been updating the docs. Thx for that - I'm almost sure other guys will help that as well ... 😅

commit 65164550 (2025-01-26): Readme: Add hint for interfering ports on Synology Closes #408 (#419)