Akkudoktor-EOS/EOS #428: [BUG]: Negativ values in load_mean_adjusted

XHalamaX, 2025-01-27:
### Describe the issue:

I get incorrect values from the load prediction (load_mean_adjusted)

I put these values to /v1/measurement/data
[payload_load_measurement.json](https://github.com/user-attachments/files/18563754/payload_load_measurement.json)

Then I verfiied that the data is in the system (now there are already some more data than in the file above) via /v1/measurement/series?key=measurement_load0_mr (all other loadx are empty)
[measurement_export.json](https://github.com/user-attachments/files/18563739/measurement_export.json)

I get a negativ load prediction via /v1/prediction/list?key=load_mean_adjusted
[load_prediction.json](https://github.com/user-attachments/files/18563785/load_prediction.json)

### Reproduceable code example:

```python

```

### Error message:

```shell

```

### Version information:

[f09658578a9898dd29eecafe3ddaf99fc3677c29](https://github.com/Akkudoktor-EOS/EOS/commit/f09658578a9898dd29eecafe3ddaf99fc3677c29)
running via docker

b0661, 2025-01-28:
Thank you for the detailed information. We will have a look into it.

XHalamaX, 2025-01-29:
Did some more tests and now I have the issue again. So the missing loadakkudoktor_year_energy was not the problem.

XHalamaX, 2025-02-28:
@b0661 Can you reproduce the issue? Would be great to have a better load prediction

Tested it today again and I still get negativ values

b0661, 2025-03-02:
The problem may arise here, if the adjustments are negative and bigger than the forecast load value:
- https://github.com/Akkudoktor-EOS/EOS/blob/d2136f1447df31734e5ede4bcb6197442be8a990/src/akkudoktoreos/prediction/loadakkudoktor.py#L123
- https://github.com/Akkudoktor-EOS/EOS/blob/d2136f1447df31734e5ede4bcb6197442be8a990/src/akkudoktoreos/prediction/loadakkudoktor.py#L126

 Asking @NormannK to have a look.

NormannK, 2025-03-02:
I had not yet a deeper look in to the load adjust. but I think you are right. I guess we can floor it at 0.

```
adjusted_value = hourly_stats[0] + weekday_adjust[date.hour]
values["load_mean_adjusted"] = max(0, adjusted_value)
```

b0661, 2025-04-20:
@XHalamaX, could you please test whether this is fixed in main now. 

Note: There is BREAKING CHANGE of the configuration format in main.

XHalamaX, 2025-08-31:
ok now I figured it out. But it's still not correct.
These are my measurements I put to
`/v1/measurement/series?key=load0_mr`

[payload_load_measurement_v2.json](https://github.com/user-attachments/files/22068225/payload_load_measurement_v2.json)

Then I try to get the prediction via
`/v1/prediction/series?key=load_mean_adjusted&start_datetime=2025-08-30T22%3A00%3A00.000Z&end_datetime=2025-09-01T22%3A00%3A00.000Z`

and receive
```
[
  515.499426286098,
  476.6909208536522,
  124.81738240180181,
  0,
  0.3691730495543766,
  0,
  52.219784690923916,
  225.7710544517692,
  777.5554080488292,
  0,
  0,
  716.6113552368245,
  0,
  0,
  83.59911784047654,
  0,
  0,
  0,
  185.8617743211163,
  26.696301890420045,
  0,
  0,
  0,
  0,
  0,
  0,
  451.30503228083865,
  437.9912775803706,
  433.7687298194354,
  524.8960270460053,
  702.3696940901351,
  714.647283278112,
  807.680528564381,
  879.6066312621899,
  891.6934607690795,
  137.6021971314201,
  0,
  375.97707528874,
  1010.5720605642882,
  0,
  0,
  370.61002302105976,
  0,
  0,
  0,
  12.087220377163703,
  0,
  14.345667831341643
]
```

Does anybody know how this works or what I could check?

kleini, 2025-09-02:
I have the same issue. Configured yearly consumption 7.6MWh, uploaded load0_mr measurements representing my total loads and the output of prediction load_mean_adjusted is very often just zero, while load_mean is between 500Wh and 1500Wh, which is not accurate to my hourly loads but way more accurate to my yearly consumption.

So, there must be an issue when adjusting the load according to measurement.

PR #491 (2025-03-27): Fix negative values load mean adjusted
Fixes #428

Waits for #477 to be merged.
closed 2026-01-01 by NormannK