Akkudoktor-EOS/EOS #436: [BUG]: API documentation load error

jwende, 2025-02-01:
### Describe the issue:

I followed the first steps to install the code on my Windows system and started the fastapi server. 

The server starts on port 8000 (not 8503). I opened localhost:8000/docs and this returned:

[image]

the server shows:
   _FastAPI   Starting production server 🚀

             Searching for package file structure from directories with __init__.py files
             Importing from C:\energy\EOS\EOS\src

    module   📁 akkudoktoreos
             ├── 🐍 __init__.py
             └── 📁 server
                 ├── 🐍 __init__.py
                 └── 🐍 eos.py

      code   Importing the FastAPI app object from the module with the following code:

             from akkudoktoreos.server.eos import app

       app   Using import string: akkudoktoreos.server.eos:app

    server   Server started at http://0.0.0.0:8000
    server   Documentation at http://0.0.0.0:8000/docs

             Logs:

      INFO   Started server process [23072]
      INFO   Waiting for application startup.
      INFO   Application startup complete.
      INFO   Uvicorn running on http://0.0.0.0:8000 (Press CTRL+C to quit)
      INFO   127.0.0.1:61605 - "GET /docs HTTP/1.1" 200
      INFO   127.0.0.1:61616 - "GET /docs HTTP/1.1" 200_ 

Thanks
Joerg

### Reproduceable code example:

```python

```

### Error message:

```shell

```

### Version information:

Windows 11
installation script from github

jwende, 2025-02-01:
it works if I use the docker image (docker compose up)
J

NormannK, 2025-02-09:
should be fixed now thanks to @Lasall. Please report here if its not working.

commit b380624c (2025-02-09): Windows: Fix EOSdash startup Closes #436 #447 (#450)

* On Windows use 127.0.0.1 as default config host (model defaults) and
   addionally redirect 0.0.0.0 to localhost on Windows (because default
   config file still has 0.0.0.0).
   Use 0.0.0.0 as default otherwise (e.g. Linux/Docker) to allow EOS
   being accessible on local network (not just same host).
   Note: Docs generation on Windows is incompatible with the Github
   pipeline tests. Address this in the nested-config feature branch.
 * Update install/startup instructions as package installation is
   required atm and Docker on Windows [... cut]
closed 2025-02-09 by NormannK