Akkudoktor-EOS/EOS #445: [BUG]: Failed to resolve 'api.akkudoktor.net' - sometimes

XHalamaX, 2025-02-04:
### Describe the issue:

Hi,
sometimes I get "Internal Server Error" response from /optimize. Below is the error message which I can see in the logs. I understand the error message but don't know why this happens sometimes.

I call the  /v1/prediction/list?key=elecprice_marketprice_wh API first and get a response. Does /optimize call api.akkudoktor.net again or is it maybe still busi with the previous request?

The problem appear just sometimes and I didn't found a way to force this issue. Had this a few days before already but didn't created a issue for it. Now it happened again

### Reproduceable code example:

```python

```

### Error message:

```shell
<details>
2025-02-04T18:18:01.883262800Z ERROR:    Exception in ASGI application
2025-02-04T18:18:01.883439556Z Traceback (most recent call last):
2025-02-04T18:18:01.883443786Z   File "/usr/local/lib/python3.12/site-packages/urllib3/connection.py", line 198, in _new_conn
2025-02-04T18:18:01.883446926Z     sock = connection.create_connection(
2025-02-04T18:18:01.883450446Z            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
2025-02-04T18:18:01.883453306Z   File "/usr/local/lib/python3.12/site-packages/urllib3/util/connection.py", line 60, in create_connection
2025-02-04T18:18:01.883467997Z     for res in socket.getaddrinfo(host, port, family, socket.SOCK_STREAM):
2025-02-04T18:18:01.883471007Z                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
2025-02-04T18:18:01.883473767Z   File "/usr/local/lib/python3.12/socket.py", line 976, in getaddrinfo
2025-02-04T18:18:01.883476587Z     for res in _socket.getaddrinfo(host, port, family, type, proto, flags):
2025-02-04T18:18:01.883479357Z                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
2025-02-04T18:18:01.883523819Z socket.gaierror: [Errno -5] No address associated with hostname
2025-02-04T18:18:01.883526919Z 
2025-02-04T18:18:01.883529679Z The above exception was the direct cause of the following exception:
2025-02-04T18:18:01.883543819Z 
2025-02-04T18:18:01.883546569Z Traceback (most recent call last):
2025-02-04T18:18:01.883549219Z   File "/usr/local/lib/python3.12/site-packages/urllib3/connectionpool.py", line 787, in urlopen
2025-02-04T18:18:01.883552050Z     response = self._make_request(
2025-02-04T18:18:01.883554760Z                ^^^^^^^^^^^^^^^^^^^
2025-02-04T18:18:01.883557750Z   File "/usr/local/lib/python3.12/site-packages/urllib3/connectionpool.py", line 488, in _make_request
2025-02-04T18:18:01.883560630Z     raise new_e
2025-02-04T18:18:01.883563340Z   File "/usr/local/lib/python3.12/site-packages/urllib3/connectionpool.py", line 464, in _make_request
2025-02-04T18:18:01.883566150Z     self._validate_conn(conn)
2025-02-04T18:18:01.883568880Z   File "/usr/local/lib/python3.12/site-packages/urllib3/connectionpool.py", line 1093, in _validate_conn
2025-02-04T18:18:01.883571770Z     conn.connect()
2025-02-04T18:18:01.883574410Z   File "/usr/local/lib/python3.12/site-packages/urllib3/connection.py", line 704, in connect
2025-02-04T18:18:01.883577290Z     self.sock = sock = self._new_conn()
2025-02-04T18:18:01.883579950Z                        ^^^^^^^^^^^^^^^^
2025-02-04T18:18:01.883582591Z   File "/usr/local/lib/python3.12/site-packages/urllib3/connection.py", line 205, in _new_conn
2025-02-04T18:18:01.883585401Z     raise NameResolutionError(self.host, self, e) from e
2025-02-04T18:18:01.883588201Z urllib3.exceptions.NameResolutionError: <urllib3.connection.HTTPSConnection object at 0x7fe020965250>: Failed to resolve 'api.akkudoktor.net' ([Errno -5] No address associated with hostname)
2025-02-04T18:18:01.883591771Z 
2025-02-04T18:18:01.883594421Z The above exception was the direct cause of the following exception:
2025-02-04T18:18:01.883597191Z 
2025-02-04T18:18:01.883605151Z Traceback (most recent call last):
2025-02-04T18:18:01.883618622Z   File "/usr/local/lib/python3.12/site-packages/requests/adapters.py", line 667, in send
2025-02-04T18:18:01.883621672Z     resp = conn.urlopen(
2025-02-04T18:18:01.883624402Z            ^^^^^^^^^^^^^
2025-02-04T18:18:01.883627052Z   File "/usr/local/lib/python3.12/site-packages/urllib3/connectionpool.py", line 841, in urlopen
2025-02-04T18:18:01.883629882Z     retries = retries.increment(
2025-02-04T18:18:01.883632552Z               ^^^^^^^^^^^^^^^^^^
2025-02-04T18:18:01.883635182Z   File "/usr/local/lib/python3.12/site-packages/urllib3/util/retry.py", line 519, in increment
2025-02-04T18:18:01.883638012Z     raise MaxRetryError(_pool, url, reason) from reason  # type: ignore[arg-type]
2025-02-04T18:18:01.883648843Z     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
2025-02-04T18:18:01.883651833Z urllib3.exceptions.MaxRetryError: HTTPSConnectionPool(host='api.akkudoktor.net', port=443): Max retries exceeded with url: /prices?start=2024-12-31&end=2025-02-06&tz=Europe/Berlin (Caused by NameResolutionError("<urllib3.connection.HTTPSConnection object at 0x7fe020965250>: Failed to resolve 'api.akkudoktor.net' ([Errno -5] No address associated with hostname)"))
2025-02-04T18:18:01.883655513Z 
2025-02-04T18:18:01.883658163Z During handling of the above exception, another exception occurred:
2025-02-04T18:18:01.883660883Z 
2025-02-04T18:18:01.883663563Z Traceback (most recent call last):
2025-02-04T18:18:01.883671233Z   File "/usr/local/lib/python3.12/site-packages/uvicorn/protocols/http/httptools_impl.py", line 409, in run_asgi
2025-02-04T18:18:01.883675413Z     result = await app(  # type: ignore[func-returns-value]
[... 64 lines cut ...]
2025-02-04T18:18:01.885894175Z     response = requests.get(url)
2025-02-04T18:18:01.885983377Z                ^^^^^^^^^^^^^^^^^
2025-02-04T18:18:01.885994768Z   File "/usr/local/lib/python3.12/site-packages/requests/api.py", line 73, in get
2025-02-04T18:18:01.885997768Z     return request("get", url, params=params, **kwargs)
2025-02-04T18:18:01.886000548Z            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
2025-02-04T18:18:01.886003218Z   File "/usr/local/lib/python3.12/site-packages/requests/api.py", line 59, in request
2025-02-04T18:18:01.886005938Z     return session.request(method=method, url=url, **kwargs)
2025-02-04T18:18:01.886008558Z            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
2025-02-04T18:18:01.886011158Z   File "/usr/local/lib/python3.12/site-packages/requests/sessions.py", line 589, in request
2025-02-04T18:18:01.886021679Z     resp = self.send(prep, **send_kwargs)
2025-02-04T18:18:01.886024459Z            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
2025-02-04T18:18:01.886026999Z   File "/usr/local/lib/python3.12/site-packages/requests/sessions.py", line 703, in send
2025-02-04T18:18:01.886029639Z     r = adapter.send(request, **kwargs)
2025-02-04T18:18:01.886032199Z         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
2025-02-04T18:18:01.886034759Z   File "/usr/local/lib/python3.12/site-packages/requests/adapters.py", line 700, in send
2025-02-04T18:18:01.886037479Z     raise ConnectionError(e, request=request)
2025-02-04T18:18:01.886040109Z requests.exceptions.ConnectionError: HTTPSConnectionPool(host='api.akkudoktor.net', port=443): Max retries exceeded with url: /prices?start=2024-12-31&end=2025-02-06&tz=Europe/Berlin (Caused by NameResolutionError("<urllib3.connection.HTTPSConnection object at 0x7fe020965250>: Failed to resolve 'api.akkudoktor.net' ([Errno -5] No address associated with hostname)"))
</details>
```

### Version information:

[1a2da76](https://github.com/Akkudoktor-EOS/EOS/commit/1a2da7636b7149169b44d37dd5103409a2a2db5e)

Via docker

XHalamaX, 2025-02-28:
Just in case other have the same issue. Look like 

```
    dns:
      - 8.8.8.8
```
in the docker compose solved/reduced the issue

drbacke, 2025-03-04:
MongoDB of api.akkudoktor.net was offline (caching) so a lot of requests = timeout. 
Should work again