Akkudoktor-EOS/EOS #447: [BUG]: ... 404 EOSdash server not reachable: 'http://0.0.0.0:8504/'

JoachimS-source, 2025-02-06:
### Describe the issue:

after istallation on doker using the compose file from github the server ist not reachable

### Reproduceable code example:

```python

```

### Error message:

```shell
404
```

### Version information:

63091741db57

b0661, 2025-02-08:
@JoachimS-source, it's very hard to guess  the reason or provide guidance with this little information.

If the error persists, please provide more information about your environment and what you did to trigger the bug.

ete72-arminius, 2025-02-08:
Hi, 

i have the same issue. 

I downloaded the Docker image and started the container with Docker.yaml from GIT. I use Portainer and I noticed the following things. 
1.) /opt/eos/config was routed as external (local volume).
 2.) the eos user cannot sudo -i. I couldn't find the PW for root. I have now set the setting in Portainer that I can access the Portainer console via root
3.) In /etc/systemctl I cannot find a service file from the video and the description by Andreas. 
4.) Commands like ps, pstree or ss don't work and so I can't see what's running 
5.) eos_dash_server set to true, but when I call up the URL of the Docker container, I get an error message. URL: 0.0.0.0:8504 does not exist. In the YAML file, port 8504 is not given to the outside world. What is port 8504 responsible for? IP:8503 shows error that dashserver is not started. I can access IP:8503/docs
6.) I then tried to implement the simple test scripts, but as eos I cannot create a file via vi or from because the commands are not available.
Could someone please tell me what components are present in the Docker image? To me it looks like something is still missing.
What settings for the server IPs need to be made for Docker?
The documentation does not describe which ports are used for what. For this reason I can't get any further in my thoughts.
What data do I have to set so that simulations can be created? Required fields are marked in the documentation, but I can't figure out which fields I have to fill out for which scenario.
I couldn't find Test.py from the video in the GIT. Has this script been replaced by the simple scripts in GIT? If so, how do I get them into the container or do I have to map another local volume and if so, is the directory predetermined or freely selectable?
Thank you in advance.

regards
Ete

ete72-arminius, 2025-02-08:
Hi, 

i think i found the problem. 

I start the eosdash server in a shell and this is the output:
root@20064e047234:/opt/eos/src/akkudoktoreos/server# python3 eosdash.py 
Traceback (most recent call last):
  File "/opt/eos/src/akkudoktoreos/server/eosdash.py", line 27, in <module>
    app = FastHTML()
          ^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/fasthtml/core.py", line 535, in __init__
    secret_key = get_key(secret_key, key_fname)
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/fasthtml/core.py", line 463, in get_key
    fname.write_text(key)
  File "/usr/local/lib/python3.12/pathlib.py", line 1047, in write_text
    with self.open(mode='w', encoding=encoding, errors=errors, newline=newline) as f:
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/pathlib.py", line 1013, in open
    return io.open(self, mode, buffering, encoding, errors, newline)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
OSError: [Errno 30] Read-only file system: '.sesskey'

root:/opt/eos# df -h .
Filesystem      Size  Used Avail Use% Mounted on
overlay         232G  121G  100G  55% /
root:/opt/eos# touch test
touch: cannot touch 'test': Read-only file system
root:/opt/eos# pwd
/opt/eos

The root filesystem is read only. 

regards
Ete

ete72-arminius, 2025-02-08:
I have fixed it. 
The problem i really a read only filesystem. 

I creazed a local directory an copy the data in the container under /opt/eos into the local directory with "docker cp". 
After that i delete all docker volumes /opt/eos/* und add a bind /opt/eos to the local directory. 
I restart the container login as root and change the ownership - chown -R eos:root /opt/eos. 
Now i am able to open the URL:8053 and i can see the EOS Dashboard

regards
Ete

ete72-arminius, 2025-02-08:
Parameter server_eos_startup_eosdash has to be set in the ENV to true.
It did not work in the config.json File

ete72-arminius, 2025-02-10:
Hi, 

i have downloaded the last main Tag container and i have still the issue that the eosdash server is not starting if i delete my local volume. I can still not touch a test file under /opt/eos and so directory .sesskey can not be created. 

Why you did not create a volume for /opt/eos and add this one. 
The old docker.yaml is gone and so can not see which configuration you use. 

regards
Ete

kptkip, 2025-02-17:
Here the same problem with the version from yesterday.

Setting volumes on my host leads to errors, while starting the container cause of write errors. 
Setting no external volumes starts the container but with the 404 Error.
As already mentioned, it is impossible to work as root (i.e. sudo) on the system.

Kind regards
Alex

ete72-arminius, 2025-02-17:
You have to connect to the docker container as root. sudoers is not set. 

docker exec -it --workdir / --user root akkudoktor-eos /bin/bash

regards
Ete

ete72-arminius, 2025-02-17:
Still the same. 

read only filesystem 

regards
Ete

Lasall, 2025-02-17:
> Setting volumes on my host leads to errors, while starting the container cause of write errors. Setting no external volumes starts the container but with the 404 Error. As already mentioned, it is impossible to work as root (i.e. sudo) on the system.

How do you set the volumes? (How do you start the container?)

The usage of sudo or manual writes inside the container should not be required. What is your base system?

Lasall, 2025-02-17:
If you use the docker-compose.yaml, can you try to expose the EOSdash port to the host (see the referenced PR) and verify it works?

ete72-arminius, 2025-02-17:
Hi, 

i recreate a new container with the yaml file and delete the part read only for a test. 

After that a can touch files. 

Then i start the eosdash server again and i have still the issue that the .sesskey can not be created under /opt/eos. 
I login in the container as root i am able to start the eosdash server. 
I check the ownership of /opt/eos and the owner is root and not eos. 
I change the owner to eos:root for /opt/eos and start the eosdash server again and it works. 

http://x.x.x.x:8503 and 8504 works

regards
Ete

Lasall, 2025-02-17:
Sorry I don't quite understand how you create the container. E.g. do you use Docker-Desktop?

ete72-arminius, 2025-02-18:
Hi, 
i have downloaded the compose file from Git and use that one. 
Yesterday i have downloaded the main. 

where can i find that i am able to use the ENV and what is the value for that?

Lasall, 2025-02-18:
Thank you for the detailed information. If I understand it correctly you start the container directly from the image (without the compose file). Unfortunately there is one required environment variable EOS_SERVER__EOSDASH_SESSKEY that has to be set to any string value. In #467 I provided a default for that variable so the container can be started without any changes.

To your external volumes: Do you mount host paths as volume into your container? If so make sure to have rw permissions for user-ID 100 (or group ID 101 or set o+rw). Also in case you mount custom named volumes, make sure to set rw permissions for user ID 100 or group ID 101 or o+rw for the root of those named volumes.

kptkip, 2025-02-19:
I now downloaded the main package and used the docker command from the shell of my QNAP-System.

Then it worked as a stack within the docker-container. I assume the environment variables were set correctly. Unfortunately the conatiner works on the same IP-Address like the host with the Ports 8503/8504. I prefer to have specific IP-Addresses for my containers. This is normally handled by the QNAP-GUI. And I have no clue, how to do this via shell and composit file.

External volumes:
To be honest, I normally don't care about any rw permissions for specific users(-groups) while exposing host directories for my docker containers (currently running 9) and all of them don't have any issues with rw-permissions.

commit d2136f14 (2025-02-23): Dockerfile: Set default for EOS_SERVER__EOSDASH_SESSKEY Closes #447 (#467)

* This allows to start the container without any extra settings
   (potentially unsafe).
   It is recommended to set EOS_SERVER__EOSDASH_SESSKEY.
commit 20621aa6 (2025-04-07): docker-compose: Expose EOSdash port Closes #447

 * Fixes direct EOSdash access on Windows localhost:8504 (required for
   redirect).
closed 2025-02-23 by NormannK