Akkudoktor-EOS/EOS #452: [BUG]: make run results in an error

RobertHerter, 2025-02-10:
### Describe the issue:

make run is not working

### Reproduceable code example:

```python

```

### Error message:

```shell
<details>
venv/bin/python src/akkudoktoreos/server/eos.py
Traceback (most recent call last):
  File "/opt/EOS/src/akkudoktoreos/server/eos.py", line 17, in <module>
    from akkudoktoreos.config.config import ConfigEOS, SettingsEOS, get_config
ModuleNotFoundError: No module named 'akkudoktoreos'
make: *** [Makefile:88: run] Fehler 1
</details>
```

### Version information:

master

b0661, 2025-02-10:
Please do "make install" before.

RobertHerter, 2025-02-11:
make install fix the problem, but is make run not to run out of development an not needing make install ?

b0661, 2025-02-17:
Seems to be solved. Reopen in case there are still Issues.

closed 2025-02-17 by b0661