Akkudoktor-EOS/EOS #481: [ENH]: Supporting Peak Shaving

RobertHerter, 2025-03-11:
### Link to discussion and related issues

```python
For the Summer Peak-Shaving sould be implemented.
Using einspeiseverguetung_euro_pro_wh to force Peak-Shaving is not really a 
good Solution.
```

### Proposed implementation

```python
Loading of Battery should me mainly done with solar peak.
```

SpectrixPvP, 2025-04-03:
I would also be interested in this enhancement, but how do you currently even implement peak shaving via `einspeiseverguetung_euro_pro_wh`?

If I set `einspeiseverguetung_euro_pro_wh` for one afternoon hour of the day to a negative remuneration (e.g. -0.0001 €/Wh), the battery will still charge to 100% using the previously available solar power before the negative value even occurs. During the negative value in the afternoon, the battery is already at 100% and hence can't charge the surplus solar power anymore, which therefore has to be fed into the grid with negative remuneration, which is of course not optimal for economic reasons...

Here the example, which is not working as expected.
For the remuneration, I used a day of dynamic prices, but manipulated one hour (14:00) to a very low negative value:
[image]

As a result, I would expect the battery to charge mainly during 14:00, to prevent the negative remuneration.
Instead the battery already charges to 100% until 12:00 and all solar power during 14:00 is fed into the grid with the negative remuneration. 
[image]

This has a high negative impact on the financial balance:
[image]

Harry79, 2025-04-04:
I recently played with a similar problem and ChatGPT came to the rescue with writing a reasonable algorithm to solve it. I discretized the problem a bit. Added some comments in brackets to make it more clear here.

Imagine I have a time series with individual timestamps. At each timestamp I can sell goods at a price p_i (Tibber has variable prices per hour), the maximum volume I can sell is limited by v_i (I actually do not sell just consume it myself instead of buying it at that point in time). At each timestamp I also get goods for free f_i (the sun is shining). I also have a limited stock s (my battery) which I cannot exceed at any point in time. How to calculate the most profitable trade volume t_i?

The dynamic programming approach dp_trade is supposed to really give the optimum result. The time complexity of the algorithm is a bit of an issue O(n * s * v_max). Where n is the number of timestamps (12-36 (hours)), s the maximum stock (need to check what granularity is actually required here), v_i needs the same quantization steps.

You may want to adapt it for your purpose or publish the data and others could help finding an optimal or just a good enough solution.

```
  #+begin_src python :cache yes :results output
    import numpy as np
    from functools import lru_cache
    from scipy.optimize import linprog

    def greedy_trade2(p, v, f, s):
        n = len(p)
        stock = 0
        trade = np.zeros(n, dtype=int)
        
        for i in range(n):
            stock = min(stock + f[i], s)  # Receive free goods, ensuring stock does not exceed max limit
            
            # Sell as much as possible within trade limit
            sell_now = min(v[i], stock)
            trade[i] = sell_now
            stock -= sell_now
        
        profit = sum(trade[i] * p[i] for i in range(n))
        return trade.tolist(), profit

    def dp_trade(p, v, f, s):
        n = len(p)
        
        @lru_cache(None)
        def dp(i, stock):
            if i == n:  # Base case: No more timestamps
                return 0, []
            
            max_profit = 0
            best_trade_seq = []
            
            for t in range(min(v[i], stock + f[i]) + 1):
                new_stock = min(stock + f[i] - t, s)
                profit_next, trade_seq_next = dp(i + 1, new_stock)
                
                profit = p[i] * t + profit_next
                if profit > max_profit:
                    max_profit = profit
                    best_trade_seq = [t] + trade_seq_next
            
            return max_profit, best_trade_seq
        
        max_profit, trade = dp(0, 0)
        return trade, max_profit

    # Example usage
    p = [10, 20, 15, 25, 5, 10, 20, 15, 25, 5, 10, 20, 15, 25, 5, 10, 20, 15, 25, 5]   # Prices
    v = [5, 5, 5, 5, 5, 5, 5, 5, 5, 5, 5, 5, 5, 5, 5, 5, 5, 5, 5, 5]       # Max volumes per timestamp
    f = [3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3]       # Free incoming goods
    s = 15                    # Max stock

    greedy_result2, greedy_profit2 = greedy_trade2(p, v, f, s)
    print("Greedy2 trade volumes:", greedy_result2, "Profit:", greedy_profit2)
    dp_result, dp_profit = dp_trade(p, v, f, s)
    print("DP trade volumes:", dp_result, "Profit:", dp_profit)
  #+end_src

  #+RESULTS[0a3dec4c13df602c7ccb8910b9dff31aabb4a563]:
  : Greedy2 trade volumes: [3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3] Profit: 900
  : DP trade volumes: [0, 5, 2, 5, 0, 0, 5, 5, 5, 0, 0, 5, 5, 5, 0, 0, 5, 5, 5, 3] Profit: 1170

```

scriptforfivem, 2025-05-10:
I'm impressed by the modularity of this codebase. Makes it very maintainable.