Akkudoktor-EOS/EOS #495: [BUG]: Cross Site Scripting

aydinnyunus, 2025-03-26:
### Describe the issue:

### XSS Report for `proxy` function: Path Parameter Injection

#### Overview:
The `proxy` function forwards requests to a backend server (EOSdash) by dynamically constructing the URL based on the `path` parameter passed in the request. This `path` parameter is directly embedded into the URL without any sanitization or validation. This can expose the system to **Cross-Site Scripting (XSS)** attacks if malicious input is provided via the `path`.

- https://github.com/Akkudoktor-EOS/EOS/blob/main/src/akkudoktoreos/server/eos.py#L910

#### Identified Issue:
The core issue lies in the **path parameter** passed to the proxy function. If the `path` parameter contains malicious scripts or HTML elements, it can potentially be executed in the client’s browser if it is improperly handled in subsequent responses.

#### Specific Vulnerability:
1. **Path Parameter Exposure**:
   The `path` parameter is directly used to construct the URL in the following line:
   ```python
   url = f"http://{host}:{config_eos.server_eosdash_port}/{path}"
   ```
   If a malicious actor sends a request with a specially crafted `path` value containing `<script>`, `onerror`, or any other form of executable HTML/JavaScript, this could lead to an XSS vulnerability.

2. **Potential Attack Scenario**:
   An attacker could craft a URL like:
   ```
   http://yourdomain.com/proxy/<PAYLOAD>
   ```
   If this path is used to generate a URL that eventually gets displayed or executed in the browser, it could trigger an XSS attack.

### Reproduceable code example:

```python

```

### Error message:

```shell

```

### Version information:

latest

b0661, 2025-03-27:
Thank you for the hint. What is your suggestion for change? Can you provide a pull request?

aydinnyunus, 2025-03-27:
Hi,

I did not analyze the feature branch yet but I create PR for the default branch

NormannK, 2025-03-27:
I accepted PR #497. id you fix the precommit problems ill accept it. Feel free to also make a PR for the dev branch.

aydinnyunus, 2025-03-27:
Can you retrigger ?

aydinnyunus, 2025-03-27:
It should be fine know.

[image]

b0661, 2025-04-20:
Fixed by #497

closed 2025-04-20 by b0661