Akkudoktor-EOS/EOS #533: [BUG]: cyclic import on /optimize

celle1234, 2025-04-22:
### Describe the issue:

When using /optimize with the new merged version I get a cyclic import error.

### Reproduceable code example:

```python
curl --json @/tmp/postEOS-testneu.txt http://10.0.3.13:8505/optimize?start_hour=21

cat /tmp/postEOS-testneu.txt
{
  "ems" : {
"pv_prognose_wh" : [ 0, 0, 0, 0, 0, 0, 2, 41, 93, 217, 2849, 2041, 2121, 1929, 1572, 1237, 981, 677, 330, 75, 2, 0, 0, 0 ,
0, 0, 0, 0, 0, 0, 2, 41, 93, 217, 2849, 2041, 2121, 1929, 1572, 1237, 981, 677, 330, 75, 2, 0, 0, 0 ]
,
"strompreis_euro_pro_wh" : [
0.0003189,0.0003118,0.0003120,0.0003110,0.0003129,0.0003325,0.0003582,0.0003750,0.0003450,0.0003270,0.0003091,0.0002904,0.0002562,0.0002408,0.0002439,0.0002796,0.0002951,0.0003250,0.0003546,0.0003885,0.0003766,0.0003582,0.0003447,0.0003315,0.0003189,0.0003118,0.0003120,0.0003110,0.0003129,0.0003325,0.0003582,0.0003750,0.0003450,0.0003270,0.0003091,0.0002904,0.0002562,0.0002408,0.0002439,0.0002796,0.0002951,0.0003250,0.0003546,0.0003885,0.0003766,0.0003582,0.0003447,0.0003315]
,
"gesamtlast" :
[250,250,250,250,250,250,650,350,350,350,1400,1600,800,350,1350,350,350,350,500,900,650,250,250,250,250,250,250,250,250,250,650,350,350,350,1400,1600,800,350,1350,350,350,350,500,900,650,250,250,250]
,
"einspeiseverguetung_euro_pro_wh": [
0.0000880,0.0000820,0.0000822,0.0000814,0.0000830,0.0000994,0.0001210,0.0001351,0.0001099,0.0000948,0.0000798,0.0000641,0.0000353,0.0000224,0.0000249,0.0000550,0.0000680,0.0000932,0.0001180,0.0001465,0.0001365,0.0001210,0.0001097,0.0000986,0.0000880,0.0000820,0.0000822,0.0000814,0.0000830,0.0000994,0.0001210,0.0001351,0.0001099,0.0000948,0.0000798,0.0000641,0.0000353,0.0000224,0.0000249,0.0000550,0.0000680,0.0000932,0.0001180,0.0001465,0.0001365,0.0001210,0.0001097,0.0000986],
"preis_euro_pro_wh_akku": 0.00020
},
"pv_akku": {
    "device_id" : "akku",
    "capacity_wh": 5500,
    "charging_efficiency": 0.87,
    "discharging_efficiency": 0.87,
    "max_charge_power_w": 2500,
    "initial_soc_percentage": "23.00",
    "min_soc_percentage": 5,
    "max_soc_percentage": 100
  },
  "inverter": {
    "max_power_wh": 4200,
    "device_id" : "inverter"
  },
  "eauto": {
    "device_id" : "auto",
    "capacity_wh": 37000,
    "charging_efficiency": 0.80,
    "discharging_efficiency": 1,
    "max_charge_power_w": 11000,
    "initial_soc_percentage": 100,
    "min_soc_percentage": 10,
    "max_soc_percentage": 10
  },
"temperature_forecast" : [ 11.7,11.8,12,11.5,11.6,11,10.7,10.6,11.2,11.9,13.1,14.5,15.9,16.6,16.7,14.9,12.1,10.5,10.2,9.7,8.8,8.2,7.9,7.8,7.6,7.4,7.4,7.1,6.8,6.5,6.3,6.3,6.4,6.6,7.1,7.4,8.1,8.7,9.5,9.6,9.8,9.7,9,8.3,7.2,6.6,6.5,6.4
]
,
"start_solution" : [17,11,13,20,10,0,16,18,9,19,22,5,22,22,22,22,22,22,10,12,7,7,8,10,10,14,0,4,11,11,11,11,7,22,22,13,22,22,22,22,22,22,11,12,11,10,12,10]
}
```

### Error message:

```shell
Application startup complete.
INFO:     Uvicorn running on http://0.0.0.0:8505 (Press CTRL+C to quit)
ERROR:    Exception in ASGI application
Traceback (most recent call last):
  File "/home/eos/eos-test/EOS-main/.venv/lib/python3.11/site-packages/uvicorn/protocols/http/httptools_impl.py", line 409, in run_asgi
    result = await app(  # type: ignore[func-returns-value]
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/eos/eos-test/EOS-main/.venv/lib/python3.11/site-packages/uvicorn/middleware/proxy_headers.py", line 60, in __call__
    return await self.app(scope, receive, send)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/eos/eos-test/EOS-main/.venv/lib/python3.11/site-packages/fastapi/applications.py", line 1054, in __call__
    await super().__call__(scope, receive, send)
  File "/home/eos/eos-test/EOS-main/.venv/lib/python3.11/site-packages/starlette/applications.py", line 112, in __call__
    await self.middleware_stack(scope, receive, send)
  File "/home/eos/eos-test/EOS-main/.venv/lib/python3.11/site-packages/starlette/middleware/errors.py", line 187, in __call__
    raise exc
  File "/home/eos/eos-test/EOS-main/.venv/lib/python3.11/site-packages/starlette/middleware/errors.py", line 165, in __call__
    await self.app(scope, receive, _send)
  File "/home/eos/eos-test/EOS-main/.venv/lib/python3.11/site-packages/starlette/middleware/exceptions.py", line 62, in __call__
    await wrap_app_handling_exceptions(self.app, conn)(scope, receive, send)
  File "/home/eos/eos-test/EOS-main/.venv/lib/python3.11/site-packages/starlette/_exception_handler.py", line 53, in wrapped_app
    raise exc
  File "/home/eos/eos-test/EOS-main/.venv/lib/python3.11/site-packages/starlette/_exception_handler.py", line 42, in wrapped_app
    await app(scope, receive, sender)
  File "/home/eos/eos-test/EOS-main/.venv/lib/python3.11/site-packages/starlette/routing.py", line 714, in __call__
    await self.middleware_stack(scope, receive, send)
  File "/home/eos/eos-test/EOS-main/.venv/lib/python3.11/site-packages/starlette/routing.py", line 734, in app
    await route.handle(scope, receive, send)
  File "/home/eos/eos-test/EOS-main/.venv/lib/python3.11/site-packages/starlette/routing.py", line 288, in handle
    await self.app(scope, receive, send)
  File "/home/eos/eos-test/EOS-main/.venv/lib/python3.11/site-packages/starlette/routing.py", line 76, in app
    await wrap_app_handling_exceptions(app, request)(scope, receive, send)
  File "/home/eos/eos-test/EOS-main/.venv/lib/python3.11/site-packages/starlette/_exception_handler.py", line 53, in wrapped_app
    raise exc
  File "/home/eos/eos-test/EOS-main/.venv/lib/python3.11/site-packages/starlette/_exception_handler.py", line 42, in wrapped_app
    await app(scope, receive, sender)
  File "/home/eos/eos-test/EOS-main/.venv/lib/python3.11/site-packages/starlette/routing.py", line 73, in app
    response = await f(request)
               ^^^^^^^^^^^^^^^^
  File "/home/eos/eos-test/EOS-main/.venv/lib/python3.11/site-packages/fastapi/routing.py", line 301, in app
    raw_response = await run_endpoint_function(
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/eos/eos-test/EOS-main/.venv/lib/python3.11/site-packages/fastapi/routing.py", line 214, in run_endpoint_function
    return await run_in_threadpool(dependant.call, **values)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/eos/eos-test/EOS-main/.venv/lib/python3.11/site-packages/starlette/concurrency.py", line 37, in run_in_threadpool
    return await anyio.to_thread.run_sync(func)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/eos/eos-test/EOS-main/.venv/lib/python3.11/site-packages/anyio/to_thread.py", line 56, in run_sync
    return await get_async_backend().run_sync_in_worker_thread(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/eos/eos-test/EOS-main/.venv/lib/python3.11/site-packages/anyio/_backends/_asyncio.py", line 2470, in run_sync_in_worker_thread
    return await future
           ^^^^^^^^^^^^
  File "/home/eos/eos-test/EOS-main/.venv/lib/python3.11/site-packages/anyio/_backends/_asyncio.py", line 967, in run
    result = context.run(func, *args)
             ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/eos/eos-test/EOS-main/src/akkudoktoreos/server/eos.py", line 1150, in fastapi_optimize
    result = opt_class.optimierung_ems(parameters=parameters, start_hour=start_hour, **extra_args)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/eos/eos-test/EOS-main/src/akkudoktoreos/optimization/genetic.py", line 574, in optimierung_ems
    self.devices.reset()
    ^^^^^^^^^^^^
  File "/home/eos/eos-test/EOS-main/src/akkudoktoreos/core/coreabc.py", line 182, in devices
    from akkudoktoreos.devices.devices import get_devices
  File "/home/eos/eos-test/EOS-main/src/akkudoktoreos/devices/devices.py", line 43, in <module>
    devices = Devices()
              ^^^^^^^^^
  File "/home/eos/eos-test/EOS-main/src/akkudoktoreos/devices/devices.py", line 35, in __init__
    self.post_setup()
  File "/home/eos/eos-test/EOS-main/src/akkudoktoreos/devices/devices.py", line 39, in post_setup
    device.post_setup()
  File "/home/eos/eos-test/EOS-main/src/akkudoktoreos/devices/devicesabc.py", line 144, in post_setup
    self._post_setup()
  File "/home/eos/eos-test/EOS-main/src/akkudoktoreos/devices/inverter.py", line 45, in _post_setup
    self.battery = self.devices.get_device_by_id(self.parameters.battery_id)
                   ^^^^^^^^^^^^
  File "/home/eos/eos-test/EOS-main/src/akkudoktoreos/core/coreabc.py", line 182, in devices
    from akkudoktoreos.devices.devices import get_devices
ImportError: cannot import name 'get_devices' from partially initialized module 'akkudoktoreos.devices.devices' (most likely due to a circular import) (/home/eos/eos-test/EOS-main/src/akkudoktoreos/devices/devices.py)
```

### Version information:

commit 8b880299b851a80e8e15e5741a8cbda9f8818c27 (HEAD -> main, origin/main, origin/HEAD)

NormannK, 2025-04-22:
which python version are you using?

Lasall, 2025-04-22:
With your example input I get a server 500 with `akkudoktoreos.devices.inverter - ERROR - Battery for PV inverter is mandatory.`

celle1234, 2025-04-23:
Here is my config /v1/config
`
{
  "general": {
    "data_folder_path": "/home/eos/eos-test/EOS-main",
    "data_output_subpath": "output",
    "latitude": <x>,
    "longitude": <y>,
    "timezone": "Europe/Berlin",
    "data_output_path": "/home/eos/eos-test/EOS-main/output",
    "config_folder_path": "/home/eos/eos-test/EOS-main",
    "config_file_path": "/home/eos/eos-test/EOS-main/EOS.config.json"
  },
  "cache": {
    "subpath": "cache",
    "cleanup_interval": 300
  },
  "ems": {
    "startup_delay": 5,
    "interval": null
  },
  "logging": {
    "level": "INFO",
    "root_level": "INFO"
  },
  "devices": {
    "batteries": [
      {
        "device_id": "byd",
        "hours": null,
        "capacity_wh": 5500,
        "charging_efficiency": 0.87,
        "discharging_efficiency": 0.87,
        "max_charge_power_w": 2500,
        "initial_soc_percentage": 0,
        "min_soc_percentage": 5,
        "max_soc_percentage": 100
      }
    ],
    "inverters": [
      {
        "device_id": "kostal",
        "hours": null,
        "max_power_wh": 4200,
        "battery_id": "byd"
      }
    ],
    "home_appliances": null
  },
  "measurement": {
    "load0_name": null,
    "load1_name": null,
    "load2_name": null,
    "load3_name": null,
    "load4_name": null
  },
  "optimization": {
    "hours": 48,
    "penalty": 10,
    "ev_available_charge_rates_percent": [
      0,
      0.375,
      0.5,
      0.625,
      0.75,
      0.875,
      1
    ]
  },
  "prediction": {
    "hours": 48,
    "historic_hours": 48
  },
  "elecprice": {
    "provider": "ElecPriceAkkudoktor",
    "charges_kwh": 0.214,
    "provider_settings": null
  },
  "load": {
    "provider": "LoadAkkudoktor",
    "provider_settings": {
      "loadakkudoktor_year_energy": 3500
    }
  },
  "pvforecast": {
    "provider": "PVForecastAkkudoktor",
    "provider_settings": null,
    "planes": [
      {
        "surface_tilt": 38,
        "surface_azimuth": -45,
        "userhorizon": null,
        "peakpower": 3.72,
        "pvtechchoice": "crystSi",
        "mountingplace": "free",
        "loss": 14,
        "trackingtype": null,
        "optimal_surface_tilt": false,
        "optimalangles": false,
        "albedo": null,
        "module_model": null,
        "inverter_model": null,
        "inverter_paco": 4200,
        "modules_per_string": null,
        "strings_per_inverter": null
      },
      {
        "surface_tilt": 38,
        "surface_azimuth": -45,
        "userhorizon": null,
        "peakpower": 0,
        "pvtechchoice": "crystSi",
        "mountingplace": "free",
        "loss": 14,
        "trackingtype": null,
        "optimal_surface_tilt": false,
        "optimalangles": false,
        "albedo": null,
        "module_model": null,
        "inverter_model": null,
        "inverter_paco": 0,
        "modules_per_string": null,
        "strings_per_inverter": null
      }
    ],
    "max_planes": 0,
    "planes_peakpower": [
      3.72,
      0
    ],
    "planes_azimuth": [
      -45,
      -45
    ],
    "planes_tilt": [
      38,
      38
    ],
    "planes_userhorizon": [
      [
        0,
        0
      ],
      [
        0,
        0
      ]
    ],
    "planes_inverter_paco": [
      4200,
      0
    ]
  },
  "weather": {
    "provider": null,
    "provider_settings": null
  },
  "server": {
    "host": "0.0.0.0",
    "port": 8505,
    "verbose": true,
    "startup_eosdash": true,
    "eosdash_host": "0.0.0.0",
    "eosdash_port": 8506
  },
  "utils": {}
}
`
The extra PV with 0 W in necessary because of another bug (https://akkudoktor.net/t/hilfe-fuer-neuling-bei-eos/28167).

How can I help to fix the problem?

Is there a working example to use /optimize in newer versions? In older ones before merging of the dev branch it was working as expected.

celle1234, 2025-04-23:
The fix seems to work.

b0661, 2025-04-24:
Sorry

commit 7ade15e9 (2025-04-23): fix: Circular runtime import Closes #533 (#535)
closed 2025-04-24 by b0661