Akkudoktor-EOS/EOS #561: UNsble to Save config to file

ThomasW69, 2025-05-25:
I use the Docker image in portainer on a raspberry PI5. When trying to export the config to file via EOSDash Admin panel I will get the follwoing error message. How to solve this.  

[image]

b0661, 2025-05-26:
EOSDash tries to import/ export files from https://github.com/Akkudoktor-EOS/EOS/blob/f88260dc3fd53e3f5cf797edacb48badace52761/src/akkudoktoreos/server/dash/admin.py#L40
If this directory is missing or cannot be accessed a failure message is raised.
There seems to be no valid user (directory) for the EOSDash user. I´m not in Docker, so the the Docker specialists should have a look.

maxx3105, 2025-06-01:
I get the same message on an Synology Nas Docker Container. 

[image]

[image]