Akkudoktor-EOS/EOS #580: [BUG]: No Prediction Updates

SamLue, 2025-06-07:
### Describe the issue:

In first step I only want to use the pv prediction.

I configrued

{
  "general": {
    "latitude": ??.???
    "longitude": ??.???
  },
  "ems": {
    "startup_delay": 5,
    "interval": 300
  },
  "logging": {
    "level": "DEBUG",
    "root_level": "DEBUG"
  },
  "pvforecast": {
    "provider": "PVForecastAkkudoktor",
    "provider_settings": null,
    "planes": [
      {
...
      }
    ],
}

Everything works, if I start my docker container und Request the predictions with GET /v1/prediction/series?key=pvforecast_ac_power

But if I try to update the prediction with POST /v1/prediction/update, the ansered data from GET /v1/prediction/series?key=pvforecast_ac_power is still old.

For Example: If I request now (07.06.2025 12:30) the predictions, the first Value is with datetime from container starting time:

{
  "data": {
    "2025-06-06T08:00:00+00:00": 2537.9403621949537,
    "2025-06-06T09:00:00+00:00": 2882.138672726043,
    "2025-06-06T10:00:00+00:00": 7040.250401620451,
    ....
    "2025-06-08T07:00:00+00:00": 305.407423718182
  },
  "dtype": "float64",
  "tz": "Etc/UTC"
}

### Reproduceable code example:

```python
- Config like described
- Wait some hours after container start.
POST /v1/prediction/update
GET /v1/prediction/series?key=pvforecast_ac_power
- Check if data is still old
```

### Error message:

```shell
With Log Level INFO there is no Log Entry if i try to update prediction with the POST reqeust.

With Log Level DEBUG I get about 48.5 MB of Loggingdata with one request.

Here some lines:
eos-1  | Timezone adapted to: Etc/UTC
eos-1  | from: 2025-06-08 06:00:00+00:00 tz=FixedTimezone(0, name="+00:00")
eos-1  | to:   2025-06-08 06:00:00+00:00 tz=Etc/UTC
eos-1  | 2025-06-07 10:26:58,597 - akkudoktoreos.utils.datetimeutil - DEBUG - 2025-06-08T07:00:00+00:00, YYYY-MM-DD, String does not match format YYYY-MM-DD z
eos-1  | 2025-06-07 10:26:58,598 - akkudoktoreos.utils.datetimeutil - DEBUG - 2025-06-08T07:00:00+00:00, DD/MM/YY, String does not match format DD/MM/YY z
eos-1  | 2025-06-07 10:26:58,598 - akkudoktoreos.utils.datetimeutil - DEBUG - 2025-06-08T07:00:00+00:00, DD/MM/YYYY, String does not match format DD/MM/YYYY z
eos-1  | 2025-06-07 10:26:58,598 - akkudoktoreos.utils.datetimeutil - DEBUG - 2025-06-08T07:00:00+00:00, MM-DD-YYYY, String does not match format MM-DD-YYYY z
eos-1  | 2025-06-07 10:26:58,598 - akkudoktoreos.utils.datetimeutil - DEBUG - 2025-06-08T07:00:00+00:00, D.M.YYYY, String does not match format D.M.YYYY z
eos-1  | 2025-06-07 10:26:58,598 - akkudoktoreos.utils.datetimeutil - DEBUG - 2025-06-08T07:00:00+00:00, YYYY.MM.DD, String does not match format YYYY.MM.DD z
eos-1  | 2025-06-07 10:26:58,598 - akkudoktoreos.utils.datetimeutil - DEBUG - 2025-06-08T07:00:00+00:00, D MMM YYYY, String does not match format D MMM YYYY z
eos-1  | 2025-06-07 10:26:58,598 - akkudoktoreos.utils.datetimeutil - DEBUG - 2025-06-08T07:00:00+00:00, D MMMM YYYY, String does not match format D MMMM YYYY z
eos-1  | 2025-06-07 10:26:58,599 - akkudoktoreos.utils.datetimeutil - DEBUG - 2025-06-08T07:00:00+00:00, YYYY-MM-DD HH:mm:ss, String does not match format YYYY-MM-DD HH:mm:ss z
eos-1  | 2025-06-07 10:26:58,599 - akkudoktoreos.utils.datetimeutil - DEBUG - 2025-06-08T07:00:00+00:00, YYYY-MM-DDTHH:mm:ss, String does not match format YYYY-MM-DDTHH:mm:ss z
eos-1  | 2025-06-07 10:26:58,599 - akkudoktoreos.utils.datetimeutil - DEBUG - Pendulum Fmt converted: 2025-06-08 07:00:00+00:00, tz=FixedTimezone(0, name="+00:00") from 2025-06-08T07:00:00+00:00, tz=Etc/UTC
eos-1  | 2025-06-07 10:26:58,599 - akkudoktoreos.utils.datetimeutil - DEBUG - 
eos-1  | Timezone adapted to: Etc/UTC
eos-1  | from: 2025-06-08 07:00:00+00:00 tz=FixedTimezone(0, name="+00:00")
eos-1  | to:   2025-06-08 07:00:00+00:00 tz=Etc/UTC
eos-1  | 2025-06-07 10:29:54,108 - akkudoktoreos.utils.datetimeutil - DEBUG - 
eos-1  | Timezone adapted to: Etc/UTC
```

### Version information:

            "Labels": {
...
"sha256:79f38539066731b10fabe46dcc4d2afe6ec14dc387d94c15b8d1a5ac6ea56347",
...
}

SamLue, 2025-06-07:
The recalculation of the prediction works, because I determined changes in the prediction values.
So the only problem seems to be, that the time window is not updated as expected.
The Window is always with start from time where the container started till + 48h (configurable)

b0661, 2025-06-07:
Maybe this endpoint is better suited to your needs: https://akkudoktor-eos.readthedocs.io/en/latest/akkudoktoreos/serverapi.html#get-v1-prediction-list.

I assume you are using [GET PVForecast](https://akkudoktor-eos.readthedocs.io/en/latest/akkudoktoreos/serverapi.html#get-pvforecast). Which has this specification:

```txt
Deprecated: PV Forecast Prediction.

Endpoint to handle PV forecast prediction.

PVForecast starts at 00.00.00 today and is provided for 48 hours.
If no forecast values are available the missing ones at the start of the series are
filled with the first available forecast value.

Note:
    Set PVForecastAkkudoktor as provider, then update data with
    '/v1/prediction/update'
    and then request data with
    '/v1/prediction/list?key=pvforecast_ac_power' and
    '/v1/prediction/list?key=pvforecastakkudoktor_temp_air' instead.
``

b0661, 2025-06-10:
Did you test the demo tab in EOSdash? Does the demo provide actual data?

b0661, 2025-06-11:
#584 is now in main. You may give it a try.

SamLue, 2025-06-11:
As you can see in the Screenshot, also the demo predictions with start in the past (time of system start).

[image]

I will install the new version and report my test here.

SamLue, 2025-06-11:
OK,

I updated and got an error at first start.

```
eos-1  | Traceback (most recent call last):
eos-1  |   File "/opt/eos/src/akkudoktoreos/server/eos.py", line 61, in <module>
eos-1  |     config_eos = get_config()
eos-1  |                  ^^^^^^^^^^^^
eos-1  |   File "/opt/eos/src/akkudoktoreos/config/config.py", line 540, in get_config
eos-1  |     return ConfigEOS()
eos-1  |            ^^^^^^^^^^^
eos-1  |   File "/opt/eos/src/akkudoktoreos/config/config.py", line 379, in __init__
eos-1  |     self._setup(self, *args, **kwargs)
eos-1  |   File "/opt/eos/src/akkudoktoreos/config/config.py", line 395, in _setup
eos-1  |     SettingsEOSDefaults.__init__(self, *args, **kwargs)
eos-1  |   File "/usr/local/lib/python3.12/site-packages/pydantic_settings/main.py", line 176, in __init__
eos-1  |     super().__init__(
eos-1  |   File "/usr/local/lib/python3.12/site-packages/pydantic/main.py", line 253, in __init__
eos-1  |     validated_self = self.__pydantic_validator__.validate_python(data, self_instance=self)
eos-1  |                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
eos-1  | pydantic_core._pydantic_core.ValidationError: 1 validation error for ConfigEOS
eos-1  | pvforecast.planes.0.surface_azimuth
eos-1  |   Input should be greater than or equal to 0 [type=greater_than_equal, input_value=-72.0, input_type=float]
eos-1  |     For further information visit https://errors.pydantic.dev/2.11/v/greater_than_equal
```

Befor updated eos accepted negativ values for "surface_azimuth" the config:

"surface_azimuth": -72.0

Now, I had to chang it to:

"surface_azimuth": 288.0

But that was a bit offtopic.

Back to the prediction time window. Unfortunately no changes with the update.
Test was again:
eos-1  | 2025-06-11 08:01:18.602 | INFO     | logging:callHandlers:1762 - 192.168.178.201:50083 - "POST /v1/prediction/update HTTP/1.1" 200
eos-1  | 2025-06-11 08:01:33.276 | INFO     | logging:callHandlers:1762 - 192.168.178.201:50085 - "GET /v1/prediction/series?key=pvforecast_ac_power HTTP/1.1" 200

Result-Body of prediction/series:

```
{
  "data": {
    "2025-06-11T07:00:00+00:00": 4939.185308531231,
    "2025-06-11T08:00:00+00:00": 6291.929389910546,
    ...
    "2025-06-13T06:00:00+00:00": 937.6764007461029
  },
  "dtype": "float64",
  "tz": "Etc/UTC"
}
```

With Update-Time 2025-06-11 08:01:18.602 I would expect:

```
{
  "data": {
    "2025-06-11T08:00:00+00:00": ???.???,
    ...
    "2025-06-13T07:00:00+00:00": ???.???  },
  "dtype": "float64",
  "tz": "Etc/UTC"
}
```

b0661, 2025-06-11:
Thank you for debugging and the info.

I think I now have a clue what is happening. 

"POST /v1/prediction/update HTTP/1.1" does not set a new start time for the energy management run. As you are requesting the prediction without start time definition "GET /v1/prediction/series?key=pvforecast_ac_power HTTP/1.1",  you get the default start time which happens to be the start time of  the latest energy management run.  As there was no energy management run for a long time -> you got it.

The short time solution is to force the energy management run start time at "POST /v1/prediction/update HTTP/1.1". This will break further development as prediction updates shall be in sync with optimization runs, but we have no optimization if there is only a prediction update. The leading definition for start time shall come from the optimization (and not from the prediction).

I think I will go for the short time solution, but let me think a little bit about it.

> Befor updated eos accepted negativ values for "surface_azimuth" the config:

Yes, this was a breaking change. It was announced in the commit and on akkudoktor.net.

SamLue, 2025-06-12:
OK,

do you think to run optimizations would solve the problem?

I could check that, if it helps!?

What are the minimum Input-Parameters for optimization? And what have to be configrued?
I have already configured:
devices->batteries->0
devices->inverters->0
and my pvforecast

And how should I configure the ems->intervall now? Is it fixed with the new version and should I configure an interval to 300 sec. again?

SamLue, 2025-06-15:
Thanks!

Now it works.

I configured the ems->interval again.
And now I get data by requesting "GET /v1/prediction/series?key=pvforecast_ac_power" or "GET /v1/prediction/list?key=pvforecast_ac_power" in the expected time window.

Let me know, if I should test some other scenarios. For example the update over "POST /v1/prediction/update".
Otherwise you can close this issue.

b0661, 2025-06-17:
> Let me know, if I should test some other scenarios. For example the update over "POST /v1/prediction/update". Otherwise you can close this issue.

I will not close until automatic energy management runs can also do optimization correctly (not only predictions). See #596 .

You may test #596. It is still very much work in progress. So expect bugs.