Akkudoktor-EOS/EOS #629: Docker: EOSDash not available

andig, 2025-07-21:
Laut https://akkudoktor-eos.readthedocs.io/en/latest/_generated/config.html#server-configuration ist EOSDash per Default verfügbar. Nach Start von

    docker run -p 8503:8503 -p 8504:8504 --name eos akkudoktor/eos

scheitert

    curl http://localhost:8504
    curl: (56) Recv failure: Connection reset by peer

Liegt das evtl. an `EOS_SERVER__EOSDASH_HOST=127.0.0.1` statt `0.0.0.0`?

MubiTec, 2025-07-21:
Ja, liegt aus meiner Sicht daran. Beide Hosts sind bei mir auch nur über den Server erreichbar. Hab es jetzt selber zu 0.0.0.0 geändert, nun geht's auch vom Laptop.
Wäre aus meiner Sicht sinnvoll als default

andig, 2025-07-22:
Hat das bei Dir über setzen den ENV Variable von aussen geklappt? Das schien bei mir nicht zu wirken?

MubiTec, 2025-07-22:
Hab erst ganz normal installiert und dann diesen Befehl abgesetzt.
curl -X PUT "http://localhost:8503/v1/config/server/eosdash_host" \
     -H "Content-Type: application/json" \
     -d '"0.0.0.0"'

Bzw.

curl -X PUT "http://localhost:8503/v1/config/server/host" \
     -H "Content-Type: application/json" \
     -d '"0.0.0.0"'

PR #656 (2025-08-21): fix(docker): make EOSDash accessible in Docker containers
Fixes the issue where EOSDash is not accessible when running EOS in a Docker container. The problem was that EOSDash was binding to 127.0.0.1 (localhost) by default, making it inaccessible from outside the container.

The fix adds a minimal default configuration file during the Docker image build that sets both the EOS server and EOSDash to bind to 0.0.0.0, allowing external access while maintaining security through Docker's network isolation.

- Add default EOS.config.json in Dockerfile with server bindings set to 0.0.0.0
- No changes  [... cut]
commit cacb2152 (2025-08-22): fix(docker): make EOSDash accessible in Docker containers (#656)

Fixes the issue where EOSDash is not accessible when running EOS in a
Docker container. The problem was that EOSDash was binding to 127.0.0.1
(localhost) by default, making it inaccessible from outside the container.

The fix adds a minimal default configuration file during the Docker image
build that sets both the EOS server and EOSDash to bind to 0.0.0.0,
allowing external access while maintaining security through Docker's
network isolation.

- Add default EOS.config.json in Dockerfile with server bindings set to 0.0.0.0
- No  [... cut]