Akkudoktor-EOS/EOS #630: FR: /optimize response ev charging signals

andig, 2025-07-21:
I'm trying to comprehend https://akkudoktor-eos.readthedocs.io/en/latest/akkudoktoreos/optimization.html:

In evcc, an ev is treated as flexible load, moving the load to the cheapest hours available. In the api response, EV charging is mixed into `Last_Wh_pro_Stunde`. It is not clear (maybe indirectly from growing social) if EV should be charged according to optimization model.

It would be great if ev charging, similar to battery AC charging, was clearly indicated in the `/optimize` response

b0661, 2025-07-23:
As I understand it the eautocharge_hours_float tells you the charge rate for the EV battery in proportion to the maximum charge rate [0.0..1.0]. The charge rate is provided by hour starting at 00:00:00. The optimization specialist may have a better explanation @NormannK, @drbacke.

In #596  I´m working on an energy management plan that translates the current optimization result. Maybe the code helps to understand the concept of eutocharge_hours_float. 

See here for the original definition: https://github.com/Akkudoktor-EOS/EOS/blob/0c8a9efb656f5d90c655c0f590fd3e592653db64/src/akkudoktoreos/optimization/genetic.py#L71

See here for the creation of a plan: https://github.com/Akkudoktor-EOS/EOS/blob/3633957c818af4a7bfeec8d7a69dffd2012e63a9/src/akkudoktoreos/optimization/geneticsolution.py#L194-L225

b0661, 2025-10-28:
Implemented